fredlcore/BSB-LAN #247: Feature request: Query parameter with target adress

MasterOfPuppets111, 2020-12-31:
same like (/P<x>,<y>,<z>) but for query a value.

fredlcore, 2020-12-31:
If we plan to implement this, I'd suggest we do it the same way as with write, like this if someone wants to query 710 from ID 1 and 712 from ID2:
/710!1/712!0

MasterOfPuppets111, 2020-12-31:
That would be awesome!

fredlcore, 2021-01-01:
Done, at least for ordinary URL-based queries, don't know if this works for JSON/MQTT queries as well.
Use for example /6224!10 to query the identification of the display unit. Also, ranges are possible with the "!" after the first parameter: /6220!10-6229
@1coderookie: Could be useful for the manual as well?

closed 2021-01-01 by fredlcore

dukess, 2021-01-01:
Do we need same function for JSON commands? https://github.com/dukess/bsb_lan/commit/87bffd24c72cc2b2a7c9a6d05aaefd84d3637d98

MasterOfPuppets111, 2021-01-01:
please @dukess =)

fredlcore, 2021-01-01:
I was just about to ask :)

1coderookie, 2021-01-02:
Great job! Does this work for both bus types BSB and LPB as well?

dukess, 2021-01-02:
I do no know but i did it like Frederik for URL command.

"U.

fredlcore, 2021-01-02:
It should.

1coderookie, 2021-01-02:
Sweet! :)

1coderookie, 2021-01-02:
@MasterOfPuppets111 Could you please give feedback once you tested it with JSON? So I can add that note to the manual as well. Thanks

MasterOfPuppets111, 2021-01-02:
@1coderookie in an hour i will bei at home and test it!

MasterOfPuppets111, 2021-01-02:
@1coderookie it doesnt work with JSON =(

/JQ=6224!0 and /JQ=6224!1

both have the same content from LPB Adress 0:
`{
  "6224": {
    "name": "Geräte-Identifikation",
    "dataType_name": "STRING",
    "dataType_family": "STRN",
    "error": 0,
    "value": "RVS43.222/100",
    "desc": "",
    "dataType": 7,
    "readonly": 1,
    "unit": ""
  }
}`

with URL Command:

/6224!0 ->  6224 Konfiguration - Geräte-Identifikation: RVS43.222/100

/6224!1 -> 6224 Konfiguration - Geräte-Identifikation: RVS46.530/100

dukess, 2021-01-03:
It will work with POST and sending parameters through JSON structure.
{"Parameter": 6224, "Destination": 1}