fredlcore/BSB-LAN #249: Empty reply from server

rroblik, 2021-01-02:
Hi @fredlcore,

I just received your BSB car, and my Due + W5500 ethernet shield.

I tried both "blink" then "webserver" example programs successfully.

Now i'm trying to run the BSB_lan programFYI using `bsb-version`
```
#define MAJOR "2"
#define MINOR "0"
#define PATCH "67"
#define COMPILETIME "20201231154027"
```

I enabled `verbose` and `monitor in `BSB_lan_config.h` and openned the IDE monitor which give me the followwing output :
```
READY
Reading EEPROM
Reading done.
EEPROM schema v.4 Program schema v.4
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 418
Address EEPROM option 7: 419
Address EEPROM option 8: 420
Address EEPROM option 9: 421
Address EEPROM option 10: 422
Address EEPROM option 11: 423
Address EEPROM option 12: 427
Address EEPROM option 13: 428
Address EEPROM option 14: 588
Address EEPROM option 15: 589
Address EEPROM option 16: 593
Address EEPROM option 17: 753
Address EEPROM option 18: 759
Address EEPROM option 19: 760
Address EEPROM option 20: 764
Address EEPROM option 21: 768
Address EEPROM option 22: 772
Address EEPROM option 23: 776
Address EEPROM option 24: 778
Address EEPROM option 25: 782
Address EEPROM option 26: 786
Address EEPROM option 27: 850
Address EEPROM option 28: 914
Address EEPROM option 29: 915
Address EEPROM option 30: 916
Address EEPROM option 31: 926
Address EEPROM option 32: 927
Address EEPROM option 33: 1087
Address EEPROM option 34: 1088
Address EEPROM option 35: 1092
Address EEPROM option 36: 1093
Address EEPROM option 37: 1094
Address EEPROM option 38: 1095
Address EEPROM option 39: 1099
Address EEPROM option 40: 1131
Address EEPROM option 41: 1163
Address EEPROM option 42: 1195
Address EEPROM option 43: 1227
Address EEPROM option 44: 1229
Address EEPROM option 45: 1230
Address EEPROM option 46: 1231
Address EEPROM option 47: 1232
Address EEPROM option 48: 1264
Size of cmdtbl1: 29472
Size of cmdtbl2: 38664
Size of cmdtbl3: 26256
free RAM: 85807
Init One Wire bus...
numSensors: 0
PPS settings:
Starting SD..ok
192.168.1.150
255.255.255.0
192.168.1.1
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...2097151 MB free
Start network services

Setup complete
GET /C HTTP/1.1

/C
```

But when I do `curl 192.168.1.150/C ` I've got an empty answer from server : 
```
*   Trying 192.168.1.150:80...
* TCP_NODELAY set
* Connected to 192.168.1.150 (192.168.1.150) port 80 (#0)
> GET /C HTTP/1.1
> Host: 192.168.1.150
> User-Agent: curl/7.68.0
> Accept: */*
>
* Empty reply from server
* Connection #0 to host 192.168.1.150 left intact
curl: (52) Empty reply from server
```

As the Arduino webserver example is working I think I'm ok on hardware side ; but I don't know how to proceed now to solver my issue, can you help me ?

FYI : I don't see my Due on my router connected devices list, strange

fredlcore, 2021-01-02:
Since you are using a board that has been placed in my Arduino before, it could have happened that I forgot to erase the contents of the EEPROM which may still contain my (fixed IP) network settings. To override these settings, you just need to change the variable 
byte UseEEPROM = 1;
to
byte UseEEPROM = 0;
in BSB_lan_config.h and flash the software. Then all configuration values will be read from the BSB_lan_config.h file. Once you have changed the settings to your settings and saved them to EEPROM, you can then change the value back to „1“ and flash the software one last time. Then everything should be running fine.

If you are still having problems after that, just let me know.

closed 2021-01-02 by fredlcore

rroblik, 2021-01-02:
Great perfect thanks a lot !