fredlcore/BSB-LAN #251: add dot after month in VT_SUMMERPERIOD

konne, 2021-01-03:
Hi,

for me, it would be more consistent if the VT_SUMMERPERIOD is written like.

25.10. and not 25.10

bye

dukess, 2021-01-03:
Why it needed? In can be added on clien side.

fredlcore, 2021-01-03:
I think so, too. Unless there is a (flash) memory consuming neutral way of implementing it, I would leave this to the external clients to deal with it.

konne, 2021-01-03:
@dukess 
25.10 could be also for example 25.10 liter, because one dot is the English representation for floats. But 25.10. ist clear not a number because dots are then only used as thousand separators.
Yes, the client can add this, but for a clean consistent API, this is something thats not urgent but should be changed.

fredlcore, 2021-01-03:
Yes, it is not pure design, but many things aren't in this project. It's not done by simply adding a dot because it also affects parsing etc.

fredlcore, 2021-01-03:
Ok, I think since @dukess changed the parsing process to sscanf, it is probably enough to add a dot at both the output as well as the parsing section. Will try if that works.

fredlcore, 2021-01-03:
Adding a dot at the end breaks the one-liner validity test via isNaN in the webinterface, so I have to remove it again before parsing it, but now the output is correct. Will have to rethink syntax checking of entered values whether there is a way to do it as compact as isNaN() while allowing the few exceptions from the isNaN() rule...

closed 2021-01-03 by fredlcore

dukess, 2021-01-03:
@konne reason is clear. Counterargument: this parameter is a string, and there is also a data type field that uniquely allows us to interpret data format.
@fredlcore we can remove validator because all parameters will sent as string and will checked when parsing.
P.S. JS setbit() is newer used.