fredlcore/BSB-LAN #252: JSON API Unit

konne, 2021-01-03:
Hi,

for a JSON API the definition of the UNIT should be "°C" and "&deg;C"
This is fine for HTML but not common in a JSON API.

thx

fredlcore, 2021-01-04:
When I tried it back then some time, it didn't work well on different browsers, that's why we switched to the HTML notation, even though it's not nice, I agree.

konne, 2021-01-04:
Hi, my idea was to make this change only by the unit. I don't know if this is easy possible in your code, but this should me not more than. These are at the moment the only three entries.

```
replace(unit, "&#037;", "%");
replace(unit, "&#181;", "µ");
replace(unit, "&deg;", "°");
```

thx Konrad

fredlcore, 2021-01-04:
That would be a solution, but I know of at least one external module (IP-Symcon) that differentiates values based on the unit.
So the problem is similar to the other issue, I would have to compile a list of external module maintainers and inform them about these changes.

fredlcore, 2021-03-30:
Changed from HTML notation to actual characters system-wide since we are using non-ASCII characters in other places as well.

closed 2021-03-30 by fredlcore