fredlcore/BSB-LAN #255: INF Messages

konne, 2021-01-04:
Hi,

I logged an INF Message 10000 (Roomtemp) with bsbJS. The message was sent from bsbLAN.

`DCC2000E023D2D021504C00026E9                 INF 42 -> 00 0x2D3D0215 Raumtemperatur 1 (10000) = -256.0`

this is the log from the bsbLAN telnet 

```
Setting temporary destination to 0
set ProgNr 10000 = 19
setting line: 10000 val: 04 C0 00
LAN->HEIZ INF      3D2D0215 04 C0 00
DC C2 00 0E 02 3D 2D 02 15 04 C0 00 26 E9
```

What is strange to me:

1. the message is sent to DST: 0x00 and not to 0x7F, because you mentioned in the docu that INF are Broadcast messages
2. the parsing of the value does not work like it is defined in the VT_TEMP, because this are 3 bytes but the first byte should be the enable byte and not already a value byte.

bye

fredlcore, 2021-01-04:
1. Does the manual say that all INF telegrams are broadcast messages?
2. Depends on where you look: https://github.com/fredlcore/bsb_lan/blob/fe4a78070b612e4981e94740304238603b46da4a/BSB_lan.ino#L4843-L4847

1coderookie, 2021-01-04:
> 1. Does the manual say that all INF telegrams are broadcast messages?

I guess he is talking about the table in [chap. 2.1.1](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap02.html#211-addressing-within-the-bsb)  / [Kap. 2.1.1](https://1coderookie.github.io/BSB-LPB-LAN/kap02.html#211-adressierung-beim-bsb). 
Afaik, all BCs are INFs - but I didn't say that all INFs are BCs. 
If I'm wrong with the content of the table, please let me know..

fredlcore, 2021-01-04:
No, the table is correct.

konne, 2021-01-04:
@fredlcore can you please close issues not so fast. I think here is still room for a valuable discussion, can you please reopen the issue

> 1. Does the manual say that all INF telegrams are broadcast messages?
@1coderookie: you are right, I read it just in the way that it is 1:1
Have you really seen the INFs that are sent not as broadcast, or only found out with try and error?
For me, it only makes sense that all INFs are also BCs because only in a BCs case it makes sense not to answer.
Maybe the Siemens designers did it, but I would expect something different.

>2. the parsing of the value ...
Here I have different subtopics.

## FL_SPECIAL_INF: Parse

For me, these lines make sense, because an INF Message is like an ANS Message that has a different enable_byte encoding
But then the FL_SPECIAL_INF can be removed
```
if(setcmd){
        param[0]=decodedTelegram.enable_byte;
        param[1]=(t >> 8);
        param[2]= t & 0xff;
      }else{ // INF message type
        if((get_cmdtbl_flags(i) & FL_SPECIAL_INF) == FL_SPECIAL_INF) {  // Case for outside temperature
          param[0]=decodedTelegram.enable_byte-1;
          param[1]=(t >> 8);
          param[2]= t & 0xff;
```
But this doesn't need a FL_SPECIAL_INF flag, because this can be handled just with INF

## FL_SPECIAL_INF: Swap command bytes

How you analyzed those two commands with FL_SPECIAL_INF there the command bytes are swapped.
For me, we could just change the order of the commands in the defs file and remove the code of swap for this INF.
Because INF message has no (N)ACK,ERR,... there you have the swapped / not swapped command.

## Parsing of this really strange INF 10000 Payload structure
Please can we just add a new VT_INF_TEMP Type with a new DT_ type so that it is clear that this has a special payload structure?

fredlcore, 2021-01-04:
> @fredlcore can you please close issues not so fast. I think here is still room for a valuable discussion, can you please reopen the issue

I close issues when the questions asked are answered. You didn't indicate that the two specific questions you asked were part of a larger context. 

> But this doesn't need a FL_SPECIAL_INF flag, because this can be handled just with INF

What does "this" refer to?
We have the case of VT_TEMP. If we want to send it as INF, we have to decide between two cases, "FL_SPECIAL_INF" (for sending (fake) outside temperature to the heater, and room temperature. Only one of the two should be executed, so that's why we need an if-then block where we have to evaluate FL_SPECIAL_INF.

> How you analyzed those two commands with FL_SPECIAL_INF there the command bytes are swapped.
For me, we could just change the order of the commands in the defs file and remove the code of swap for this INF.

I don't understand what you want to say.

> Please can we just add a new VT_INF_TEMP Type with a new DT_ type so that it is clear that this has a special payload structure?

All INFs are special in that way. There are a several still unknown INFs that go across the bus and we don't know their meaning yet. I don't want to make a case (yet) where we introduce new data types for each single specific case but rather address this generally once we know more about them.

konne, 2021-01-04:
> I don't understand what you want to say.

My idea here is the following:

Change the following lines from:
```
{0x0500021F,  CAT_USER_DEFINED,     VT_TEMP,          10003, STR8700,  0,                    NULL,         DEFAULT_FLAG+FL_SPECIAL_INF, DEV_ALL}, // Außentemperatur wie von Funkempfänger übermittelt
{0x053D0521,  CAT_USER_DEFINED,     VT_TEMP,          10004, STR8700,  0,                    NULL,         DEFAULT_FLAG+FL_SPECIAL_INF, DEV_ALL}, // Außentemperatur wie von Funkempfänger übermittelt
```

to:
```
{0x0005021F,  CAT_USER_DEFINED,     VT_TEMP,          10003, STR8700,  0,                    NULL,         DEFAULT_FLAG, DEV_ALL}, // Außentemperatur wie von Funkempfänger übermittelt
{0x3D050521,  CAT_USER_DEFINED,     VT_TEMP,          10004, STR8700,  0,                    NULL,         DEFAULT_FLAG, DEV_ALL}, // Außentemperatur wie von Funkempfänger übermittelt
```
and remove lines:
````
  if((get_cmdtbl_flags(i) & FL_SPECIAL_INF) == FL_SPECIAL_INF) {
    c=((c & 0xFF000000) >> 8) | ((c & 0x00FF0000) << 8) | (c & 0x0000FF00) | (c & 0x000000FF); // because send reverses first two bytes, reverse here already to take care of special inf telegrams that don't reverse first two bytes
  }
```

Does this make sense?

What I don't understand is your wording "(fake) outside". Yes, BSBLan is just sending the inf message, but the message itself is the same message what the outdoor temp would send.

fredlcore, 2021-01-04:
Ok, I think I got it.
The problem with the outside temperature INF (and which is also an example of the seemingly unstandardized way that INFs are used by Siemens) is that it **must** be sent as an INF. This is why we need the `}else{ // INF message type`clause.
Then, we have to differentiate between two ways of sending a temperature. The outside temperature INF telegram has the structure `00 HIGH LOW`. A SET telegram would have 01 in the first byte, so we can't use SET. Also, we have to swap the first two bytes because the Send function swaps them again, and this specific telegram needs it unswapped. 

For sending the room temperature we have to have the data payload in the first two bytes and the last (third) byte is either ignored or irrelevant. So we need an else case for this as well. Here, the first two bytes of the command ID are swapped (done in Send()), as it would be in SET and other INF telegrams we know of. 
So from my understanding we cannot save or simplify the code here (as much as I'd like).

 > What I don't understand is your wording "(fake) outside". Yes, BSBLan is just sending the inf message, but the message itself is the same message what the outdoor temp would send.

Yes, but as mentioned just now, the structure of the outside temperature telegram is different from the room temperature telegram, while the encoding of the value is the same. I guess INF telegrams were the "do what you want" playground for Siemens' programmers ;).

closed 2021-02-08 by fredlcore