fredlcore/BSB-LAN #262: MQTT JSON formatting issue

rroblik, 2021-01-05:
Dear @fredlcore,

I'm trying to use MQTT with bsb_lan but I have seems to have a JSON formatting issue (`byte mqtt_mode = 3;`) :

From the serial monitor for example : 

 ```
23:35:49.729 -> Output topic: BSB-LAN/json
23:35:49.729 -> Payload Output : {"PAC":{"id":8700","name":"STR8700_TEXT","value": "","desc": "","unit": "UNIT_DEG_TEXT","error": 261}}
23:35:52.728 -> Output topic: BSB-LAN/json
23:35:52.728 -> Payload Output : {"PAC":{"id":8743","name":"STR8743_TEXT","value": "","desc": "","unit": "UNIT_DEG_TEXT","error": 261}}
```

The `"` after `8700` break json format so my MQTT broker client see it as string instead of parseable JSON.

From what I see the error is here for all `id` values.

Note : i'm not connected to the bsb bus for now, that why I've 261 errors.

Can you help ?

fredlcore, 2021-01-05:
That's an error, yes. @dukess: Do you know how to fix it? Otherwise I'll look into it.

dukess, 2021-01-06:
Done. https://github.com/fredlcore/bsb_lan/pull/263

Sleepless night.

rroblik, 2021-01-06:
Impressive !

thanks a lot @dukess for the hard work. 

Waiting the merge 👌🏽

dukess, 2021-01-06:
Thanks for report!

rroblik, 2021-01-06:
Can be closed #263 

Thanks !

closed 2021-01-06 by fredlcore