fredlcore/BSB-LAN #264: MQTT single topic vs multiple sub

rroblik, 2021-01-06:
Hi,

I'm trying MQTT to connect my heater with my automation software.

An example a playload (using `byte mqtt_mode = 3;`) looks like `{"PAC":{"id":8700,"name":"STR8700_TEXT","value": "","desc": "","unit": "UNIT_DEG_TEXT","error": 261}}`, received under `{BSB-LAN}/json` topic.

On the automation software side, commands/informations can easily and automatically be created based on topic depth. With this MQTT playload, topic il always `json` ; that's - to my mind - the pain point.

From my POV it should look like (at least) : 

```
BSB-lan/json/8700
BSB-lan/json/8743
...
```

And later, maybe, use "category" to group topic

```
BSB-lan/temperature/outside #8700
BSB-lan/temperature/ambient #8721
...
```

I don't know if the mapping job should be done on BSB Lan side or elsewhere, I'm new user of MQTT protocol  

Thanks

fredlcore, 2021-01-06:
Please try MQTT mode 1 which sends parameter number and value in each line.

rroblik, 2021-01-06:
What a surprise because I read
` // MQTT: 0 - disabled, 1 - send messages in plain text format, 2 - send messages in JSON format, 3 - send messages in rich JSON format.`

But ok let me try it ...

rroblik, 2021-01-06:
Ok working but it's plain text (and not JSON).

Is that normal ?

fredlcore, 2021-01-08:
We're talking about the MQTT functionality here, which, in addition to plain text MQTT messages also gives two options to send parameters in JSON format. If these two JSON formats don't work for you, but the plain text does, I don't see the need to replicate the plain text version into JSON format. Or am I missing something?

rroblik, 2021-01-08:
Hmmm ok ; I need to read more documentation about MQTT protocol.

Thanks

closed 2021-01-08 by rroblik