fredlcore/BSB-LAN #268: Please clarify JL & JW

konne, 2021-01-07:
Hi,

if I read a config via JL and get for example an element with the JSON Key 3 and Id 6
```
  "3": {
    "id": "6",
```
What is important for write. The correct ID (6) or the correct key (3)

bye
Konrad

dukess, 2021-01-07:
"3" is a displaying order. You can ignore it.
"6" is real ID.

konne, 2021-01-08:
@dukess: so if I understand it right I have to send back:

````json
{
  "ANYKEY": {
    "id": "6",                  
    "value": "new Value to Write" 
  }
}
```

correct?

dukess, 2021-01-08:
No. Syntax is same as for /JS because same parser used:
{{"parameter": 6, "value": "new Value to Write"}, {"parameter": 22, "value": "something new"}}

hacki11, 2021-01-08:
Is this command ne or already present in 1.x?
What does this really do? I found 'Reads the list of configuration created with /JL and adjusts the settings' but what configuration can be set? What is different to /JS?

dukess, 2021-01-08:
> Is this command new or already present in 1.x?

No. It is new feature.
/JL and /JW used for configuring BSB-LAN not heater/cooler etc.

hacki11, 2021-01-08:
Is it GET or POST?
Can you give me real-life examples for both in terms of request and response?

dukess, 2021-01-08:
POST. /JS, /JW support POST only.
/JQ, /JC, /JR, can be queried with GET and POST.

/JL, /JI, /JV queried without parameters.

/JL. Payload can be named as combination /JC and /JQ. It contain possible values for ENUMs and current values

{
  "0": {
    "id": "0",
    "type": "1",
    "format": "2",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_USEEEPROM_TEXT",
    "value": "150",
    "possibleValues": [
      { "enumValue": "105", "desc": "MENU_TEXT_OFF" },
      { "enumValue": "150", "desc": "MENU_TEXT_ON" }
      ]
    },
  "3": {
    "id": "6",
    "type": "1",
    "format": "3",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_BUSTYPE_TEXT",
    "value": "0",
    "possibleValues": [
      { "enumValue": "0", "desc": "BSB" },
      { "enumValue": "1", "desc": "LPB" },
      { "enumValue": "2", "desc": "PPS" }
      ]
    },
  "4": {
    "id": "7",
    "type": "1",
    "format": "1",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_OWN_BSBADDR_TEXT",
    "value": "66"
    },
  "5": {
    "id": "8",
    "type": "1",
    "format": "1",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_OWN_LPBADDR_TEXT",
    "value": "66"
    },
  "6": {
    "id": "9",
    "type": "1",
    "format": "1",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_DEST_LPBADDR_TEXT",
    "value": "0"
    },
  "7": {
    "id": "10",
    "type": "1",
    "format": "2",
    "category": "CAT_GENERAL_TEXT",
    "name": "CF_PPS_WRITE_TEXT",
    "value": "0",
    "possibleValues": [
      { "enumValue": "0", "desc": "MENU_TEXT_OFF" },
      { "enumValue": "1", "desc": "MENU_TEXT_ON" }
      ]
    },
  "8": {
[... 527 lines cut ...]
  "44": {
    "status": 1
  },
  "37": {
    "status": 1
  },
  "38": {
    "status": 1
  },
  "39": {
    "status": 1
  },
  "40": {
    "status": 1
  },
  "41": {
    "status": 1
  },
  "42": {
    "status": 1
  },
  "46": {
    "status": 1
  }
}

hacki11, 2021-01-08:
Thanks for the explanation!

Is this intentionally:
- Parameter in /JL as field "id" is of type _string_ and
- parameter in /JW Request as field "parameter" is of type _integer_
Shouldn't it be the same data type?

In 
```
{
            id: '0', 
            type: '1', 
            format: '2', 
            category: 'CAT_GENERAL_TEXT', 
            name: 'CF_USEEEPROM_TEXT', 
            value: '150', 
            possibleValues: [
              {
                enumValue: '105', 
                desc: 'MENU_TEXT_OFF'
              }, 
              {
                enumValue: '150', 
                desc: 'MENU_TEXT_ON'
              }
            ]
          },
```
Can you help me with /JL Response:
- type: Which types exists, what values do they have, really a string not an integer?
- format: Which formats exists, what values do they have, really a string not an integer?

Updated: https://app.swaggerhub.com/apis/hacki11/BSB-LAN/2.0#/general/getConfiguration
Thanks

dukess, 2021-01-09:
> Shouldn't it be the same data type?

Of course, all mentioned fields must be numbers. Moreover i think to rename "id" to "parameter" while a little time has passed and this module is not used everywhere.

These two fields allow to build user interface:

> type: Which types exists, what values do they have, really a string not an integer?

0 -  CDT_VOID, // byte array (for internal using. Never sent through JSON)
1 -  CDT_BYTE, //unsigned byte. can be various format (text, switch, dropdown)
2 -  CDT_UINT16, //CPI_TEXT field with format for unsigned integer
3 -  CDT_UINT32, //CPI_TEXT field with format for unsigned long integer
4 -  CDT_STRING, //CPI_TEXT field
5 -  CDT_MAC, //CPI_TEXT field with format for MAC input/output
6 -  CDT_IPV4, //CPI_TEXT field with format for IPv4 input/output
7 -  CDT_PROGNRLIST, //CPI_TEXT field with format for programs list input/output
8 -  CDT_MAXDEVICELIST,//CPI_TEXT field with format for MAX! devices list input/output
9 -  CDT_DHTBUS //CPI_TEXT field with format for DHT bus

> format: Which formats exists, what values do they have, really a string not an integer?

Using for select HTML input type. 0 - none 1 - text field, 2 - switch, 3 - dropdown

konne, 2021-01-09:
> Of course, all mentioned fields must be numbers. Moreover i think to rename "id" to "parameter" while a little time has passed and this module is not used everywhere.
@dukess: that would be great. The API is new and it make sense to make the changes now. I also agree with @hacki11 that consistent naming and datatypes are important. So I like your idea to rename id -> parameter

hacki11, 2021-01-09:
I also recognised many hashmaps as responses. It is more common to use arrays and place the key as field inside the object. 
But openapi supports both, no real issue here

konne, 2021-01-09:
@dukess  please can you here use also names instead of ids for the type.
If you ever has to change the "C" code the api is independent of that changes.
so type: "VOID" or type: "BYTE" ,....

thx

hacki11, 2021-01-09:
That would be great, I then can automatically generate the enum type

dukess, 2021-01-09:
@hacki11 
>I also recognised many hashmaps as responses. It is more common to use arrays and place the key as field inside the object.

Parser will be more complex, so i leave it for client side.

@konne To be honest i do not want do it right now because every text string consume space. It can be critical for Mega users. So i think do stable release first and then made improvements.

hacki11, 2021-01-09:
I forgot one thing to ask. In /JW Response, the status field:
1 = OK
2 = NOT_FOUND
Is that correct?

dukess, 2021-01-09:
Correct!

dukess, 2021-01-12:
@konne @hacki11 Note: /JK and /JC command can return HTTP code 304 "Not Modified" without payload if ETag will be same. As ETag BSB-LAN version is used.

hacki11, 2021-01-12:
Is the E-Tag feature available with plain /JC /JK or also if the user gives parameters?

Then we would have 3 different responses with the same E-Tag, which will cause issues:
/JC
/JC=700
/JC=700,710
How is this implemented?

dukess, 2021-01-12:
user parameters allowed.
BSB-LAN return same ETag for different queries.
By fact /JK=x /JC=x commands return static content which can be changed only after software upgrading. I.e ETag describe actual version.

hacki11, 2021-01-12:
That means, the client has to determine if the content will be different or not (e.g. change parameters) prior the request? Imho that could have been solved by ETag taking the parameter keys into account.

dukess, 2021-01-12:
Form Wiki:
In typical usage, when a URL is retrieved, the Web server will return the 
resource's current representation along with its corresponding ETag value, 
which is placed in an HTTP response header "ETag" field:

ETag: "686897696a7c876b7e"

The client may then decide to cache the representation, along with its 
ETag. Later, if the client wants to retrieve the same URL resource again, 
it will first determine whether the locally cached version of the URL has 
expired (through the Cache-Control and the Expire headers). If the URL has 
not expired, it will retrieve the locally cached resource. If it is 
determined that the URL has expired (is stale), the client will send a 
request to the server that includes its previously-saved copy of the ETag 
in the "If-None-Match" field.[2]

If-None-Match: "686897696a7c876b7e"

On this subsequent request, the server may now compare the client's ETag 
with the ETag for the current version of the resource. If the ETag values 
match, meaning that the resource has not changed, the server may send back 
a very short response with a HTTP 304 Not Modified status. The 304 status 
tells the client that its cached version is still good and that it should 
use that.

However, if the ETag values do not match, meaning the resource has likely 
changed, a full response including the resource's content is returned, just 
as if ETags were not being used. In this case, the client may decide to 
replace its previously cached version with the newly returned 
representation of the resource and the new ETag.

hacki11, 2021-01-12:
Thanks for the explanation, backed with caching makes sense for queries having everything in the request uri.

I assume the POST /JC call with the request body does not support etags, right? Because the cache does only use the request uri which would be the same for different requests.

dukess, 2021-01-12:
I made fix for /JC & POST: ETag will be disabled for this case.

fredlcore, 2021-02-08:
Can we consider this closed?

hacki11, 2021-02-08:
From my side I'm fine!

closed 2021-02-15 by fredlcore