fredlcore/BSB-LAN #269: Web strange behavior / missing text

rroblik, 2021-01-07:
Hi @fredlcore,

I'm now connected successfully with my Atlantic Heater (RVS) 👍 

But I've some strange issues : 

1. The Web interface show my only "placeholder text" like that 
[image]

2. I'm not able to set value using web even if in conf file I've `#define DEFAULT_FLAG 0` :
 `http://192.168.1.203/710`
[image]
`http://192.168.1.203/S710=21`
[image]
 

Side notes : 
* I'm using [@dukess ajax alternative webserver](https://github.com/dukess/bsb_lan_ajax) configured as requested ; but the default webserver is displayed I think. My SD card have all required files.

Thanks

dukess, 2021-01-07:
The problem is that you can only use one language at a time.
Strings in "C" used as text resource identifiers for later substitute with real language.
By the way you can download latest ajax.html. It already support Web configuration.

rroblik, 2021-01-07:
Hu @dukess,

Ok yes; but ajax.html (renamed to index.html) is already here on the SD card

dukess, 2021-01-07:
If you get it from release i recommend to you download fresh version.

dukess, 2021-01-07:
Update: ...and latest language files from this project: when ajax.html can't find needed text string it display text ID.

rroblik, 2021-01-07:
I get it from your master branch ; is that ok ?

dukess, 2021-01-07:
Yes, sure

fredlcore, 2021-01-08:
Since this is a matter of @dukess' alternative webserver, I can't say anything about it, really, I'm sorry...
I would close this for now, but that should still make it possible for your two to continue the discussion if necessary.

closed 2021-01-08 by fredlcore

rroblik, 2021-01-08:
Yes no problem. Doing without it for now 👍