fredlcore/BSB-LAN #270: Localization / translation best practices

rroblik, 2021-01-08:
Hi @fredlcore / @1coderookie 

I've read a lot of issues / documentation about the internationalization but I don't find a *KISS* way of helping you in translation work.

I'm French native, and English speaker ; I've configured my BSB LAN to use `EN` language -- as it's generally the most translated language that I can understand -- but it see a lot of missing translation, falling back into German into my English 😸.
[image]

I would like to help you into a 100% EN / FR coverage ; could you explain me how can I help you ? I see language related string into 3 files (`.h`, `.json`, `.js`) ; but I hope we only have to update one to let CI or something else build them all for us ?

Furthermore, I think a _How to contribute_ section in README file would help this great project (or maybe a `CONTRIBUTE` file)

Thanks

fredlcore, 2021-01-08:
Thanks for your help offer, highly appreciated, and it is actually quite simple indeed:
You just open the destination language (say, LANG_EN.h) on one side of your screen and LANG_DE.h (which is the master, because that's where most of the documentation was/is available) on the other. Then you just copy the #define lines over to the destination language and translate the term. All other files (.json, .js etc.) are created automatically when a new version of the file is compiled.

We tried our best to avoid doublettes, so even if the term "Reduziert" (reduced) appears several times in the software, there is (or should) only be one line in the language file. If you spot doublettes in the German LANG_DE.h, please let us know.

fredlcore, 2021-01-08:
One more note on the translation of technical terms: Please do not translate technical terms by "guessing" their meaning. It's best to look up the exact term on your heating system or the documentation that came with it. If you just use Google translate, it may come up with all different kind of things, but not the term that is actually technically accurate. In that case it's best to leave it as it is until somebody finds an authoritative translation.

rroblik, 2021-01-09:
Ok will try it! Will start in French 😃

dukess, 2021-01-09:
@rroblik You can try to find and download manuals like this https://kupdf.net/download/service-luna-platinum-ed-8_5c51bc1ae2b6f5646c8fedfb_pdf for your boiler.

rroblik, 2021-01-09:
Yes I have thanks!

1coderookie, 2021-01-09:
> One more note on the translation of technical terms: Please do not translate technical terms by "guessing" their meaning. 

Exactly, that's a very important note here! Especially because there are many abbreviations, more in English then in German right now. So _please_ only translate those parameters to french, which are either really 'simple' (like outside temperature) or which you found in controller manuals in french language. 
I'll review the latest eliminations of abbreviations within the german language file soon (I spotted some mistakes there) and probably start eliminating the abbreviations within the english version also.

Btw @rroblik: I think I/we didn't get a feedback from you yet which heating system exactly you are using an what /Q puts out. Can you please send me an email with that infos? Thanks.

1coderookie, 2021-01-09:
@rroblik: I'll try to work on the LANG_EN-file soon to 
a) add missing elements like "Beginn Tag/Monat" and 
b) eliminate the existing abbreviations (so that the parameters&functions will appear in full text). 
But I'll need some time for that. 
So I'd suggest that you either wait until I'm done or just start with the existing file for the translation for the french version and add the rest later once I'm done with the english language file. 

If you want to add/get names etc in french language from/based on french manuals (what's imho definitely the best solution to avoid any translation-problems): I just found a french manual within my documentation-'library' which I could send to you if you're interested in that..

rroblik, 2021-01-09:
Ok for French I’ve the manual ;)

1coderookie, 2021-01-09:
I don't know which system/controller you are using and which manual you have.. I just can tell you that the specific manuals for the diffenrent controllers contain also the specific parameters&descriptions. E.g.: heatp-pump-specific parameters (RVS61/21 controller) are not listed/described within a manual of an oil-fired burner (RVS63/43) or an gas-fired system (LMS14/15).
The manual I can offer you is one for RVS63/43-type controllers.

rroblik, 2021-01-09:
Let me try first ;)

fredlcore, 2021-02-08:
Any progress here, @rroblik?

closed 2021-02-15 by fredlcore