fredlcore/BSB-LAN #272: MQTT and broadcast messages

rroblik, 2021-01-09:
Hello

From what I read, mqtt eligible parameters must be explicitly defined in settings, so as the push frequency. 

I would like to know if it’s possible to update this behavior to (also?) send any broadcasted values ? (INF, if I’m right...)

To clarify things, I would imagine also differents subtopics :
* ˋBSB-LAN/p` For pulled messages (actually `json`)
* `BSB-LAN/b` For broadcast messages

What do you think about it @fredlcore ? Don’t know if it’s technically realist !

thanks

fredlcore, 2021-01-09:
What broadcast messages do you want to evaluate that cannot be polled? The problem with broadcasts is that we don't know if BSB-LAN will be listening right in that moment for an incoming telegram and not be doing something else at that moment. Therefore it's always more reliable to query that specific telegram you are interested in. I can't think of a value that would require polling more often than once every minute. If you display it in a graph, the finer granularity will get lost anyway.

closed 2021-01-09 by fredlcore

rroblik, 2021-01-09:
My "need" is to update the changing value only when they change ; but in fact I don't know if BSBS protocol do it in the heater (change detection). I didn't know that BSB was a blocking protocol ; ok let pull pull pull !

fredlcore, 2021-01-09:
That's what the home automation software should take care of. In FHEM, I can set for each reading whether the database where the retrieved values are stored in should be updated at every poll or at every changed value.
Also, INFs are not sent every time any parameter changes, but only for few selected parameters such as the ones that result in changed icons in the display.