fredlcore/BSB-LAN #273: Due cold startup issue workaround

rroblik, 2021-01-09:
Hello @fredlcore,

I noticed I need to manually push on reset button on my Due to let the BSB LAN working after a cold startup / power outage / sketch transfer.

I Google it and I read that is a know "design bug" of (some of) Due (maybe clone, maybe not).

There is hardware workarounds around the web but I also [read a dead simple software workaround](https://forum.arduino.cc/index.php?topic=256771.msg4643749#msg4643749). I'm not sure that I can add the `delay(2000)` at the beginning of `setup()` for BSB LAN ... in `BSB_lanustom_setup.h` it may be to late (?).

Any feedback about this ?

Thanks !

dukess, 2021-01-09:
You need solder resistor.
https://forum.arduino.cc/index.php?topic=256771.msg2512504#msg2512504

fredlcore, 2021-01-09:
I haven't had this issue yet, but feel free to try it out in your setup (just add it to wherever you think it should go in the BSB_lan.ino). If it works reliably, I can add it as a general option, but we already have a 3 second delay before network is being used, so I'm not sure if this really changes things.

closed 2021-01-09 by fredlcore

rroblik, 2021-01-09:
FYI : delay workaround not working :(

@dukess : I've not tools to solder this kind of resistor ; will see how to do because in case of power failure my bsb lastay offline for ever (until manual reset) ; so bad !

hacki11, 2021-01-13:
I want to add that there is also a software workaround.
The atmega16u2 on the due needs to be reflashed with a new firmware.
An additional Arduino is needed, here are the instructions
https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoISP
https://petervanhoyweghen.wordpress.com/2013/02/25/modifying-the-atmega16u2-firmware-on-the-due/

Firmware:
https://forum.arduino.cc/index.php?action=dlattach;topic=167492.0;attach=52901
https://forum.arduino.cc/index.php?topic=167492.msg1373055#msg1373055

working for me

1coderookie, 2021-01-13:
Thanks for that info! I added the link to the capacitor solution in the manual, but I'll also add your mentioned solution.