fredlcore/BSB-LAN #275: Settings not saved

rroblik, 2021-01-10:
Hello @fredlcore 

I read a read the doc again but I've an issue / misunderstanding about configuration saving.
I've now fixed the "cold start bug" of my due using a 10K resistor (#273) 😎 

I simulate power outage by disconnecting card power supply ; but when I reconnect power supply I can see on settings page (`/CI`) that settings are not saved ; value seems to be retaken taken from `BSB_lan_config.h`. If I edit settings (using web) they are stored without error ; but lost on next power outage.

In `BSB_lan_config.h` I've (related to EEPROM / config settings only)
```c
byte UseEEPROM = 1;
// "External" web server. Read files from SD-card. Only static content: html, js, css, jpg, etc.
#define WEBSERVER
// Configuration will be store in EEPROM
#define CONFIG_IN_EEPROM
// Compile web-based configuration and EEPROM config store module extension.
#define WEBCONFIG
// Compile JSON-based configuration and EEPROM config store module extension.
#define JSONCONFIG
```

Documentation is not clear about that, to my mind.

Thanks for your help, again !

fredlcore, 2021-01-10:
`UseEEPROM = 1` is not enough. You have to set "Read config from EEPROM" in the web configuration to "on" and also save these settings at the bottom of the page. Have you done all that?

rroblik, 2021-01-10:
Hmmm yes ; here is my configuration

[image]

In fact I just moved the card next to my heater (so powerd off) ; and I just reconnected ; all look ok... 😮

To be sure : I when I read `Free space: 0 MB` that mean I'm on EEPROM ?

fredlcore, 2021-01-10:
No, that's the SD card.
So it seems it's working now, because all the settings are there?

1coderookie, 2021-01-10:
> Documentation is not clear about that, to my mind.

> `UseEEPROM = 1` is not enough. You have to set "Read config from EEPROM" in the web configuration to "on" and also save these settings at the bottom of the page. 

It's true that this fact isn't mentioned in chapter 5.2 due to the fact that that's not part of the config.h. UseEEPROM is (iirc) set to 1 per default though, but when I write the chapter 5.1 about the webconfig, then I'll also mention that point of course.

fredlcore, 2021-02-08:
@rroblik: As you mentioned that it's working ok for you now, can we consider this closed?

rroblik, 2021-02-08:
Yes !

closed 2021-02-08 by fredlcore