fredlcore/BSB-LAN #276: PlatformIO CI errors empty

fredlcore, 2021-01-10:
@fl42 and @dukess: Since a few days I get an error report with every push I make to the repo:
https://github.com/fredlcore/bsb_lan/runs/1677593942?check_suite_focus=true
To me, this is not helpful at all if there is no error message telling me where to look for the error. 
It seems to have started since I added a new (but not required) definement to the configuration file. I have copied the new config to the tests directory and named it accordingly, but the error remains.

Could you two tell me what the workflow should be so that I can make changes not only in the main default configuration file but also in the other files? Or should we do away with this CI because a general error is not very helpful, especially when the actual version compiles fine nevertheless?

fl42, 2021-01-10:
PR #277 
Please note the CI fails in the PR but that looks not related to the PR :)

fredlcore, 2021-01-10:
Thanks a lot both for the explanation as well as the PR, @fl42!
I am logged into the account, but all I get to see is this:
[image]
With the documentation you have linked, I found the raw logs and that's fine for me, I just had hoped that in the screen shown above would already be the actual error message. Because a plain "Error: " is a bit misleading ;).
Anyway, thanks for that!

closed 2021-01-10 by fredlcore