fredlcore/BSB-LAN #282: Naming within the Webconfig-Interface

1coderookie, 2021-01-13:
Hi @fredlcore and @dukess,
I'm working on the translations and the general naming of the functions within the webconfig right now (LANG_DE). Before I'll make a PR, I'd like to ask you to have a look at it and I'd like to mention/suggest some other stuff:

1.)
German translations/names for LANG_DE-entries:
```
//WEBCONFIG
#define CF_USEEEPROM_TEXT "Konfiguration aus EEPROM lesen"
#define CF_BUSTYPE_TEXT "Bustyp"
#define CF_OWN_ADDR_TEXT "BSB/LPB: Eigene Adresse"
#define CF_DEST_ADDR_TEXT "BSB/LPB: Zieladresse"
#define CF_PPS_WRITE_TEXT "PPS: Schreiben aktiviert"
#define CF_LOGTELEGRAM_TEXT "Bustelegramme loggen"
#define CF_LOGCURRVALUES_TEXT "Parameterwerte loggen"
#define CF_LOGCURRINTERVAL_TEXT "Logintervall (Sekunden)"
#define CF_AVERAGESLIST_TEXT "24h-Durchschnittswerte: Parameter"
#define CF_CURRVALUESLIST_TEXT "Logging: Parameter"
#define CF_MAX_DEVICES_TEXT "MAX!-Geräte"
#define CF_DHCP_TEXT "DHCP verwenden"
#define CF_IPADDRESS_TEXT "IP-Adresse (manuell)"
#define CF_TRUSTEDIPADDRESS_TEXT "Vertrauenswürdige IP-Adresse"
#define CF_MASK_TEXT "Subnetzmaske"
#define CF_GATEWAY_TEXT "Gateway"
#define CF_DNS_TEXT "DNS-Server"
#define CF_WWWPORT_TEXT "TCP/UDP-Port"
#define CF_WIFI_SSID_TEXT "WLAN-SSID"
#define CF_WEBSERVER_TEXT "Webserver SD-Karte"
#define CF_PASSKEY_TEXT "URL-Passkey"
#define CF_BASICAUTH_TEXT "HTTP-Authentifizierung"
#define CF_ONEWIREBUS_TEXT "OneWire: Pins"
#define CF_ONEWIREBUS_DEVICES_TEXT "OneWire: Geräte"
#define CF_DHTBUS_TEXT "DHT: Pins"
#define CF_IPWE_TEXT "IPWE verwenden"
#define CF_IPWEVALUESLIST_TEXT "IPWE: Parameter"
#define CF_MAX_TEXT "MAX! verwenden"
#define CF_MAX_IPADDRESS_TEXT "CUNO/CUNX/modifizierter MAX!Cube: IP-Adresse"
#define CF_WRITEMODE_TEXT "Schreibzugriff erlaubt"
#define CF_DEBUG_TEXT "Debugging verwenden"
#define CF_MQTT_TEXT "MQTT verwenden"
#define CF_MQTT_IPADDRESS_TEXT "MQTT: IP-Adresse"
#define CF_MQTT_USERNAME_TEXT "MQTT: Username"
#define CF_MQTT_PASSWORD_TEXT "MQTT: Passwort"
#define CF_MQTT_TOPIC_TEXT "MQTT: Topic prefix"
```
As you can see, I tried to keep the names as short as possible and clear at the same time. I also changed the naming a bit within the functions where we can choose between activated/deactivated, because in the field of the pulldown we have (e.g.) 'on/Ein' and 'off/Aus', so that it fits better to the meaning. 
If you are alright with the naming, I can make a PR.
But I have some more stuff:

2.)
We have two rows for the 24h-average calculation right now:
[image]
The first row says 'Calculation of 24h-average values for the following parameters' and comes from somewhere else within the language file. But that name is not correct there as it is the label for the parameters which should be calculated and not if the function itself is activated or not. So that name would fit better in the row underneath. But I think it's too long for there anyway, so I'd suggest the following solution:
24h-Durchschnittswerte berechnen (EN: "Calculate 24h-averages")
24h-Durchschnittswerte: Parameter (EN: "24h-averages: parameters")
That means that the existing entry has to be thrown out, where we are coming to the next point:

3.)
I know it saves memory if we are using the existing names/description which are already existing. But: 
a) Right now we have a mixture of English and German -> that's not nice.
b) The existing names doesn't always fit really good (as mentioned above; also affects last entry "Neuere Version gefunden" which should be "Check for update"/"Auf Update überprüfen" to fit the meaning of the enabled/disabled function itself).
c) That only affects a few entries (about 8 or so). I think it wouldn't cost a lot more memory if we just would add own entries for them for the //WEBCONFIG-area but it would definitely make it nicer.
Due to the shortening of the existing names we are saving some space now anyway, so maybe at the end it wouldn't make such a big difference..

4.)
There are no own entries / there is no own "//WEBCONFIG"-area within the LANG_EN right now I think, so it would be good/necessary to have it. I'm not sure if it's enough if I would just add it at the same place within the file as where it is within the LANG_DE, so could you guys please take care about it?

5.)
Suggestion: Let's get rid of the categories (left column) within the table. 
Reason: Right now there are three categories: General, IP settings, MQTT settings. 
In my opinion that's not the best solution, because on one hand it's not a precised subdivision (e.g.: 'MAX!-stuff' is within 'General', but it's not 'general' up to me; 'Checking for newer versions' is within 'MQTT settings' but that has nothing to do with MQTT) and on the other hand it's not even really necessary because a clear naming should be sufficient anyway. 
So my point here is: We should either get rid of the categories there and go with a clear naming instead (what also would save us some memory again I think) or make a more precise subdivision of the category names. I'd prefer getting rid of it. 
    
Once we found a solution, I can start with the description of the webconfig in chapter 5.1 of the manual, cuz the function itself won't get any big changes anymore anyway I think. Small changes afterwards aren't a problem of course, but before I start with the description, I'd like to get a 'clear' base about the names etc to start with..

1coderookie, 2021-01-13:
Oh, and one thing that I forgot:
We have the line "Free space: xxMB" right above the top of the webconfig interface, which refers to the space of the SD card. I'd suggest that we add "SD" or "SD card" to the name, so that it's more clear  which space exactly is meant: "Free space SD card: xxMB".

dukess, 2021-01-13:
1. I approve (my opinion for German language can be ignored because i'm _kann nicht fließend Deutsch sprechen_)
2. MENU_TEXT_AVT is used for displaying program list in /C command when WEBCONFIG disabled. So we can change text to without big regret. I will make little change for /C command for displaying your variant.
3. If it is not possible to change the text so that it fits both webconfig and the existing interface, then, of course, we need to create new localization strings. List of SUSPICIOUS strings: MENU_TEXT_NVA, MENU_TEXT_MMD, MENU_TEXT_VBL, MENU_TEXT_MAC, MENU_TEXT_AVT. All other strings are unique.
4. I can copy piece of text from _DE to _EN but it still contain same error.
5. Maybe in the future, when there are a lot of parameters, we will need categories, now, I agree, we can disable them.

fredlcore, 2021-01-13:
Thanks, @1coderookie, that's a great start - I agree with all the points, except maybe with 5 - of course the current categories are not enough, but if we have more (and more fitting ones), people will know with one look whether this or that is relevant for them. Also, in the end it might save memory if we have a category "Bus" and then just write "Destination address", because then it's clear that it's meant for whatever bus is active.

As with no. 3, I think the translation just hasn't been done yet, that's why there is a mix of English and German. Of course as much as possible should be in the same language.

As for the //WEBCONFIG area, you can just add that, it's just a comment. It doesn't matter which way the defines are listed in the file, but grouping them makes it of course easier to find them later.

1coderookie, 2021-01-13:
> ..because i'm _kann nicht fließend Deutsch sprechen_)

Duuude - that's great! Much more than I can say in Russian - I only know three important expressions: njet, da & nastrovje ;)

About 5:
Ok, so we stick with the categories, fine with me. Then I could start adding other categories (like 'Bus', 'Sensoren' etc for the German version and the LANG_EN accordingly):
```
#define CAT_BUS_TEXT "Bus"
#define CAT_SENSORS_TEXT "Sensoren"
#define CAT_IPWE "IPWE"
```
but where&how do I tell the program which name (of the function from the other column) belongs to which category?

For //WEBCONFIG in LANG_EN I can also try it, will test it locally then. 

About 2:
> MENU_TEXT_AVT is used for displaying program list in /C command when WEBCONFIG disabled. So we can change text to without big regret. I will make little change for /C command for displaying your variant.

But then it wouldn't fit in that output really well anymore, I'd prefer to make the clear distinction between that output and the one from the webconfig.

About 3:
> If it is not possible to change the text so that it fits both webconfig and the existing interface, then, of course, we need to create new localization strings. List of SUSPICIOUS strings: MENU_TEXT_NVA, MENU_TEXT_MMD, MENU_TEXT_VBL, MENU_TEXT_MAC, MENU_TEXT_AVT. All other strings are unique.

I'd definitely prefer that solution. But I have no idea how to do that with the localization strings, so I'd ask you guys to take care about that ;)
But I can start with the other stuff and then we/you can take care of that afterwards..

dukess, 2021-01-13:
> Duuude - that's great! Much more than I can say in Russian

You're exaggerating, I had to use a translator. :-) Although I studied German long time ago, it turned out that I had no place to apply it and I gradually forgot the grammar. I can still remember some words, but I won't be able to communicate.

> I only know three important expressions: njet, da & nastrovje ;)

At least it is good base to start communicate. Which language will be used - not important :-)

> #define CAT_SENSORS_TEXT "Sensoren"

Moment, i do it.

> but where&how do I tell the program which name (of the function from the other column) belongs to which category?

Definitions for web confg placed in BSB_lan_EEPROMconfig.h

> I'd definitely prefer that solution. But I have no idea how to do that with the localization strings, so I'd ask you guys to take care about that ;)
But I can start with the other stuff and then we/you can take care of that afterwards..

Can you propose translations for replacing? Now it is
#define MENU_TEXT_NVA "Neuere Version gefunden"
#define MENU_TEXT_MMD "Monitor Modus"
#define MENU_TEXT_VBL "Verbositäts-Level"
#define MENU_TEXT_AVT "Berechnung von 24h-Mittelwerten für die folgenden Parameter"

MENU_TEXT_MAC can be leaved without changes i think because it mean "MAC address".

I'll add the new definitions myself.

1coderookie, 2021-01-13:
> You're exaggerating, I had to use a translator. :-) 

lol - ok, that doesn't count then ;)

> Although I studied German long time ago, it turned out that I had no place to apply it and I gradually forgot the grammar. I can still remember some words, but I won't be able to communicate.

Yeah I know that effect, it's a pity, but I'm sure it's still hidden _somewhere_ in our brains though ;) 

> > I only know three important expressions: njet, da & nastrovje ;)
> 
> At least it is good base to start communicate. Which language will be used - not important :-)

Hehe, I agree :)

> Can you propose translations for replacing? Now it is
> #define MENU_TEXT_NVA "Neuere Version gefunden"

Well that's what I meant with creating a new entry: The existing fits in the meaning and the content where it's displayed now at the first page of BSB-LAN-Web if a newer version was found, but it doesn't fit in the context where it's displayed within the webconfig.
For the webconfig it should be something like 'Check for updates' and 'Auf Updates überprüfen' imo, cuz it's a function where we have on and off as possible settings.  

> #define MENU_TEXT_MMD "Monitor Modus"

Already fits in German, in English should be 'monitor mode' (didn't check the english webconfig though).

> #define MENU_TEXT_VBL "Verbositäts-Level"

Should be changed to 'Verbositätsmodus' / 'verbosity mode' anyway I think, cuz we don't have different levels, just on or off.

> #define MENU_TEXT_AVT "Berechnung von 24h-Mittelwerten für die folgenden Parameter"

Fits in the context of the (alternative) output, but not within the webconfig. 
For webconfig: Depends if we add a category "24h-Durchschnittswertberechnung" / "24h average calculation". If we would do so, we could use (already existing?) expressions 'aktiviert' / 'enabled' and 'Parameter'/'parameters' for the second column.  
'Aktiviert'/'enabled' because of the reason that it's a function where we have on and off as possible settings, so that will fit better imo.

> MENU_TEXT_MAC can be leaved without changes i think because it mean "MAC address".

Yes, but it's English though.. In German: 'MAC-Adresse'. 

> I'll add the new definitions myself.

Great, thanks!

dukess, 2021-01-14:
Done but for German language only. (For avoiding interferring edits in English)

1coderookie, 2021-01-14:
Great, thanks! Will have a look at it after merge :)

1coderookie, 2021-01-14:
What do you guys think: how about adding another field "Multicast DNS" (category Netzwerk/Network) where one can see/edit the mDNS name?

fredlcore, 2021-01-14:
Hm, most people will only have one instance of BSB-LAN running, I guess, and setting a default name in a define statement makes it possible to change it, but people will have to know what they are doing. So that way it's easier to use that hostname in examples etc.

dukess, 2021-01-14:
...or do name generated on the fly, ex: bsb-lan-0-42 for standart BSB, bsb-lan-1-06 for RGT1 on LPB, bsb-lan-2-xx for PPS ("bsb-lan" + bus_mode + "-" + own_address in hex)

1coderookie, 2021-01-14:
RGT1 (and/or RGT2 = 7) will/has to be still connected to BSB like the original room units I think, doesn't work as an RGT at LPB.
But we don't _have_ to add mDNS field in webconfig - just thought that (long term speaking) we wanted to change pretty much completely to webconfig w/o the need of touching the config.h file - therefore it would have been needed.

fredlcore, 2021-01-14:
Yes, that's true, I wouldn't veto against it, I was just thinking if it would making references in the manual easier if we no longer have to write `http://<ip-of-adapter>`. So, @1coderookie: your call!

1coderookie, 2021-01-14:
Ah ok, didn't get it that way. Honestly speaking: I won't change _all_ of the `http://<ip-address>` entries within the manual anyway, too much of a hustle ;) 
So then I'd vote for making that mDNS-naming also available within the webconfig. It's activated per default anyway already, and if someone (like me for example) uses more than one setup, it could make things easier to distinguish between the setups and see the whole config with one glance at the webconfig page..

fredlcore, 2021-01-14:
I just realize that it's not activated by default because it's not yet working with WIFI. That's why I reconsider now and would stick with the definement. Otherwise we would have to disable the setting in the webconfig to avoid confusion...

dukess, 2021-01-14:
Need more RAM for Mega!
We can try change
```
const char *averagesFileName = "averages.txt";
const char *datalogFileName = "datalog.txt";
const char *journalFileName = "journal.txt";
```
to
```
const char averagesFileName[] PROGMEM = "averages.txt";
const char datalogFileName[] PROGMEM = "datalog.txt";
const char journalFileName[] PROGMEM = "journal.txt";
```
and free 42 bytes but I'm not sure if this will work correctly.

fredlcore, 2021-01-14:
Ok, but what is the relation between the filenames and the mDNS setting?

dukess, 2021-01-14:
RAM consumption only :-)

We have reached time when adding an option to the configuration can make it impossible to run the program on Mega due to lack of RAM

fredlcore, 2021-01-14:
Ok, good to know... Combined with the fact that it's not (yet) usable with WiFi, I would vote for leaving it in the config.

1coderookie, 2021-01-14:
Ok, fine with me ;)

But just to understand it right:
```
#ifndef WIFI                            // mDNS currently does not work with the WiFiSpi library
#define MDNS_HOSTNAME "BSB-LAN"         // Advertises the hostname in the local network. Disable this if you don't want your device to be found under this name in your network.
#endif
```
imho it _is_ activated per default in case WiFi isn't activated!? At least I didn't activate mDNS last time when I flashed that version to my LAN-setup and it worked with the 'bsb-lan' within the URL..

fredlcore, 2021-01-14:
Yes, it is activated if WIFI is NOT defined (if*n*def). But those who use WiFi won't (yet) be able to use it, therefore it's a bit misleading to have it in WebConfig.

1coderookie, 2021-01-14:
Ok, then I'm glad I got it right ;)
We could let it just appear if WiFi isn't activated, but let's keep this for the future (when mDNS will work with WiFi anyway I hope), so that we'll still have something to do later :)

1coderookie, 2021-01-14:
> We have reached time when adding an option to the configuration can make it impossible to run the program on Mega due to lack of RAM

Btw: Are you using the v2&Mega-Setup on a regular basis as your one and only setup or are you already using v3&Due and checking the Mega-compatibility everytime any changes occur also..?

dukess, 2021-01-14:
Due is preferred for me but Mega tested and still work even current version.

1coderookie, 2021-01-14:
So you are using v3&Due but you are also testing the Mega-compatibility every time? Great, thanks for that!

dukess, 2021-01-14:
Not every time but often

1coderookie, 2021-01-19:
@dukess: I'd like to change the name of the button at the end of the webconfig from "Datei absenden" to "Einstellungen speichern" / "Save settings", but I don't find the text of that one within LANG_DE / LANG_EN. Where can I find them and would it be ok to change that name or is that field used somewhere else also?

dukess, 2021-01-19:
"Daten absenden" - standart naming for "Submit" button. For changing here is only one way: add "value" field to HTML code.

fredlcore, 2021-01-19:
I agree with @1coderookie that the naming in German is not the best way to phrase it, but if we add a value to the HTML field now, we would have to translate it to all languages, whereas if we stick to the standard, the default translation will still be understood by most.

dukess, 2021-01-20:
Can we use #define STR6204_TEXT "Parameter speichern" as button name?

fredlcore, 2021-01-20:
Great idea!

closed 2021-01-31 by 1coderookie