fredlcore/BSB-LAN #283: Consensus for naming & abbreviations within language files

1coderookie, 2021-01-13:
I'd like to find a little consensus about the naming of the parameters and probable abbreviations within the language file(s). 
We often have terms like temp/temperature, so what do you think - wouldn't it be sufficient if we use continuous abbreviations for 'usual' expressions like that? 
So: 
Temperatur = Temp ; temperature = temp (I don't believe that people would mess it up with 'temporary' within that context)
Außentemperatur = AT
Heizkreis = HK, heat circuit = HC 
etc.?
Many times it's already existing, but when I made some changes within LANG_DE, I also changed to fulltext many times which maybe (in these pretty much obvious cases) isn't needed. Could save us some memory..

dukess, 2021-01-13:
The main problem that was solved with the help of abbreviations is the insufficient screen size on room units. When translating into Russian, I try to remove abbreviations with the exception of well-known ones, such as TWW, HC (but in Russian HC looks equally for "Heating Circuit" and "Cooling Circuit". So decision will be different for different languages.

fredlcore, 2021-01-13:
I wouldn't veto against the abbreviations, but for me as a novice user back in the day, I was quite puzzled what "HK1" and "AT Lieferant" would mean. Yes, it's somewhere in the manual, but at best at the first time it's mentioned, and afterwards, it's often also just referred to as "AT". So I think for newbies, the full expression is easier to handle.

1coderookie, 2021-01-13:
Ok, so we'll go with _full names_ wherever it's possible (and clear from the manual about specific terms from existing abbreviations).
Will make it also easier for people who wants to translate from English to other languages.

closed 2021-01-14 by 1coderookie