fredlcore/BSB-LAN #288: HTTP auth not working right?

1coderookie, 2021-01-15:
Based on a recent error report in the FHEM forum about a problem with the HTTP authentification, I just did a test of that function. And: indeed, it really seems that it's not working right. 
What I did:
- disabled "Read config from EEPROM" within webconfig first
- activated the example (base64 encoded atari:800xl), deactivated the definement with the 'plain' value
- flashed and queried the URL.
The field where the username and password has to be entered appears, but when I enter "atari" and "800xl" and commit it, the field just appears again. 
So either I am/we are doing something wrong here or there is a bug. Could you please check that @fredlcore and @dukess? 
Thanks!

fredlcore, 2021-01-15:
Correct, the basic auth is not working, I can't even find the right spot to identify where it fails. Could you have a look at it, @dukess ?

fredlcore, 2021-01-15:
It looks to me that in line 6765++ it is only checked if the variable USER_BASS_B64 is set, but it is not actually compared to the credentials sent by the browser, so it will always fail?

dukess, 2021-01-15:
Can you drop here HTTP headers of query and answer? (Ctrl+Shift+A in Mozilla or Chrome, then Network)

user/password comparsion will be success when
"Authorization: Basic" header was not previounsly found AND
length of USER_PASS_B64 > 0 AND
"Authorization: Basic" was found in header AND
USER_PASS_B64 was found in header

```
              if (!(httpflags & HTTP_AUTH) && USER_PASS_B64[0] && strstr_P(outBuf + buffershift,PSTR("Authorization: Basic"))!=0 && strstr(outBuf + buffershift,USER_PASS_B64)!=0) {
                httpflags |= HTTP_AUTH;
```
After headers parsing we check conditions.
If length of USER_PASS_B64 > 0 AND
HTTP_AUTH was NOT found then HTTP Auth fail

`if (USER_PASS_B64[0] && !(httpflags & HTTP_AUTH))`

dukess, 2021-01-15:
Fixed.

fredlcore, 2021-01-15:
Ok, the strange thing is this: It does not work as it is, but if I add two Serial.printlns, it works:
```
             else
              //Execute only if flag not set because strstr more expensive than bitwise operation
Serial.println(outBuf);
Serial.println(outBuf+buffershift);
              if (!(httpflags & HTTP_AUTH) && USER_PASS_B64[0] && strstr_P(outBuf + buffershift,PSTR("Authorization: Basic"))!=0 && strstr(outBuf + buffershift,USER_PASS_B64)!=0) {
                httpflags |= HTTP_AUTH;
              }
              memset(outBuf + buffershift,0, charcount);
              charcount=buffershift; //Reserve space in buffer for ETag
            } else if (c != '\r') {
              // you've gotten a character on the current line
              currentLineIsBlank = false;
            }
```
Do you have an idea why that changes things?

dukess, 2021-01-15:
https://github.com/fredlcore/bsb_lan/pull/286/commits/6f2576c7c79bf2c0b1a1a9139e253db7cf417ce9

fredlcore, 2021-01-15:
Great, thanks a lot! Still strange that it worked with the added printlns for me, but now it also works, so that's that :)!

closed 2021-01-15 by fredlcore

dukess, 2021-01-15:
else
              //Execute only if flag not set because strstr more expensive than bitwise operation
Serial.println(outBuf);

this println never displayed because "else" will work only when "ETag" header was found.

I've never noticed it because
Accept-Encoding: gzip, deflate
was always higher than
If-None-Match: "05122020184214773550"

fredlcore, 2021-01-15:
Well, it was executed on my system, that's how I checked whether the header came in correctly, that's what was wondering me, doesn't make sense...

1coderookie, 2021-01-15:
Just checked with the example from the config.h - works! Thanks! You guys are soooo quick :1st_place_medal:

hacki11, 2021-03-20:
> Great, thanks a lot! Still strange that it worked with the added printlns for me, but now it also works, so that's that :)!

the else has no brackets. Only the first line after the else will be executed in that branch.
the else should have `else { code }` if more than one line shall be executed.

I currently stumbled over "authentication does not work" and found the missing brackets and this thread. shall the memset and charcount update really be not called within the else branch?

fredlcore, 2021-03-20:
Thanks! That's why I always use brackets because it makes it immediately clear what code belongs to which if/else block, especially when indention is not correct (of which we still have quite a lot of code). I really should write down some code guidelines ;)...

hacki11, 2021-03-20:
Yeah, a auto format commit would be a good idea :)