fredlcore/BSB-LAN #292: Broken texts

fredlcore, 2021-01-16:
In some entries of the configuration menu, the first character of the first dropdown entry is cut off. At the same time, the option value of that entry is "0" which is tricky because it still works with "Off" entries (which are zero anyways), but if you look at the QAA model dropdown, the QAA 50 entry does not have the entry 82, but 0 as well.
@dukess: Do you have an idea why this could be the case? The entries in BSB_lan_defs.h and LANG_DE.h look allright to me.

1coderookie, 2021-01-16:
Do you mean the entry by the pulldown itself (like Ein/Aus) or the transmitted value/code?
Just tried with v2.0.76-20210114232524 - here everything looks ok within the pulldown entries (checked every pulldown):
[image]
Tried some switching between on/off (but not with every function though) and also that worked.
Tested with Chrome and FireFox.

fredlcore, 2021-01-16:
Ok, sorry, I should upload my own devices with the most recent firmware before I make such tests ;), sorry for the confusion, and thanks @1coderookie for double-checking :)!

closed 2021-01-16 by fredlcore