fredlcore/BSB-LAN #297: Add warning for missing (new) definements

fredlcore, 2021-01-20:
Could/should we add a section that puts out a compiler warning if a (newly added) config definement (or even variable, if possible) is not yet added to an existing installation? That would make troubleshooting for users easier if they upgrade and do not immediately know what they should add to their existing configuration.

dukess, 2021-01-20:
The only thing that comes to mind is to add #define with version number to the configuration file, and compare it with the program version number when compiling. If the versions are different, display a warning.

fredlcore, 2021-01-20:
I was thinking of adding something like this to _defs.h:
```
#if !defined(BUTTONS) || !defined(RGT_EMULALTOR) ...
error "Mandatory definement missing in BSB_lan_config.h - please compare with BSB_lan_config.h.default
#endif
```

dukess, 2021-01-20:
In this case warning will appear always when user will disable unused modules.

fredlcore, 2021-01-20:
You're right, then let's drop this.

closed 2021-01-20 by fredlcore