fredlcore/BSB-LAN #308: Setting time and weekday on PPS bus

fredlcore, 2021-01-23:
I have now figured out that it is also possible to set time and day of week on PPS bus. So far, we have one read-only text field that shows both day of week as well as time. The question now is how to include setting time and day of week:
1. Use the current text field and parse day of week and time from there
2. Add another dropdown wher the user can set the day of week and change the current one to only display (and set) the time.

Option 1 is probably taking up more resources as we need to parse the string entered and also have to take into account that day of week is different for different languages. With a dropdown, we would be on the safe(r) side - what do you think, @dukess? I would need your help if we needed to add another field because I haven't fully understood the way you register new variables with the EEPROM etc.

dukess, 2021-01-23:
Time and weekday stored in one bit-separated PPS field?

fredlcore, 2021-01-23:
Data format is four bytes, one for day of week (1=monday, 7=sunday), and one each for hours, minutes and seconds.
Since there will be (hopefully) a few more PPS parameters coming from my new testing system, I have reordered the PPS parameter list and inserted a few empty spaces so that in the future, we won't have to annoy users with shifting parameter numbers in the PPS category anymore.

A few things I haven't figured out how to do:

1. When I open the PPS category 48, I get all the newly sorted parameters, but only up to number 15067. 67 was previously the amount of PPS variables and the size of pps_values[]. I have increased PPS_ANZ from 68 to 92, so I thought that this was it, but it didn't help. 
2. There is CF_PPS_VALUES which has a size calculated like this: `sizeof(pps_values[PPS_TWS]) * (PPS_BRS - PPS_TWS + 1)`. This could be set to PPS_ANZ now, but I didn't want to do it because I'm not sure whether this would affect EEPROM versioning.

I hope I removed all calculations like `i<=PPS_TWI`... because now every parameter can have a FL_EEPROM flag which determines whether the value should be stored in EEPROM or not.

For the moment, this will mean around 20 "wasted" bytes of RAM which are not (yet) in use. I hope that this will still allow for Mega users to compile.

dukess, 2021-01-24:
>Data format is four bytes, one for day of week (1=monday, 7=sunday), and one each for hours, minutes and seconds.

Idea: we can add handler for PPS like for progam 0 and set full date/time. BSB-LAN source code already contain function for weekday calculation so here no problem to convert full date to PPS weekday/time.

>Since there will be (hopefully) a few more PPS parameters coming from my new testing system, I have reordered the PPS parameter list and inserted a few empty spaces so that in the future, we won't have to annoy users with shifting parameter numbers in the PPS category anymore.

It's a good idea!

>There is CF_PPS_VALUES which has a size calculated like this: sizeof(pps_values[PPS_TWS]) * (PPS_BRS - PPS_TWS + 1). This could be set to PPS_ANZ now, but I didn't want to do it because I'm not sure whether this would affect EEPROM versioning.

If the actual number of members increases, then the size in the EEPROM will also increase and default values will be loaded.

fredlcore, 2021-01-24:
The problem is that PPS does not support day of month and month of year, so only time and weekday. So we can't have a calculation of weekday. 
But if you say that it's ok to increase the size of pps_values[], then I can just use one of the spares and use the ENUM_WEEKDAY dropdown as I have done in the current test.

As for CF_PPS_VALUES, can I just safely set the size to PPS_ANZ (i.e. 92) or do I have to change anything else? Do I have to increase the EEPROM version number? If yes, could you tell me how and where? If increasing the size is all I need to do, then I'll do just that :).

dukess, 2021-01-25:
> As for CF_PPS_VALUES, can I just safely set the size to PPS_ANZ (i.e. 92) or do I have to change anything else?

if first EEPROM-writable element still pps_values[PPS_TWS] and all writable elements placed between PPS_BRS and PPS_TWS then nothing more needed.

> Do I have to increase the EEPROM version number?

No, it not help us: i.e. array size was increased CRC32 will be changed and default values will be loaded at startup.

> could you tell me how and where?

Version number placed in second column in config[] table in BSB_lan_EEPROMconfig.h. Increase version when you adding new options.

in BSB_lan_EEPROMconfig.h
```
#ifdef WEBCONFIG
  {CF_PPS_MODE,         1, CCAT_BUS,      CPI_DROPDOWN,  CDT_BYTE,           CF_PPS_WRITE_TXT, sizeof(pps_write)},//need handler
```
should be 
```
  {CF_PPS_MODE,         1, CCAT_BUS,      CPI_DROPDOWN,  CDT_BYTE,           CF_PPS_WRITE_TXT, sizeof(pps_write)},//need handler
#ifdef WEBCONFIG
```
because pps_write can be set through URL command and stored in EEPROM without WEBCONFIG.

fredlcore, 2021-01-25:
> if first EEPROM-writable element still pps_values[PPS_TWS] and all writable elements placed between PPS_BRS and PPS_TWS then nothing more needed.

No, I have changed this so that those values are written to EEPROM which have the FL_EEPROM flag.

As for BSB_lan_EEPROMconfig.h, I still do not understand. Currently, we have this line:
```
  {CF_PPS_VALUES,       0, CCAT_GENERAL,  CPI_NOTHING,   CDT_VOID,           NULL, sizeof(pps_values[PPS_TWS]) * (PPS_BRS - PPS_TWS + 1)}, //printlnToDebug(PSTR("Reading EEPROM..."));  for (int i=PPS_TWS;i<=PPS_BRS;i++){ ...}
#ifdef CONFIG_IN_EEPROM
#ifdef WEBCONFIG
```
Could/should I change this now to 
```
  {CF_PPS_VALUES,       0, CCAT_GENERAL,  CPI_NOTHING,   CDT_VOID,           NULL, PPS_ANZ},
```
?

As for CF_PPS_MODE, it is here already above `#ifdef WEBCONFIG`. Can you check the most recent version in the repo? Thanks!

dukess, 2021-01-25:
In my version we save in EEPROM only part of pps_settings (because some settings are volatile and not need for saving, AFAIK). In your case we have three way:
1. simpler: we can save/restore whole array 
`{CF_PPS_VALUES,       0, CCAT_GENERAL,  CPI_NOTHING,   CDT_VOID,           NULL, sizeof(pps_values[0]) * PPS_ANZ},` or
`{CF_PPS_VALUES,       0, CCAT_GENERAL,  CPI_NOTHING,   CDT_VOID,           NULL, sizeof(pps_values)},`
Contra: EEPROM consumption is equal pps_values array size; unpredicable values in array when restoring.
2. harder: we can save/restore parameters with FL_EEPROM flag. We need to add handler for CF_PPS_VALUES (for saving/restoring parameters with flag), add exclusion and handler for CF_PPS_VALUES to code like this: 
```
      for(uint8_t i = 0; i < CF_LAST_OPTION; i++){
        //do not write here MAX! settings
        if(i != CF_MAX_DEVICES && i != CF_MAX_DEVADDR) {
          writeToEEPROM(i);
        }
      }
```
never to use writeToEEPROM(CF_PPS_VALUES) and readFromEEPROM(CF_PPS_VALUES) for avoiding unpredicable restoring.
Contra: EEPROM consumption is equal pps_values array size.

2. Nightmare: calculate number of parameters with FL_EEPROM flag; add handler and exclusions as above; calculate shift for every pps_values when they are saving and restoring (for lesser EEPROM consumption).

fredlcore, 2021-01-25:
If we go for 1, I know we will waste a few bytes that are not used because they are volatile, read-only data. But at least I think these are not written to EEPROM and therefore are not incorrect when the EEPROM is being read at startup when we do it like this:
```
    uint8_t flags=get_cmdtbl_flags(i);
    if ((flags & FL_EEPROM) == FL_EEPROM && EEPROM_ready) {
      printFmtToDebug(PSTR("Writing EEPROM slot %d with value %u"), cmd_no, pps_values[cmd_no]);
      writelnToDebug();
      writeToEEPROM(CF_PPS_VALUES);
    }
```
This is how I have implemtented it in the current master repository. Previously, there was a `for` loop that would go through selected entries of pps_values[], so I was thinking that this here would work as well?

If that works, the question remains how I should change 
```
  {CF_PPS_VALUES,       0, CCAT_GENERAL,  CPI_NOTHING,   CDT_VOID,           NULL, sizeof(pps_values[PPS_TWS]) * (PPS_BRS - PPS_TWS + 1)}, //printlnToDebug(PSTR("Reading EEPROM..."));  for (int i=PPS_TWS;i<=PPS_BRS;i++){ ...}
```
then.

dukess, 2021-01-25:
{CF_PPS_VALUES, 0, CCAT_GENERAL, CPI_NOTHING, CDT_VOID, NULL, sizeof(pps_values)},

enough

fredlcore, 2021-01-25:
Thanks! It generally works now, but even though I get debug messages that parameters are written to EEPROM with the correct value, some parameters are not read back after restart while others do.
The ones that do not work are (in the new sorting order): 15000-15002 and 15004-15005. 15003 and all settable parameters above 15005 do work fine. Do you have any idea where the problem could lie?

dukess, 2021-01-25:
Ouch.
You forgot to change form 
registerConfigVariable(CF_PPS_VALUES, (byte *)&pps_values[PPS_TWS]);
to
registerConfigVariable(CF_PPS_VALUES, (byte *)pps_values);

fredlcore, 2021-01-25:
Ok, changed it - is it bad to have it run without that change?

dukess, 2021-01-25:
pps_values starts reading/writing from first element
&pps_values[PPS_TWS] - from PPS_TWS element.
So some first elements was never saved/restored.