fredlcore/BSB-LAN #312: Telnet debug output no longer working

fredlcore, 2021-01-26:
(At least) since the release 2.0.76-20210114232524 (i.e. since 14 January), a connection to the telnet port no longer works, a "connection refused" immediately comes up. This is different from connecting to a port on that IP that is not served (such as 4321) where the error takes much longer. @dukess, do you have an idea where I could be looking at?

dukess, 2021-01-26:
Last sync was 20.01.2021. Telnet was functional.
How many network services started?

fredlcore, 2021-01-26:
That may be the cause because after reboot, telnet was working but not webinterface. Is there a way to count sockets? Strange thing is that the same problem also occurs on W5500 Ethernet shield which should allow for 8 sockets.

dukess, 2021-01-26:
Now we have:
HTTP server for web interface/JSON
Telnet server
MQTT client with permanent connection
HTTP client for update checking.
MAX client
mDNS client

fredlcore, 2021-01-26:
Right, that's quite a bit, but in my configuration here I have turned off MAX, MQTT and update checking. That should leave three sockets which should still be fine even with the four sockets of the WIFI-ESP solution. Plus, sockets such as update checking are only occupied for a short time, so they would probably be the one who "lose out" when other sockets such as web server and telnet server have been opened first?

dukess, 2021-01-26:
Hmm... flashed fresh firmware - all ok with telnet but found and fix other bug.

fredlcore, 2021-01-26:
Ok, seems to work here as well again, but not with the other user who reported it. Strange, but will close this now until we have more info. Thanks!

closed 2021-01-26 by fredlcore