fredlcore/BSB-LAN #318: DHT22 never shows all values at the same time

1coderookie, 2021-01-30:
Hi guys,
can/did someone of you check if a DHT22 works correct with the current version? 
Mine always shows only one value when I query (which value then seems to be chosen randomly, in most cases it's temperature). It never shows all values at the same time. Tested with two different setups.
The sensor itself is ok though, just tested it with a Mega and v0.44 - three values are reported as expected.

dukess, 2021-01-31:
It is possible that DHT22 cannot transmit data at short intervals. I see two ways: 1. do buffering of data from the sensor; 2. add a delay after accessing the sensor. The second option is easier.

dukess, 2021-01-31:
Found info: interval between queries must be at least 2 sec.

1coderookie, 2021-01-31:
Thanks for answering. That's not the problem though. When I query at the old Mega setup, all three measured values are displayed. Now with Due & the current version, a query only shows one out of three measured values. Doesn't matter if I retry every minute or once in an hour. 
Here some screenshots:
[image]
[image]
[image]
 At the hardware side, everything seems to be ok though: sensor, voltage, wiring etc., and everything works fine at the Mega-setup also.
So I was wondering if maybe something was changed back in time that it doesn't work as expected.

dukess, 2021-01-31:
This PR may help: https://github.com/fredlcore/bsb_lan/pull/319

> Thanks for answering. That's not the problem though. When I query at the old Mega setup, all three measured values are displayed. Now with Due & the current version, a query only shows one out of three measured values. Doesn't matter if I retry every minute or once in an hour.

On version < 2.0 we read DHT22 once per measure temp and humidity.
On version 2.0 we need query every sensor 3 times: when measuring temperature, humidity and abs. humidity. (And here is raised problem: sensor can't be queried so quick)
I add some variables for save state between queries. If we query same sensor multiple times we get values without repeating sensor reads.

1coderookie, 2021-01-31:
Ah ok, got it, thanks! I thought you meant the query from my side.. Just because I'm curious: why the sensor has to be read out 3 times at v2.x?
Flashing your ino for testing right now..

1coderookie, 2021-01-31:
Now it works! Thanks! ..now I can disconnect that DHT and put it back in the box.. :)
[image]

dukess, 2021-01-31:
>Just because I'm curious: why the sensor has to be read out 3 times at v2.x?

Three different programs with data so when we build category we go through the program numbers in order and called sensor every time.
BME280 will queried 5 times. %) But it works with it.

dukess, 2021-01-31:
Fine!

1coderookie, 2021-01-31:
Did any changes occur with the query of DS18B20 also? Should I check with (multiple?) DS18B20, just to make sure..?

dukess, 2021-01-31:
Hope Dallas has no quirks. But yes DS sensors will called twice: for temperature value and serial number.

1coderookie, 2021-01-31:
Ok so I'll test it - just to make sure that everything works as expected ;)

1coderookie, 2021-01-31:
Nice, quick test with 4 sensors at pin 7 - works :)
[image]

closed 2021-01-31 by 1coderookie