fredlcore/BSB-LAN #320: Getting serial data transmission to work properly on ESP32

fredlcore, 2021-01-31:
This is the follow-up to Sub-Issue 4 from here: https://github.com/fredlcore/bsb_lan/issues/316#issuecomment-770295034
@do13, I changed the port setup in bsb.cpp so that it does not crash after a short time anymore, at least here. Please let me know if it works for you, too, it should actually just take the same configuration as in esp32-hal-uart.c.

You mentioned that the heater receives your data. Could you try and test whether it actually accepts it the first time it is sent? If you open a serial monitor and change the comfort temperature setpoint, you'll see that it is sent frequently to the heater. In my setup, it takes several telegrams for the value to be displayed in the heater's display, indicating that the transmission quality/timing is still a problem.

What is really weird still is that even when connecting the ESP32 to my Arduino Due which is hooked up to my BSB bus, I don't even see the data sent from the ESP32 on the Arduino. Consequently, neither BSB nor LPB work so far.

fredlcore, 2021-01-31:
Found the issue with the problem on BSB/LPB: Due code issues a flush() after each byte which somehow lead to a delay of 50ms between each byte - far too much! Why this was not a problem with PPS is beyond me, but I now changed the code so that it works. Not sure if collision detection also works the way we have implemented on the Arduino, maybe something you can look into. Let me know if the current code works without modifications for you, too, then we can close this issue.

do13, 2021-01-31:
Nope, it doesn't work. Looks good from a timing perspective, but each frame consists of 40 bytes:
[image]
Strange

do13, 2021-01-31:
That's probaly caused by the collision detection. If I get rid of the `goto retry;` It looks better

fredlcore, 2021-01-31:
Sorry, I don't understand: You say it's not working, but the timing looks good? Please be a bit more specific. Also, what "frame" are you referring to? I don't see 40 bytes there. What your image shows looks fine. If you could debug a bit more what the problem is on your side with the UART configuration, that would also be helpful if you want these changes to make it into the main repository.

do13, 2021-01-31:
Sorry, I was short in time.
The collision detection won't work this way. The byte is transferred to the uart layer (`serial->write(data);`) but not send out on the wire. That's the reason, why collision detection sometimes incorrectly detects its own data and retries the messages.

I don't know how this was supposed to work. Ideally the `rx_pin_read()` function should be called directly after the last bit has been send on the wire but we don't know when this happens.

Idea:
If the RS485 Half Duplex mode is enabled, the ESP32 provides such an automatic collision detection. Maybe this could be used?

fredlcore, 2021-01-31:
I think the way it was originally conceived was to pull the bus low (i.e. send a bit), read from the bus and see if it stays low. However, this does not detect a collision if the other bus member was sending a 0 (i.e. leaving the bus high).
If we don't have control over bitwise sending, there is no point in collision detection and we can just leave it out for the ESP32. Won't do that much harm because the other "official" devices will adhere to the collision detection and thus tolerate the "bullying" behaviour of the ESP32.

do13, 2021-02-01:
Yep, agreed.
Let's close this issue and step into the other challenges ...

fredlcore, 2021-02-01:
The question is still why my changes are not working on your ESP32. Is it because I set pins 16 an 17 as serial pins by default? Still, your setup should not crash or anything because of that.

fredlcore, 2021-02-02:
@do13, could you get back to us in this matter? Otherwise I assume it works and can close this issue.

do13, 2021-02-02:
Sorry, I was quite busy in the last days.
I haven't checked the latest implementation, but I can confirm that it had worked after you removed the collision detection. We can close this issue.

closed 2021-02-02 by fredlcore