fredlcore/BSB-LAN #322: Compiling for ESP32 under Arduino IDE fails in several points

fredlcore, 2021-02-02:
Compiling the source code for the ESP32 architecture under PlatformIO / VSCode works fine, but not under Arduino IDE.
Apart from some libraries that have to be removed/zipped because they are incompatible with ESP32, the main issue lies with this error:
```
/tmp/arduino_build_761330/sketch/BSB_lan.ino.cpp -o /tmp/arduino_build_761330/sketch/BSB_lan.ino.cpp.o
BSB_lan:6279:23: error: conflicting declaration of 'void loop()' with 'C' linkage
In file included from /tmp/arduino_build_761330/sketch/src/BSB/bsb.h:6:0,
                 from /home/db/BSB_lan/BSB_lan.ino:439:
/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Arduino.h:119:6: note: previous declaration with 'C++' linkage
 void loop(void);
      ^
```

The Arduino library pre-declares the `loop()` and `setup()` functions there, so that should actually not be a problem. Commenting out the `loop()` line allows compilation to proceed, but at the end the compiler fails because the loop-function was not predeclared, so it's a catch-22 situation.

I _think_ it could have something to do with the way the Arduino.h library is being included. Some libraries include it with quotes (`" "`) and some with brackets (`< >`). The former tries to load it from the working directory, the other one from the system library folder. It could be that with PlatformIO, these are the same and with the Arduino IDE, they are not, for whatever reason. 

A search for Arduino.h on the Arduino IDE system would probably be a good start to see where the library is located and whether there are any differences.

@do13 and @dukess, do you have any other ideas?

dukess, 2021-02-02:
i google same problem 
```
https://github.com/espressif/arduino-esp32/issues/2982
https://github.com/intel/linux-sgx/issues/208
```
And have ideas:
1. Move #include <arduino.h> to first non-commented line in .ino file
2. Try to around 
```
void setup(void);
void loop(void);
```
with 
```
#ifdef __cplusplus
extern "C" {
#endif
```
in ESP32's arduino.h.

Can't to try it right now: esp32 tools downloads very slow.

dukess, 2021-02-02:
Ufff... Too many errors:
First compilation:
```
"C:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/pthread" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/sdmmc" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/soc" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spi_flash" "-IC:\\Users\\DukeSS\\AppData\\Local\\Arduino15\\packages\\esp32\\h [... cut ...]
Используем библиотеку ESPmDNS версии 1.0 из папки: C:\Users\DukeSS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 
Используем библиотеку Wire версии 1.0.1 из папки: C:\Users\DukeSS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire 
Используем библиотеку SPI версии 1.0 из папки: C:\Users\DukeSS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 
Используем библиотеку WiFi версии 1.0 из папки: C:\Users\DukeSS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
exit status 1
```

fredlcore, 2021-02-02:
Found a workaround:

Moving
```
void setup(void);
void loop(void);
```
inside the 
```
#ifdef __cplusplus
extern "C" {
#endif
```
block prevents the error. Moving the `#include <Arduino.h>` to the first line did not help.
That's really strange, because the same Arduino.h works fine with PlatformIO :(.
The link you mentioned above (https://github.com/espressif/arduino-esp32/issues/2982) mentions `Time.h`, so I tried using the standard TimeLib, but that also did not help.

In any case, you have to move the following folders away from the main directory, otherwise compilation will fail:
```
ArduinoMDNS
BlueDot_BME280
WiFiSpi
WiFiSpiESP
```

The only remaining error then is related to the multiple definition of EEPROM, for which @do13 mentioned to add a compiler flag. However, I don't know how to add this flag inside the Arduino IDE, and also I would hope that this could be prevented with some kind of `#ifdef`. Any suggestions, @do13?

dukess, 2021-02-02:
I think i fix BME280 library: https://github.com/fredlcore/bsb_lan/pull/323/

dukess, 2021-02-02:
>That's really strange, because the same Arduino.h works fine with PlatformIO :(.

Different compilers/flags?

fredlcore, 2021-02-02:
The EEPROM problem lies in the automatic creation of the EEPROM object. We then recreate it with a specific size and name for ESP32. The source code says that with `NO_GLOBAL_INSTANCES` definement, we should be able to prevent that, but if I set this in the main code before including EEPROM.h, it does not seem to be active in the library (which makes sense in a way because I don't think that definements are passed through to libraries. Or are they?)

fredlcore, 2021-02-02:
> > That's really strange, because the same Arduino.h works fine with PlatformIO :(.
> Different compilers/flags?

Most likely, but I wonder why this is not a problem with other ESP32 sketches? I mean, the pre-declaration of loop() and setup() etc. should affect any sketch that has these functions (which are basically all Arduino sketches).

dukess, 2021-02-02:
>Most likely, but I wonder why this is not a problem with other ESP32 sketches? I mean, the pre-declaration of loop() and setup() etc. should affect any sketch that has these functions (which are basically all Arduino sketches).

BSB-LAN too complex for naming "sketch". Now it is "Enterprise software" :-)

dukess, 2021-02-02:
Found one more topic about EEPROM issue:
`https://github.com/espressif/arduino-esp32/issues/1238`

fredlcore, 2021-02-02:
Found a dirty hack:
Change
```
EEPROMClass EEPROM("eeprom1", 0x1000);
```
to
```
EEPROMClass EEPROM_ESP("eeprom1", 0x1000);
#define EEPROM EEPROM_ESP
```

fredlcore, 2021-02-02:
Now the only error left is an esptool.py error which complains about some load_module stuff, but that's probably Python 2.7 related.

dukess, 2021-02-02:
One more hack:
i added -DNO_GLOBAL_EEPROM to compiler.cpreprocessor.flags string in platform.txt for ESP32

fredlcore, 2021-02-02:
Yes, that would also work, but the goal would be for end users to not have to change any system files. That would be my goal for this strange "extern C" error as well...

dukess, 2021-02-02:
Understand this.

dukess, 2021-02-02:
Ok. One more solution: just remove EEPROMClass EEPROM("eeprom1", 0x1000);
In this case software will use "GLOBAL EEPROM". Their size is set in 
  #ifdef ESP32
  EEPROM.begin(4096); // size in Byte
  #endif
so here should be not a problem. I hope.

fredlcore, 2021-02-02:
Great!

fredlcore, 2021-02-02:
Weird: PlatformIO honors `#define NO_GLOBAL_EEPROM` and passes it on to the EEPROM.h library. Arduino IDE does not :(.

dukess, 2021-02-02:
Just forgot to add.

fredlcore, 2021-02-02:
Instead of forcing people to add the system Arduino.h library, we could (on ESP32) provide the "working" Arduino.h and include them from our src folder? I don't assume that it will change much in the future as it is so central?

dukess, 2021-02-02:
We can try it at least

fredlcore, 2021-02-02:
Changed it now, @1coderookie: Could you download the most recent master and try to compile first for the Due (no modifications necessary) and then (if all is fine) remove the two directories ./src/BSB_LAN/src/ArduinoMDNS and ./src/BSB_LAN/src/WiFiSpi - then switch to ESP32 and try to compile. That's probably the least effort we can provide for Users to install it on an ESP32 because the Arduino IDE does not provide means to exclude libraries from compiling. If this works, feel free to add this to the manual if you like :).

1coderookie, 2021-02-02:
> Changed it now, @1coderookie: Could you download the most recent master and try to compile on the Arduino IDE first for the Due (no modifications necessary)

Works!

> and then (if all is fine) remove the two directories ./src/BSB_LAN/src/ArduinoMDNS and ./src/BSB_LAN/src/WiFiSpi - then switch to ESP32 and try to compile. 

Hmm, following error appeared:
```
Gebauter Kern wird archiviert (zwischengespeichert) in: /tmp/arduino_cache_164501/core/core_f3536b11a825564661a7993787920424.a
Linking everything together...
/home/db/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc -nostdlib -L/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib -L/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group /tmp/arduino_build_527515/sketch/BSB_LAN.ino.cpp.o /tmp/arduino_build_527515/sketch/src/BSB/BSBSoftwareSerial.cpp.o /tmp/arduino_build_527515/sketch/src/BSB/bsb.cpp.o /tmp/arduino_build_527515/sketch/src/BlueDot_BME280/BlueDot_BME280.cpp.o /tmp/arduino_build_527515/sketch/src/CRC32/CRC32.cpp.o /tmp/arduino_build_527515/sketch/src/DHT/dht.cpp.o /tmp/arduino_build_527515/sketch/src/DallasTemperature/DallasTemperature.cpp.o /tmp/arduino_build_527515/sketch/src/OneWire/OneWire.cpp.o /tmp/arduino_build_527515/sketch/src/PubSubClient/src/PubSubClient.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/FatFile.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/FatFileLFN.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/FatFilePrint.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/FatFileSFN.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/FatVolume.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/FmtNumber.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/StdioStream.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/fstream.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/istream.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/FatLib/ostream.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/MinimumSerial.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SdCard/SdSpiCard.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SdCard/SdSpiCardEX.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SdCard/SdioCardEX.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SdCard/SdioTeensy.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SpiDriver/SdSpiESP8266.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SpiDriver/SdSpiParticle.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SpiDriver/SdSpiSAM3X.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SpiDriver/SdSpiSTM32.cpp.o /tmp/arduino_build_527515/sketch/src/SdFat/SpiDriver/SdSpiTeensy3.cpp.o /tmp/arduino_build_527515/sketch/src/Time/DateStrings.cpp.o /tmp/arduino_build_527515/sketch/src/Time/Time.cpp.o /tmp/arduino_build_527515/libraries/EEPROM/EEPROM.cpp.o /tmp/arduino_build_527515/libraries/ESPmDNS/ESPmDNS.cpp.o /tmp/arduino_build_527515/libraries/WiFi/ETH.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFi.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiAP.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiClient.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiGeneric.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiMulti.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiSTA.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiScan.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiServer.cpp.o /tmp/arduino_build_527515/libraries/WiFi/WiFiUdp.cpp.o /tmp/arduino_build_527515/libraries/SPI/SPI.cpp.o /tmp/arduino_build_527515/libraries/Wire/Wire.cpp.o /tmp/arduino_build_527515/core/core.a -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr -lstdc++ -Wl,--end-group -Wl,-EL -o /tmp/arduino_build_527515/BSB_LAN.ino.elf
/tmp/arduino_build_527515/core/core.a(main.cpp.o):(.literal._Z8loopTaskPv+0x4): undefined reference to `setup()'
/tmp/arduino_build_527515/core/core.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `loop()'
/tmp/arduino_build_527515/core/core.a(main.cpp.o): In function `loopTask(void*)':
/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp:14: undefined reference to `setup()'
/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp:17: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
[... cut ...]
Bibliothek WiFi in Version 1.0 im Ordner: /home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI  wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: /home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.
```

Of course I'll add it to the manual - I'm just waiting for a working solution for the ArduinoIDE - then I'll write the ESP32 chapter ;)

fredlcore, 2021-02-02:
The error is strange, do you have these lines in code or was that not yet replicated:
```
#if !defined(ESP32)
#include <Arduino.h>
#else
#include "src/Arduino.h"  // workaround for https://github.com/fredlcore/bsb_lan/issues/322#issuecomment-771544542
#endif
```

fredlcore, 2021-02-02:
I also made a comparison between a simple ESP32 sketch and ours. Ours seems to need 
```
 extern "C" void loop();
```
in the preprocessed file whereas the standard example just has
```
void loop();
```
Does that make things clearer?

1coderookie, 2021-02-02:
> The error is strange, do you have these lines in code or was that not yet replicated:
> 
> ```
> #if !defined(ESP32)
> #include <Arduino.h>
> #else
> #include "src/Arduino.h"  // workaround for https://github.com/fredlcore/bsb_lan/issues/322#issuecomment-771544542
> #endif
> ```

Yepp, I have them (downloaded the recent version before the test).

1coderookie, 2021-02-02:
> Does that make things clearer?

For me? No _lol_ - but I guess you meant @dukess.. Should I add that somewhere..?

fredlcore, 2021-02-02:
Grrr... The Arduino IDE seem to still load the system Arduino.h first :(.
But could you add this as the first line after the comments / before the inclusion of Arduino.h:
```
void loop();
```

At least then it compiles here without any problems, not sure though if it will run still ;)...

1coderookie, 2021-02-02:
Not sure where exactly: in the .ino right after the changelog comments and before the "#define LOGTELEGRAM_OFF 0" as first entry:
```
 *        - added some images of the BSB adapter
 *
 */
void loop();
#define LOGTELEGRAM_OFF 0
``` 
or like this:
```
#if !defined(ESP32)
void loop();
#include <Arduino.h>
```
?

fredlcore, 2021-02-02:
Both ways is fine, as long as it's before the Arduino.h include. Or you just pull the latest version from the repo.

1coderookie, 2021-02-02:
Like this 
```
#if !defined(ESP32)
void loop();
#include <Arduino.h>
#else
#include "src/Arduino.h"  // workaround for https://github.com/fredlcore/bsb_lan/issues/322#issuecomment-771544542
#endif
```
still the error comes up

```
/tmp/arduino_cache_164501/core/core_f3536b11a825564661a7993787920424.a(main.cpp.o):(.literal._Z8loopTaskPv+0x4): undefined reference to `setup()'
/tmp/arduino_cache_164501/core/core_f3536b11a825564661a7993787920424.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `loop()'
/tmp/arduino_cache_164501/core/core_f3536b11a825564661a7993787920424.a(main.cpp.o): In function `loopTask(void*)':
/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp:14: undefined reference to `setup()'
/home/db/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp:17: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
```
..but maybe that wasn't the right place where I put it..?

fredlcore, 2021-02-02:
No, before the #if

1coderookie, 2021-02-02:
```
void loop();
#if !defined(ESP32)
#include <Arduino.h>
#else
#include "src/Arduino.h"  // workaround for https://github.com/fredlcore/bsb_lan/issues/322#issuecomment-771544542
#endif
```
leads to "hello again, it's me, the loop-with-C-linkage"-error:
```
In file included from /home/db/bsb_lan-master/BSB_LAN/BSB_LAN.ino:443:0:
/tmp/arduino_build_527515/sketch/src/Arduino.h:127:15: error: conflicting declaration of 'void loop()' with 'C' linkage
 void loop(void);
               ^
/home/db/bsb_lan-master/BSB_LAN/BSB_LAN.ino:439:6: note: previous declaration with 'C++' linkage
 void loop();
      ^
```

fredlcore, 2021-02-02:
Ok, to be sure, just download the most recent version from GitHub please.

1coderookie, 2021-02-02:
Eehm:
```
Pyserial is not installed for /usr/bin/python. Check the README for installation instructions.
Traceback (most recent call last):
  File "/home/db/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 37, in <module>
    import serial
ImportError: No module named serial
```
ok, installed python-serial and did it again - IT COMPILED! :) (So python-serial-installation and probably Python in general must be mentioned in the manual later also?!)
Will try flashing now and after that will try to compile and flash for Due, hang on..

1coderookie, 2021-02-02:
ESP32: works! (And the MAC address is also displayed correct now btw!)

Due (WiFi-ESP8266 solution): Please don't hit me now..
```
Sketch wird kompiliert...
/home/db/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10808 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI/src -I/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire/src /tmp/arduino_build_954732/sketch/BSB_LAN.ino.cpp -o /tmp/arduino_build_954732/sketch/BSB_LAN.ino.cpp.o
BSB_LAN:439:6: error: previous declaration of 'void loop()' with 'C++' linkage
 void loop();
      ^
In file included from /home/db/bsb_lan-master/BSB_LAN/BSB_LAN.ino:441:0:
/home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/Arduino.h:54:24: error: conflicts with new declaration with 'C' linkage
 extern void loop( void ) ;
                        ^
Bibliothek SPI in Version 1.0 im Ordner: /home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/libraries/SPI  wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: /home/db/.arduino15/packages/arduino/hardware/sam/1.6.12/libraries/Wire  wird verwendet
exit status 1
previous declaration of 'void loop()' with 'C++' linkage
```

fredlcore, 2021-02-02:
Ok, please try and replace the single `void loop();` line with this with both systems:
```
#if defined(ESP32)
void loop();
#endif
```

1coderookie, 2021-02-02:
:dancers: HOOORAY!!! NOW YOU GOT IT! :dancers: 
Compiled and flashed both boards (Due with WiFi-solution though, but also tried to compile w/o WIFI-definement - that also worked) - access to webinterface works! :1st_place_medal:

fredlcore, 2021-02-02:
Perfect, thanks everyone! @1coderookie for intensive testing, @dukess for all the right pointers and @do13 for the general idea - I think this is really a great improvement for all the users who prefer WiFi over LAN and don't want to meddle with the ESP8266 WiFi bridge solution.
I'll then start designing a board that fits the NodeMCU so that these users will have a physically stable option as well...

1coderookie, 2021-02-02:
Yes, also many thanks from my side to all of you guys! 
So I'll start writing the ESP32 chapters..

do13, 2021-02-02:
> Ok. One more solution: just remove EEPROMClass EEPROM("eeprom1", 0x1000);
> In this case software will use "GLOBAL EEPROM". Its size is set in

Did you checked this on a real device?
I added this for a very good reason. If the size parameter isn't used, a default value of 0 is set internally and the function EEPROM.length() returns 0 which breaks functions like clearEEPROM() and EEPROM_dump()

fredlcore, 2021-02-02:
Thanks for the hint, this seems to be a design flaw in the ESP32 code: https://github.com/espressif/arduino-esp32/issues/2280
However, since we will always use 4kB of EEPROM memory (because that is what the Arduino Mega has built-in and which is what we're using the the PCB board for the Due), we can just generally use a `#define EEPROM_SIZE 4096` and replace this for EEPROM.length(). Saves us more `#if`s in the code.

fredlcore, 2021-02-02:
I'm not sure whether this is related to my changes now or whether this was a problem before, but writing settings to EEPROM does not seem to work, and when trying to clear the EEPROM, I get lots of 
```
[E][EEPROM.cpp:199] commit(): error in write                                    
[E][EEPROM.cpp:199] commit(): error in write     
```
lines. 

**EDIT:**
When I change back to 
```
EEPROMClass EEPROM_ESP("eeprom1", 0x1000);
#define EEPROM EEPROM_ESP     // This is a dirty hack because the Arduino IDE does not pass on #define NO_GLOBAL_EEPROM which would prevent the double declaration of the EEPROM object
```
I can write to EEPROM, but when writing to or clearing the EEPROM, I still get these error messages.

Any ideas?

do13, 2021-02-02:
I guess this is due to the global instance which interferces with EEPROM_ESP class in the sketch. 
Could you try building with NO_GLOBAL_EEPROM or try the following:
`EEPROMClass EEPROM_ESP("eeprom_esp1", 0x1000);`

fredlcore, 2021-02-02:
If I write "EEPROM_ESP" it would not work because all references are to EEPROM.
So I added `-DNO_GLOBAL_EEPROM` to platformio.ini and used this:
`EEPROMClass EEPROM("eeprom1", 0x1000);`
However, the effect is the same.
The strange thing is that when writing to EEPROM, it's always exactly 13 such error lines, no more, no less. @dukess, does that number ring a bell regarding the EEPROM layout?

I also modified EEPROM.h to give me _handle, _name and _size and these are always 1, "eeprom1" and 4096 respectively. _data in these 13 error lines is
`Ö???mKEQ0502326`
where the KEQ... part is the value of the first MAX id.

Upon booting the ESP32, I get a 
`EEPROM schema CRC mismatch`
line. @dukess, what does this mean? I also cleared the EEPROM via /NE, but that didn't help.

fredlcore, 2021-02-02:
Ok, the error message that nvs_set_blob() returns may actually be helpful: It returns ESP_ERR_NVS_NOT_ENOUGH_SPACE (0x1105): There is not enough space in the underlying storage to save the value.

fredlcore, 2021-02-02:
It seems we have to adjust the partition table in order to store more than some 300 bytes:
https://esp32.com/viewtopic.php?t=5871#p25644

fredlcore, 2021-02-02:
I think we have to do away with the built-in EEPROM and its library because it seems that it is not able to store floats or doubles:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html
(see keys and values).

I suggest we go the same way as with the Due and use the I2C EEPROM that is already on the board.

dukess, 2021-02-03:
@fredlcore 
> does that number ring a bell regarding the EEPROM layout?

Nope. Looks like EEPROM page size is 16 bytes. (1+1+4+10, where 10 - first 10 bytes of MAX! devices list)

>EEPROM schema CRC mismatch

This mean that stored in EEPROM CRC32 value not equal to value which calculated at startup. Reasons:
1. EEPROM was erased or first used.
2. EEPROM broken.
3. Settings schema was updated without version number increasing.

> I think we have to do away with the built-in EEPROM

It is simpler for support and board development but little bit expensive for ESP32 users.

>because it seems that it is not able to store floats or doubles (necessary for PPS at least if you want to be able to store fractions of temperature such as 19.5):

Here is no problem because we store all parameters as byte arrays. (That's why I used the checksum calculation for the scheme, since with this approach any value in the EEPROM is valid.)

fredlcore, 2021-02-03:
Thanks for the clarifications - so it seems that the EEPROM size is too big for the standard partiotion layout of the ESP32 which is why the EEPROM CRC errors show up. @1coderookie, could you confirm this before updating to the new version?

As for changing the partition layout, this is not trivial if you want to change the size of the NVS partition as the Arduino IDE only allows you to change the size between flash and SPIFFS file system. So at least for Arduino IDE users, it would not be possible to get the system up and running. I'm happy to stand corrected here, but if that remains the case and if the EEPROM size is indeed limited to 512 bytes, I see no other way than to get back to the I2C EEPROM. In the end, it's just 30 cents or so of extra cost.

1coderookie, 2021-02-03:
Well, I'm not sure if I got the problem right, but first of all: 
I just changed some settings via webconfig just for testing and it seems like it was successful. Reboot occured when I activated 'check for updates' which is ok though iirc, deactivation was possible w/o reboot. 
MQTT, averages - every changed setting seems to be accepted.
Somewhere you mentioned an unneccessary reboot after changing the buttons for the room unit emulation - but that's needed to apply the settings, so the reoboot is totally fine in that case also (see comment in BSB_LAN_EEPROMconfig.h). 
I'd have to make further tests though to see if the ESP is stable when changing the settings more often - cuz right now it seems like the ESP crashed and isn't accessable anymore, even though the router shows its online..

About EEPROM size: 
In ArduinoIDE I used the default settings
- Flash size: 4MB (32MB)
- Partition scheme: Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
And I didn't get any EEPROM CRC yet (during flashing, right?). 
Or am I on the wrong path..? Let's maybe get that clear via email..?! ;)

1coderookie, 2021-02-03:
Ok, ESP32 wasn't accessable anymore, so I tried to re-flash the version from yesterday - seems that changing the settings via webconfig killed something:
```
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 4c:11:ae:9b:dd:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0x2C)
A fatal error occurred: Invalid head of packet (0x2C)
``` 
Then I noticed that the upload speed wasn't set to 115200 - I changed it and right now it's flashing.

fredlcore, 2021-02-03:
Please check for error messages in serial monitor and load configuration via /C, otherwise they just stay in memory.

1coderookie, 2021-02-03:
..I'll send you the stuff via email, don't wanna spam the issue.. ;)

dukess, 2021-02-03:
> I see no other way than to get back to the I2C EEPROM. In the end, it's just 30 cents or so of extra cost.

At this time is better to stay with I2C EEPROM

We can reduce numbers of parameters that require reboots  after configuration changes but then we need handlers for applying new values on the fly.

fredlcore, 2021-02-03:
Ah, I didn't know this! So there is a regular reboot after saving configuration? Then that's fine, I just thought that's a crash followed by a reboot.

1coderookie, 2021-02-03:
Yes, at least certain settings require and initiate a reboot like I mentioned above. But the reboot didn't seem to be 100% successful as I reported to you via email, but let's test and see..

fredlcore, 2021-02-03:
That was my understanding, too, that _certain_ settings require a reboot - and I thought that only in that cases a reboot is initiated. So for simply changing log parameters or something like that, I thought that there is no reboot. But I stand corrected, and there's no need right now to change that.

1coderookie, 2021-02-03:
Yes, you're right - also from my perspective a reboot doesn't _always_ occur - at least with the Due. -> Is that correct @dukess?
It seemed to occur with the ESP though, as I mentioned in the edit above. But we'll test, compare and see, once the ESP runs again..

dukess, 2021-02-03:
> So for simply changing log parameters or something like that, I thought that there is no reboot. 

These parameters not needed to reboot.
Not need reboot now:
bus type, addresses, PSP mode
RGT emulation (sensors)
averages,
MAX settings
MQTT settings

Do reboot:
Network parameters: MAC. DHCP mode, IP address, etc.
RGT emulation (buttons settings)

fredlcore, 2021-02-03:
Thanks for the clarification!

[29 more replies]

closed 2021-02-08 by fredlcore