fredlcore/BSB-LAN #326: Sending commands via serial no longer works (and crashes on ESP32)

fredlcore, 2021-02-03:
Until at least version 0.44, it was possible to send commands similar to the URL commands via the serial interface to the Arduino. Now these commands are no longer evaluated. Furthermore, entering characters result in a crash of the ESP32 if the serial-entered string does not contain a space.

The problematic code seems to be this:
```
        char *u_s = strchr(cLineBuffer,' ');
        char *u_e = strchr(u_s + 1,' ');
        if(u_e) u_e[0] = 0;
        strcpy(cLineBuffer, u_s + 1);
```
In version 0.44, the code read:
```
        String urlString = String(cLineBuffer);
        urlString = urlString.substring(urlString.indexOf('/'), urlString.indexOf(' ', urlString.indexOf('/')));
        DebugOutput.println(urlString);
        urlString.toCharArray(cLineBuffer, MaxArrayElement);
```
(BTW: I know my code is often hard to read, but if I hadn't compared with the 0.44 code, I would have no idea what the current code snippet does ;).)

In the previous version, urlString would still be a valid string if no space was contained. So whether I enter via serial `/4444/710` or via HTTP `GET /4444/710`, urlString would always be `/4444/710` because a missing space would result in an end-index of -1 and thus cover the whole string. In the current code, a missing space would probably evaluate to 0xFF or something else that makes the `strcpy` line crash.

Another observation: cLineBuffer already 'arrives' with some extra baggage:
Entering `GET /710` shows this when printed back to serial:
`GET /710|??h0???????`
I wonder whether there is a null-termination missing somewhere when the string is composed, but it does not seem to be a problem when the URL string is parsed later on. For serial entered commands, there is no result being printed.

fredlcore, 2021-02-03:
To summarize what needs to be done here is: Take the string that begins with a slash and ends with either a space or the end of the string.

fredlcore, 2021-02-03:
I've reverted to the (slightly modified) previous approach now. I assume that this eats up more memory, so if that's the case, @dukess, and you can convert this to less memory consuming code, feel free to go ahead :)!

closed 2021-02-03 by fredlcore

dukess, 2021-02-03:
The incorrect behavior of this piece of code was most likely caused by an error. And yes, cLineBuffer is better to completely reset before filling.
I look on this little bit later.

dukess, 2021-02-03:
@fredlcore Can you test this patch?
I tried it on Mega and now possible to send commands from serial console.

fredlcore, 2021-02-03:
Great, it works, thanks!