fredlcore/BSB-LAN #329: Using the internal flash as EEPROM on the ESP32

fredlcore, 2021-02-04:
@dukess, @do13 and @1coderookie: As it seems that the ESP32 has insufficient pull-ups necessary for using I2C without adding extra pull-up resistors, I would like to at least make sure that there is no "easy" way for end-users to configure their ESP32 so that 4kB of the ESP32's internal flash can be used as EEPROM. While adding two more resistors is not that much of work or cost, it would prevent switching the BSB-LAN board between the systems because the Arduino Mega and Due already have properly sized external pull-ups on their respective boards.

If accessing 4kB of flash is not possible without complex changes in the partitioning scheme, then I think we'll have to go for the I2C EEPROM, but I want to make sure that we have exhausted all other ways.

1coderookie, 2021-02-04:
First of all: thanks for the hint about the pullups - didn't think of that when I connected the EEPROM, but that could be the solution for my problems from yesterday! @fredlcore

About EEPROM: first of all, please keep in mind that I'm not a programmer and that I probably didn't get the problem right or that my comment could be nonsense - so please don't be mad at me in that case.. ;) 
1. The 4kB probably have to be set by the partition table: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html
2. [Here](https://tinkerman.cat/post/eeprom-rotation-for-esp8266-and-esp32/) I read about the need to make a full sector erase before writing new data to EEPROM - as a solution, the guy wrote something called "EEPROM rotation" - maybe that could be a solution for a probably problematic writing..?
3. The ESP32 has NVS instead of EEPROM, so why don't/can't you guys use that?  
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html
4. Don't know if that's the same 'way' of using/programming/accessing the nvs, but besides the API from Espressif I found examples which are using `#include <Preferences.h>` to access nvs (German examples for @fredlcore: https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/esp32-daten-und-einstellungen-dauerhaft-speichern and http://esp32-server.de/eeprom/)

1coderookie, 2021-02-04:
5. Found something else: 
Converted EEPROM library to use NVS instead of partition:
https://github.com/espressif/arduino-esp32/commit/619568db5bcc219091c653a5ced5e378b3a5643b
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv

fredlcore, 2021-02-04:
Thanks for your research. The thing is that the ESP32 does not have an EEPROM per se, it takes part of its flash as nvs (non volatile storage), so the EEPROM library wraps around that to "look like it".

The errors I got when using the EEPROM library indicated that there is not enough space to write which means that the (my) partition size for nvs is to small. The partition you linked to may be better or even working fine, but I have not found a(n easy) way to wire this from the Arduino IDE which only offers pre-configured partitions.

1coderookie, 2021-02-04:
> Thanks for your research. The thing is that the ESP32 does not have an EEPROM per se, it takes part of its flash as nvs (non volatile storage), so the EEPROM library wraps around that to "look like it".

I know - even though I understood it that way, that the "EEPROM" is taken from the flash and the NVS is not used for "EEPROM" from Arduino per se. So that's why I mentioned the NVS & NVS-API extra, also the `#include <Preferences.h>` solutions maybe to use it explicitely with a different NVS-specific approach. But of course I can be wrong ;)
  
> The errors I got when using the EEPROM library indicated that there is not enough space to write which means that the (my) partition size for nvs is to small. The partition you linked to may be [...]

Besides creating a csv with a decent partition table and copy it manually to the ESP32-partitions-path of the Arduino-ESP32-packages to make it accessable within the entry "Partition Scheme" in the Arduino IDE, there maybe is an easier way by using this ESP32 partition manager as a plugin tool for the Ardunio IDE: https://github.com/francis94c/ESP32Partitions  ?!?

fredlcore, 2021-02-04:
I understood what you wrote, but again: everything is flash on the ESP32, its just different partitions for different usages, and nvs is one of them. Preferences.h is no different from the general approach.
ESP32Partitions _could_ work, but I'm not sure if this is an add-on that runs only on some OS (Windows?) only, and still people would have to fiddle with configurations they don't understand, that can break things, or worse: lead to inconsistent behaviour later on which will be very difficult to pinpoint (eventually by us). So unless one of the default partition settings in the Arduino IDE is working for 4k "EEPROM-style" flash, this is not an option for me.

1coderookie, 2021-02-04:
Ok, sorry, thought it could be a solution if we write the how-to-install about that ESPtool etc in the manual also.. ;)

fredlcore, 2021-02-04:
As requested by @dukess, here is the output the serial monitor when reverting back to EEPROM.h. I have modified EEPROM.cpp to output the parameters sent to `nvs_set_blob()` when doing a `EEPROM.commit()`, i.e. _handle, _name, _data and _size. Size is always 4096 bytes (which makes sense as commit writes everything at once), and the error code is always 0x1105 which is `ESP_ERR_NVS_NOT_ENOUGH_SPACE (0x1105): There is not enough space in the underlying storage to save the value`

I also changed EEPROM.cpp in such a way that it would also print a debug message if a value has been written successfully, but that never happens. So I'm wondering whether maybe `_name`which is set to `eeprom1` could be wrong because this is not defined in the(/my) partition table?

fredlcore, 2021-02-04:
[EEPROM-log.txt](https://github.com/fredlcore/bsb_lan/files/5925493/EEPROM-log.txt)

dukess, 2021-02-04:
I can't understand anything. There are no messages in the log except for the error 0x1105 (not enough space). But there are no messages at the initialization stage.
Right now I have two idea:
1. Try to reduce EEPROM size to 4000 (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#c.ESP_ERR_NVS_NOT_ENOUGH_SPACE: [in] length: length of binary value to set, in bytes; Maximum length is 508000 bytes or (97.6% of the partition size - 4000) bytes whichever is lower.)
2. Check free space in NVS partition: it can be filled after experiments

fredlcore, 2021-02-04:
I think I found some parts of the problem:

1. First of all, the name in the EEPROMClass declaration must match an existing partition. Maybe "eeprom1" exists on Olimex or @do13 created a custom partition table that contains "eeprom1", but in almost any other case it won't work. I also tried "eeprom" which is the default in EEPROMClass, but at least here on my NodeMCU-ESP32, this also does not exist as default. I then checked the default partition table in PlatformIO, and among the partitions was "nvs" which I then successfully used. We would have to check the partition names that are created by default in the Arduino IDE to pick one that will work in that setting (and also provides OTA capabilities, another thing to keep in mind)
2. Once I had identified "nvs" as the correct partition name, things looked much better, but I still got an error here and there, and most importantly this warning at boot: `[E][EEPROM.cpp:104] begin(): Not enough space to expand EEPROM from 0 to 4096`. So I changed the size to 512 and it was working fine, but 2048 gave me again quite a number of errors. The problem is that it somehow seems to be stuck at 2048 now, no matter what I enter: If I set it to 1024, the `_size` variable in the logs still shows 2048. Also when I set it from 1024 to 4096, I get `Not enough space to expand EEPROM from 2048 to 4096
Using settings from config file`. Clearing the EEPROM with /NE did not help either.

So the questions are now: 
1. How do I figure out the size of my nvs partition. Is it really just 512 bytes (as many sources say that this is the limit)? Or is it larger than that but (in my case) smaller than 2048?
2. How do I "shrink" my "EEPROM" from 2048 down to a size that at least runs without errors?
3. Is 512 a hard limit, no matter how big the "nvs" partition is? If not, is there a standard partition layout in the Arduino IDE that we can refer to in the manual that will make sure that we will have 4096 bytes on all systems (Mega, Due, ESP32)?
4. Would 512 bytes be enough (for the moment)?

fredlcore, 2021-02-04:
Bizarre: I now uploaded a standard "blink" sketch from the Arduino IDE using the "Minimal SPIFFS" partition layout which should have an nvs partition size of 20480 bytes, but when I then re-upload BSB-LAN and try a size of 4096, I get this error:
`[E][EEPROM.cpp:104] begin(): Not enough space to expand EEPROM from 2049 to 4096`
...and 2049 was the size I tried before in BSB-LAN prior to uploading the "blink" sketch. So somehow this value must have "survived" the flash procedures in the meantime.
I'm now moving myself upwards in terms of size, and while I first found a barrier at 2432, I can now get slightly above it (currently 2448 works, but 2560 does not). Also, errors are coming back with these larger sizes, but no matter what I enter, I can't "downsize" it again. I wonder whether PlatformIO stores partition data somewhere and uploads it with every flash?

fredlcore, 2021-02-04:
Ok, it must have to do with the way PlatformIO creates/writes/updates the partition table. When I use the "Minimal SPIFFS" partition layout in the Arduino IDE and flash the BSB-LAN sketch, I get no EEPROM errors whatsoever. When I use the same CSV in platformio.ini, there seems to be no change.

However, what is strange is that with the Arduino flashed version, the configuration seems not to be written to EEPROM/NVS. Upon boot, it says `Using settings from config file` and the EEPROM dump contains all zeros, even after /NE.

fredlcore, 2021-02-04:
I switched back to internal ESP32's EEPROM now in the master repository, just in case you guys want to try it out yourselves...

fredlcore, 2021-02-04:
Ok, I don't know why this made a difference, but once I completely erased the flash with this command:
`esptool.py --port /dev/tty.SLAB_USBtoUART --chip esp32 erase_flash`
(`/dev/tty.SLAB_USBtoUART` is the COM-port of the ESP32)
I now no longer get the errors when reading/writing from the ESP's "EEPROM" and values are in fact written and read back after resets :)! Thanks for all the ideas and testing!
Could you please test this command (probably only works well on a Linux/Mac machine) which prints out the current partition of your ESP32:
`(python esptool.py --port /dev/tty.SLAB_USBtoUART read_flash 0x8000 0xc00 /dev/fd/3 >&2) 3>&1 | python3 gen_esp32part.py /dev/fd/0`
You would have to enter this command in the directory where esptool.py and gen_esp32part.py are located, with PlatformIO this is `~/.platformio/packages/framework-arduinoespressif32/tools`. And of course adjust your COM-port.
I hope that "nvs" will be a partition name that all of your have. Once people use the min_spiffs partition layout, it will be created anyway, but it would be even more failsafe if that was a name (other than "eeprom" or "eeprom1") that every ESP32 model has.

fredlcore, 2021-02-04:
If you don't get a partition table from the above command, you may have to change 0x8000 to 0x9000 (for newer boards - whatever "newer" means).

1coderookie, 2021-02-05:
Partition (still from "Default 4MB with spiffs" though) NVS exists:

Espressif ESP32 Partition Table
Name, Type, SubType, Offset, Size, Flags
_nvs,data,nvs,0x9000,20K,_
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,1280K,
app1,app,ota_1,0x150000,1280K,
spiffs,data,spiffs,0x290000,1472K,

fredlcore, 2021-02-08:
Thanks! @1coderookie also had a similar result, so we can safely assume that nvs is the correct partition name across multiple settings. Maybe @do13 can post his partition table where eeprom1 was used, too, but for now, I'll close this issue.

closed 2021-02-08 by fredlcore