fredlcore/BSB-LAN #331: (New) configuration options that should be included into web-based configuration

fredlcore, 2021-02-06:
Here I would like to list all configurable parameters/options that should be added to the web-based configuration at some point. No need to do this right now, because if I understand correctly, each newly added batch of parameters increases the EEPROM version number of which we only have 254(?). So these could be added at a time when we have to make changes to the list of parameters anyway.

- Replace definement `ENABLE_ESP32_OTA` with web-based configuration

dukess, 2021-02-08:
> if I understand correctly, each newly added batch of parameters increases the EEPROM version number of which we only have 254(?)

Yes, you right. So you right one more time: it is better to add some options at one time than one-by-one adding.

fredlcore, 2021-09-15:
@dukess and @1coderookie, can you think of other settings that we have added in the last couple of months that are missing in this list? Once it's "complete", could you add these parameters to the configuration, @dukess? No need to add the code that replaces the ifdefs, just to get the configuration up to date.

1coderookie, 2021-09-15:
Will think about it, but right now only the MQTT-clientID-issue/solution pops up in my head (https://github.com/fredlcore/BSB-LAN/pull/376) - maybe that should be settable via Webconfig also..?

dukess, 2021-09-15:
Found MDNS_HOSTNAME in config

dukess, 2021-09-15:
ENABLE_ESP32_OTA - can be in Webconfig
WIFI - leave as definition because different #include is used. In other case we should think about HAL development.
ESP32_USE_SD - leave as definition for same reason as above.
ONE_WIRE_BUS - can be in Webconfig. But we can use zero value of One_Wire_Pin as off state and non-zero as on-state.
DHT_BUS - DHT_Pins already in Webconfig. Or we should to add additional parameter for DHT on/off? Empty value will disable DHT.
BME280 - can be in Webconfig

dukess, 2021-09-15:
> Will think about it, but right now only the MQTT-clientID-issue/solution pops up in my head (#376) - maybe that should be settable via Webconfig also..?

MQTTDeviceID already presented in Webconfig.

fredlcore, 2021-09-15:
Thanks for your quick feedback, you two!
As for WiFi, you are right, that won't be able to configure via WebIF. Once we provide pre-compiled binaries, we'll have to create several versions then (or focus just on one of them).
As for the sensors: Can we use zero value  as "off" for 1Wire, DHT and BME280 respectively? I think it makes sense that if the value is empty, then the function is off. Saves us several knobs to further configure...

1coderookie, 2021-09-15:
About the sensors: "0" for OFF would be ok for me, but then 
a) for DHT&OneWire the pin number(s) if connected and
b) for BME (I2C) the amount of sensors attached, 
so that one can change config fully using webconfig, right? Or isn't that possible? Cuz @dukess wrote "ONE_WIRE_BUS - can be in Webconfig. But we can use zero value of One_Wire_Pin as off state and _non-zero as on-state_".

dukess, 2021-09-16:
> As for the sensors: Can we use zero value as "off" for 1Wire, DHT and BME280 respectively? I think it makes sense that if the value is empty, then the function is off. Saves us several knobs to further configure...

Yes, we can. 

> so that one can change config fully using webconfig, right? Or isn't that possible?

I think we can do it, but we need some experiments with BME library.

Right now we have "enableOneWireBus" variable which looks useless and can be replaced with "One_Wire_Pin" without any quirks.

dukess, 2021-09-16:
Patches:  https://github.com/dukess/bsb_lan/commit/064a582c848412c4251a3926462d9488c31ecc29 and https://github.com/dukess/bsb_lan/commit/072550ea5ebd3cb7ee2dfc1c4f5a4a70b31dc4ef

fredlcore, 2021-09-16:
Thanks for the patches and the explanation; since @1coderookie has BME280 sensors running(?), maybe he can help with the experiments?

dukess, 2021-10-20:
Oh, you right, we're forgot about "show_unknown".

fredlcore, 2021-10-30:
Could you also add a web-config option for the bus pins? For the moment, we won't need this, but once we provide pre-compiled binaries, people would have to be able to change the pins via web-interface.

dukess, 2021-10-31:
Why not? I see no problem for implement this.

dukess, 2021-10-31:
should i add fixed_device_family and fixed_device_variant options too?

fredlcore, 2021-10-31:
Yes, that would be great! Not that I know of any non-debugging use  right now, but if we're working on the EEPROM layout, it's best to add it right away...

1coderookie, 2021-11-01:
I think we actually would need (pretty much) everything from the config.h changeable using the webconfig when we wanna switch to the precompiled binaries, right..?

fredlcore, 2021-11-01:
Well, I had hoped that we could do away with some settings as well as take away some of the very special configuration options (such as fixed device family/variant etc.), but I think these would have been the last ones(?) we didn't have already, so I thought we can just add them as well...

dukess, 2021-11-01:
As a last resort, we can always hide these options

fredlcore, 2021-11-01:
Good idea. Maybe we should then have an option for basic and advanced settings? Because I think half of the options (at least) is not going to be used by 90% of the users ;)...

fredlcore, 2021-11-01:
We wouldn't have to store this in EEPROM, it could just be a CGI variable or a URL command such as /CE (configuration extended) or so...

dukess, 2021-11-01:
If we want divide configuration for "basic settings" and "advanced settings" we should to add one more field in options table like "flags" field in cmdtbl and build configuration page depending on flags states.

dukess, 2021-11-01:
There is not much difference how we will separate the settings, but I would prefer to store the switching flag in the EEPROM instead two URLs :)

fredlcore, 2021-11-01:
If you prefer it that way, I'm of course fine with it :)

dukess, 2021-11-01:
which options should be in advanced mode only?
advanced:
TX/RX pins, 
device family/model

dukess, 2021-11-01:
Debug options:
CF_DEBUG
CF_VERBOSE
CF_MONITOR
CF_SHOW_UNKNOWN

fredlcore, 2021-11-01:
I think for "basic" configuration, we should only display those options that are necessary to get it running (like the way my Fritz Box router does it). Everything else, including MQTT, IPWE, room unit emulation etc. is only for people who have some level of experience.
So I would show "General", "Network" and _maybe_ "Logging"/"Averages" categories and leave the rest in advanced settings.

fredlcore, 2021-11-01:
Within "Bus", I would keep "buy type" and "PPS mode". Within "Network", I would also move HTTP Auth, TCP port, MAC address, trusted IPs and mDNS hostname to advanced.

dukess, 2021-11-01:
[image]

fredlcore, 2021-11-01:
Awesome :)!

fredlcore, 2021-12-07:
@dukess: Could you add a configuration option for ESP32 to choose between using SD card and internal flash so we're able to remove the ESP32_USE_SD definement? Thanks!

dukess, 2021-12-08:
I'm not sure that is possible without HAL. I don't know how to do it without much effort.

1coderookie, 2021-12-08:
HAL: "I'm sorry Dave, I'm afraid I can't do that." (..sorry, I had to write this.. ;) )

fredlcore, 2021-12-08:
Hm, I'm not near the source code now, but I thought it could be handled the same way we are dealing with the serial interface. Both SoftwareSerial as well as HardwareSerial are of the Stream class and we just use a pointer to the device that is actually used. But maybe the SPIFFS, SD_MMC and SdFat have nothing like that in common. 
In that case we would probably have to force Olimex users to use the SD card (if so desired) and the NodeMCU users to SPIFFS. That would kick out those users who use a ESP32 board that has SD on board but is not an Olimex, but I guess that would be acceptable as well...

dukess, 2021-12-08:
...Or follow the path of compiling many binaries for different architectures:
ESP32 - SPIFFS, SD_MMC, SdFat
NodeMCU - SPIFFS and, may be, SdFat
Due - SdFat (+WiFi or Ethernet shield)

fredlcore, 2021-12-08:
Yes, but we would have _so_ many binaries then, because all of these will have to be compiled for all the different languages, and then again one with WiFi and one with Ethernet etc. The amount of time it will take until on set is fully compiled will take ages ;)...

fredlcore, 2021-12-25:
@dukess: Could you confirm/explain why switching between internal flash and (external) SD card on ESP32 could work the same way as we're doing it with ComServer/ComClient or &Serial that refers to HardwareSerial or SoftwareSerial?

closed 2023-03-21 by fredlcore

fredlcore, 2024-03-02:
Just for the record, WiFi and SD card access are now also changeable via configuration in webinterface, which makes all relevant configuration options available there.