fredlcore/BSB-LAN #336: Not logging to SD from 2.0 ver.

proasnet, 2021-02-16:
Hello, from version 2.0 I have BSB-LAN on Arduino Due with ethernet shield WZ5100 and 2GB SD card. Before this update, I had BSB-LAN on Arduino Mega with the same ethernet shield. Now from 2.0 my BSB-LAN not logging data to SD card. Yesterday I upgraded to the last version and nothing, in the web UI I have an empty graphs and logs. SD is formated to FAT32, not file. I try re-create file by /D0 parameter to URL and nothing...
In the older version 1 on Mega this was working.
Can help me somebody?
Thank you.

fredlcore, 2021-02-16:
Please post your BSB_LAN_config.h, a screenshot of your web configuration (/C) and the output of the serial monitor during startup.

proasnet, 2021-02-17:
Hi, I am sorry for later answer. Now I can connect to my BSB Lan.
Here is serial debug:
`READY
Reading EEPROM
Reading done.
EEPROM schema v.5 Program schema v.5
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
Address EEPROM option 50: 1420
Address EEPROM option 51: 1440
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Logging output to Telnet
Size of cmdtbl1: 29568
Size of cmdtbl2: 38616
Size of cmdtbl3: 26496
free RAM: 85599
Init One Wire bus...
numSensors: 0
PPS settings:
[... 209 lines cut ...]
  {CF_MAX_DEVICES,      0, CCAT_MAX,      CPI_TEXT,      CDT_MAXDEVICELIST,  CF_DEVICES_TXT, sizeof(max_device_list)}, //Need to call UpdateMaxDeviceList() before saving to EEPROM
  {CF_MAX_DEVADDR,      0, CCAT_MAX,      CPI_NOTHING,   CDT_VOID,           NULL, MAX_CUL_DEVICES * sizeof(uint32_t)},
  {CF_PPS_VALUES,       0, CCAT_GENERAL,  CPI_NOTHING,   CDT_VOID,           NULL, sizeof(pps_values)}, //printlnToDebug(PSTR("Reading EEPROM..."));  for (int i=PPS_TWS;i<=PPS_BRS;i++){ ...}
#ifdef CONFIG_IN_EEPROM
#ifdef WEBCONFIG
  {CF_IPWE,             2, CCAT_IPWE,     CPI_SWITCH,    CDT_BYTE,           CF_USE_TXT, sizeof(enable_ipwe)},//immediately apply
  {CF_IPWEVALUESLIST,   2, CCAT_IPWE,     CPI_TEXT,      CDT_PROGNRLIST,     CF_PROGLIST_TXT, sizeof(ipwe_parameters)},//immediately apply
  {CF_DEBUG,            2, CCAT_DEBUG,    CPI_DROPDOWN,  CDT_BYTE,           CF_USE_TXT, sizeof(debug_mode)},
  {CF_VERBOSE,          3, CCAT_DEBUG,    CPI_SWITCH,    CDT_BYTE,           CF_VERBOSE_TXT, sizeof(verbose)},
  {CF_MONITOR,          3, CCAT_DEBUG,    CPI_SWITCH,    CDT_BYTE,           CF_MONITOR_TXT, sizeof(monitor)}
#endif
#endif
};

typedef struct{
  uint16_t eeprom_address;//start address in EEPROM
  byte *option_address; //pointer to parameter variable
} addressesOfConfigOptions;
addressesOfConfigOptions options[CF_LAST_OPTION];`

[image]
[image]
[image]

Thank you

fredlcore, 2021-02-17:
So is logging to SD card working now or not?
Try /D instead of /DG to see the raw data. What does the file look like?

proasnet, 2021-02-18:
Not working.
I tried both commands /D and /DG but without solve problem. I tried more cards, not solved.
Here is config.h
`/************************************************************************************/
/************************************************************************************/
/* Settings -   BEGIN                                                               */
/************************************************************************************/
/************************************************************************************/

// Upon first installation, rename this file from BSB_lan_config.h.default to BSB_lan_config.h and adjust settings accordingly
// Users who still use Arduino Mega2560 boards instead of Arduin Due: Please have a look at the very end of this configuration
// where certain modules are disabled by default due to limited flash memory.

/* Select language; so far German is the most complete, with English following.
 * Available languages are: Czech (CS), German (DE), Danish (DA), English (EN), Spanish (ES), Finnish (FI),
 * French (FR), Greek (EL), Hungarian (HU), Italian (IT), Dutch (NL), Polish (PL), Russian (RU), Swedish (SV),
 * Slovenian (SI) and Turkish (TR).
 * Incomplete languages will automatically be filled up with English translations first, and if no English translation
 * is available, fallback will take place to German.
*/
#define LANG EN

/*
Allow to initialize program configuration by reading settings from EEPROM
byte UseEEPROM = 0; // Configuration is read from this config file.
                    // Configuration can be stored in EEPROM but will not used while UseEEPROM is zero.
                    // Set zero for fallback startup in case EEPROM configuration is broken.
byte UseEEPROM = 1; // Configuration will be read from EEPROM. This is the default.
*/
byte UseEEPROM = 1;

/*
 *  Enter a MAC address, found either on the EthernetShield or use the one below.
 *  Change this if you have more than one BSB-LAN adapter in your LAN, so that there aren't any address conflicts.
 *  MAC address here only affects the LAN shield, it doesn't apply to the WiFi-ESP-solution.
*/
byte mac[6] = { 0x00, 0x80, 0x41, 0x19, 0x69, 0x90 };

/*
 * Initialize the Ethernet server library
 * with the IP address and port you want to use
 * (port 80 is default for HTTP):
*/

uint16_t HTTPPort = 80;
bool useDHCP = true;                 // Set to false if you want to use a fixed IP.
byte ip_addr[4] = {192,168,178,88};     // Please note the commas instead of dots!!!  Set useDHCP to true if you want DHCP.
byte gateway_addr[4] = {192,168,178,1}; // Gateway address. This is usually your router's IP address. Please note the commas instead of dots!!! Ignored if first value is 0.
byte dns_addr[4] = {192,168,178,1};     // DNS server. Please note the commas instead of dots!!! Ignored if first value is 0.
byte subnet_addr[4] = {255,255,255,0};  // Subnet address. Please use commas instead of dots!!! Ignored if first value is 0.

//#define WIFI                          // Activate this definement if you want to use WiFi. Note: MAC address can't be set individually.
char wifi_ssid[32] = "YourWiFiNetwork"; // enter your WiFi network name (SSID) here
char wifi_pass[64] = "YourWiFiPassword";// enter your WiFi password here
#define WIFI_SPI_SS_PIN 12              // defines SPI-SS pin for Arduino-ESP8266 connection

#define MDNS_HOSTNAME "BSB-LAN"         // Advertises the hostname in the local network. Disable this if you don't want your device to be found under this name in your network.

#define DEBUG         // Compile with verbose DEBUG module if defined
byte debug_mode = 1;  // Debug mode: 0 - disabled, 1 - send debug messages to serial interface, 2 - send debug messages to telnet client
byte verbose = 1;     // If set to 1, all messages on the bus are printed to debug interface
byte monitor = 0;     // Bus monitor mode. This is only necessary for in-depth debug sessions.
bool show_unknown = true; // true - show all parameters, false - hide unknown parameters from web display (parameters will still be queried and therefore take time!)

/* SECURITY OPTIONS
 * There are several options to control and protect access to your heating system. However, keep
 * in mind, that even activating all three options are no guarantee that a versatile intruder with
 * access to your (W)LAN won't be able to gain access. In any case, no encryption of data streams
 * is provided from the Arduino itself. Use VPN or a SSL proxy if that is a must for you and connect
 * the Arduino wired to the VPN server or SSL proxy. On the other hand, someone with this amount
[... 245 lines cut ...]
// By default, the following modules are disabled for the Arduino Mega2560 due to lack of memory.
// This overwrites any definements above.
// You are of course free to make changes here so that they fit your needs and the Mega's memory constraints.
#if defined(__AVR__)
//#undef CONFIG_IN_EEPROM
//#undef WEBCONFIG
#undef JSONCONFIG
//#undef WEBSERVER
#undef AVERAGES
#undef DEBUG
#undef IPWE
#undef MQTT
#undef MDNS_HOSTNAME
#undef OFF_SITE_LOGGER
#undef ROOM_UNIT
#undef VERSION_CHECK
#undef MAX_CUL
#undef BME280
#endif
/************************************************************************************/
/************************************************************************************/
/* Settings -   END                                                                 */
/************************************************************************************/
/************************************************************************************/
`

fredlcore, 2021-02-18:
Config looks ok. The thing is that your card gets detected (it says 1907 MB free space in your first post), so I don't know what could be wrong as logging works here on my system. Could you try  again an older version (1.1 or 0.44) from releases and see if it works there?

1coderookie, 2021-02-18:
Looking at the config.h, you have these settings:
- logging to card: _deactivated_
- interval: 3600 seconds
- parameters: 8700, 8740, 8326, 8314, 710, 8310, 9005, 700

and the screenshot of your webconfig shows
- logging to card: activated
- interval: 60seconds
- parameters: only 8700

My suggestion: Switch off "Read config from EEPROM" in the webconfig, adjust the settings within the config.h to your needs, then flash again so that it reads the settings from the config.h and have a look if it works. Maybe something got tangled up there somewhere..
General note: Check before by querying the parameters, if your system really supports the parameters (e.g. the burner state and stages).

proasnet, 2021-02-18:
Okay. In webconfig, I had all parameters ( 8700, 8740, 8326, 8314, 710, 8310, 9005, 700), before some testings. No effect. Then I edit it to 8700 only for test. Then I made screenshot.
Today after some works I try your reqiements, about afternoon

fredlcore, 2021-02-18:
@1coderookie: The settings in _config.h are overridden if UseEEPROM is set to 1 and Reading from EEPROM in web-config is "on". Both is the case with proasnet, so it should work. But to be sure, it might help to try as you suggest.

1coderookie, 2021-02-18:
Yes, I know ;) 
But maybe he missed clicking on the submit button at one point or something got tangled up somewhere - so I thought I'll make that suggestion, just to make sure that everything gets written correctly in the EEPROM..

dukess, 2021-02-19:
@proasnet 
> ..ethernet shield WZ5100... Before this update, I had BSB-LAN on Arduino Mega with the same ethernet shield.

Can this shield be incompatible with Due?

Can you flash current version to Mega and try Logging?

proasnet, 2021-02-26:
Hi, today I have a time for experiments.
I do, what 1coderookie wrote. I tried reflash due with using config.h, logging to SD card with test interval 360s, nothing data. I tried two cards, FAT32 format, the first 512MB, the second 2GB. Used device is Baxi Luna Platinum with LMS15, with version of BSB lan under v.2 was logging work. Now not...
But If I thing, I have a Home Assistant on RPI with InfluxDB, but I am banana and I cant set it for using with BSB lan directly, I see that in config.h is some setting for use external logger.

proasnet, 2021-02-26:
Oh, what I see now ! I removed a card from a BSB Lan device, and I instert to a PC reader. On the card is 3 files. In datalog is data from today, but this data I not see in a graph of webUI and not see in Download data

1coderookie, 2021-02-26:
- Which three files do you see on the card? How are they named?
- What happens when you execute /D when the card with the datalog.txt file is plugged into the LAN shield's card reader?
- Do you have any script blocker (addon) active in your browser, like NoScript or something like that?
- When you tried the two SD cards, did you execute /D0 before starting to log?

dukess, 2021-02-26:
>  In datalog is data from today, but this data I not see in a graph of webUI and not see in Download data

Which errors was shown? (Error 404 or something else?)

fredlcore, 2021-02-28:
@proasnet: If you want us to help you, it would be helpful (and polite) if we didn't have to wait days or a week until we get (useful) feedback from you. Please get back to our questions, several of which are not answered yet. Otherwise I'll close this issue because this does not seem to be a problem that other users are facing.

proasnet, 2021-03-01:
I am sory for this waiting. In my Job, my boss send me to big industrial
automation project and I am only one programmer in our company. No free
time. If is possible, after done this we can write about it.

fredlcore, 2021-03-01:
Ok, sure, just wanted to make sure your're still onto this issue...

proasnet, 2021-03-01:
Yes I am still. But my bsb lan I have at home and is my private problem,
but in a my job I am very bussy this two week. If will be possible I return
to.
Thank you for understand.

proasnet, 2021-03-05:
Hi fredlcore, I am back now. I am sorry for long waiting, now I am at home.
Now I removed a card from LAN shield and I am put in a PC.
I see three files:
averages.txt
datalog.txt
journal.txt
All three files has date of edit 1.1.2000
[averages.txt](https://github.com/fredlcore/BSB-LAN/files/6090795/averages.txt)
[averages.txt](https://github.com/fredlcore/BSB-LAN/files/6090799/averages.txt)
[datalog.txt](https://github.com/fredlcore/BSB-LAN/files/6090801/datalog.txt)
 Here is code, from /D command and serial monitor:
```
GET /D HTTP/1.1

Duration: 1
POST /JQ?Parameter=700,710,711,712,714,730,900,8000,8740,8749 HTTP/1.1

LAN->HEIZ QUR  700 Heating circuit 1 - Operating mode: 
DC C2 00 0B 06 3D 2D 05 74 9C 4B 
HEIZ->LAN ANS  700 Heating circuit 1 - Operating mode: 3 - Komfort
DC 80 42 0D 07 2D 3D 05 74 00 03 87 10 
#700: 3 - Komfort
LAN->HEIZ QUR  710 Heating circuit 1 - Room temperature Comfort setpoint: 
DC C2 00 0B 06 3D 2D 05 8E D2 1E 
HEIZ->LAN ANS  710 Heating circuit 1 - Room temperature Comfort setpoint: 22.0 &deg;C
DC 80 42 0E 07 2D 3D 05 8E 00 05 80 10 E3 
#710: 22.0 &deg;C
LAN->HEIZ QUR  711 Heating circuit 1 - Komfortsollwert Maximum: 
DC C2 00 0B 06 3D 2D 05 A5 47 17 
HEIZ->LAN ANS  711 Heating circuit 1 - Komfortsollwert Maximum: 24.0 &deg;C
DC 80 42 0E 07 2D 3D 05 A5 00 06 00 FD 69 
#711: 24.0 &deg;C
LAN->HEIZ QUR  712 Heating circuit 1 - Room temp reduced setpoint: 
DC C2 00 0B 06 3D 2D 05 90 21 E1 
HEIZ->LAN ANS  712 Heating circuit 1 - Room temp reduced setpoint: 18.0 &deg;C
DC 80 42 0E 07 2D 3D 05 90 00 04 80 9A 2F 
#712: 18.0 &deg;C
LAN->HEIZ QUR  714 Heating circuit 1 - Room temp frost protection setpoint: 
DC C2 00 0B 06 3D 2D 05 92 01 A3 
HEIZ->LAN ANS  714 Heating circuit 1 - Room temp frost protection setpoint: 6.0 &deg;C
DC 80 42 0E 07 2D 3D 05 92 00 01 80 88 B2 
#714: 6.0 &deg;C
LAN->HEIZ QUR  730 Heating circuit 1 - Summer/winter changeover temp: 
DC C2 00 0B 06 3D 2D 05 FD 9C EA 
HEIZ->LAN ANS  730 Heating circuit 1 - Summer/winter changeover temp: 18.0 &deg;C
DC 80 42 0E 07 2D 3D 05 FD 00 04 80 FA 7B 
#730: 18.0 &deg;C
LAN->HEIZ QUR  900 Heating circuit 1 - Operating mode: 
DC C2 00 0B 06 3D 05 07 BE AD 48 
HEIZ->LAN ANS  900 Heating circuit 1 - Operating mode: 4 - Automatik
DC 80 42 0D 07 05 3D 07 BE 00 04 C3 E3 
#900: 4 - Automatik
LAN->HEIZ QUR 8000 Status - Status heating circuit 1: 
DC C2 00 0B 06 3D 05 07 A3 6E D4 
HEIZ->LAN ANS 8000 Status - Status heating circuit 1: 114 - Heizbetrieb Komfort
DC 80 42 0D 07 05 3D 07 A3 00 72 DC 80 
#8000: 114 - Heizbetrieb Komfort
LAN->HEIZ QUR 8740 Diagnosis consumer - Room temp 1 actual value: 
DC C2 00 0B 06 3D 2D 05 1E 51 A7 
HEIZ->LAN ANS 8740 Diagnosis consumer - Room temp 1 actual value: 22.2 &deg;C
DC 80 42 0E 07 2D 3D 05 1E 00 05 8E 37 B2 
#8740: 22.2 &deg;C
LAN->HEIZ QUR 8749 Diagnosis consumer - Raumthermostat 1: 
DC C2 00 0B 06 3D 05 0C 7D 98 9D 
HEIZ->LAN ANS 8749 Diagnosis consumer - Raumthermostat 1: 0 - Kein Bedarf
DC 80 42 0D 07 05 3D 0C 7D 00 00 E2 DF 
#8749: 0 - Kein Bedarf
```
In the browser I have not any blocker. I not see any data on any PC connected to BSB_LAN.
After putting a cards to reader, I try /D0 on all cards before testing.

fredlcore, 2021-03-05:
Ok, please send the logs from right where the Arduino starts (start the logging and press the reset button). I will say something about starting the SD card and the free space.
Also, can you confirm that the data in the SD card files is correct? So if you log every x seconds, you also have entries in datalog.txt every x seconds?

fredlcore, 2021-03-05:
Also, can you check that after issuing /D0, the data logs are reset (deleted and created anew)?

proasnet, 2021-03-05:
This is serial monitor from reset and boot:
```
RGT1->Hin Bedarf
S 8749 Diagnosis consumer - Raumthermostat 1: 0 - Kein Bedarf
DC 80 42 0D 07 05 3D 0C 7D 00 00 E2 DF 
#8749: 0 - Kein Bedarf
READY
Using settings from config file
Reading done.
EEPROM schema v.0 Program schema v.5
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
Address EEPROM option 50: 1420
Address EEPROM option 51: 1440
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Size of cmdtbl1: 29568
Size of cmdtbl2: 38616
Size of cmdtbl3: 26496
free RAM: 85595
[... 126 lines cut ...]
DC 80 42 0E 07 2D 3D 05 FD 00 04 80 FA 7B 
#730: 18.0 &deg;C
LAN->HEIZ QUR  900 Heating circuit 1 - Operating mode: 
DC C2 00 0B 06 3D 05 07 BE AD 48 
HEIZ->LAN ANS  900 Heating circuit 1 - Operating mode: 4 - Automatik
DC 80 42 0D 07 05 3D 07 BE 00 04 C3 E3 
#900: 4 - Automatik
LAN->HEIZ QUR 8000 Status - Status heating circuit 1: 
DC C2 00 0B 06 3D 05 07 A3 6E D4 
HEIZ->LAN ANS 8000 Status - Status heating circuit 1: 114 - Heizbetrieb Komfort
DC 80 42 0D 07 05 3D 07 A3 00 72 DC 80 
#8000: 114 - Heizbetrieb Komfort
LAN->HEIZ QUR 8740 Diagnosis consumer - Room temp 1 actual value: 
DC C2 00 0B 06 3D 2D 05 1E 51 A7 
HEIZ->LAN ANS 8740 Diagnosis consumer - Room temp 1 actual value: 22.4 &deg;C
DC 80 42 0E 07 2D 3D 05 1E 00 05 97 B4 AA 
#8740: 22.4 &deg;C
LAN->HEIZ QUR 8749 Diagnosis consumer - Raumthermostat 1: 
DC C2 00 0B 06 3D 05 0C 7D 98 9D 
HEIZ->LAN ANS 8749 Diagnosis consumer - Raumthermostat 1: 0 - Kein Bedarf
DC 80 42 0D 07 05 3D 0C 7D 00 00 E2 DF 
#8749: 0 - Kein Bedarf
HEIZ->EM1 INF      0504024E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
DC 80 03 1B 02 05 04 02 4E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 1B 
```

fredlcore, 2021-03-05:
Ok, so everything looks good so far, the SD card is detected, it says that there is 1.9GB free space left and it seems that the files are created correctly. Can you please check and call /D0 and see (on the PC) if the files are created correctly and post the content of datalog.txt here?

Then, could you please search for the line 
`  while (chars_read == logbuflen) {`
in BSB_LAN.ino (around line 6245)
and put _above_ it these lines:
```
Serial.print("Chars read: ");
Serial.println(chars_read);
Serial.print("logbuflen: ");
Serial.println(logbuflen);
Serial.print("bigBuff: ");
Serial.println(bigBuff);
```
And then flash it and call /D again and post the output of the serial log here.

proasnet, 2021-03-05:
datalog.txt file I attached in my post before your. Now I am going reflash it. Please wait.

proasnet, 2021-03-05:
Here is serial output after insert code and reflash, restart:
```
READY
Using settings from config file
Reading done.
EEPROM schema v.0 Program schema v.5
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
Address EEPROM option 50: 1420
Address EEPROM option 51: 1440
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Size of cmdtbl1: 29568
Size of cmdtbl2: 38616
Size of cmdtbl3: 26496
free RAM: 85595
Init One Wire bus...
numSensors: 0
PPS settings:
Starting SD..ok
192.168.0.150
255.255.255.0
192.168.0.254
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...1907 MB free
Start network services
Device family: 123
Device variant: 231

Setup complete
Connect to MQTT broker, updating will topic
Will topic: BSB-LAN/status
HEIZ->EM1 INF      05040227 00 00 00 00 
DC 80 03 0F 02 05 04 02 27 00 00 00 00 D8 E6 
HEIZ->ALL  INF 10103 User defined - INFO HK2 - TBD: 00 00 FF FF FF FF FF FF 00 00 
DC 80 7F 15 02 2E 00 02 11 00 00 FF FF FF FF FF FF 00 00 F8 F1 
HEIZ->EM1 INF      0504024E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
DC 80 03 1B 02 05 04 02 4E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 1B 
Connect to MQTT broker, updating will topic
Will topic: BSB-LAN/status
GET /D HTTP/1.1

Chars read: -1
logbuflen: 1024
bigBuff: Milliseconds;Date;Parameter;Description;Value;Unit
361578;26.02.2021 07:10:35;8700;Outside temp sensor local;-1.3;&deg;C
361769;26.02.2021 07:10:35;8740;Room temp 1 actual value;21.7;&deg;C
361962;26.02.2021 07:10:35;8326;Relative output;30;&#037;
362156;26.02.2021 07:10:35;8314;Return temp actual value;49.5;&deg;C
362373;26.02.2021 07:10:36;710;Room temperature Comfort setpoint;22.0;&deg;C
362594;26.02.2021 07:10:36;8310;Boiler temp actual value;58.1;&deg;C
362762;26.02.2021 07:10:36;9005;Wasserdruck H1;1.5;bar
362970;26.02.2021 07:10:36;700;Operating mode;1;
723200;26.02.2021 07:16:36;8700;Outside temp sensor local;-1.3;&deg;C
723389;26.02.2021 07:16:37;8740;Room temp 1 actual value;21.6;&deg;C
723597;26.02.2021 07:16:37;8326;Relative output;35;&#037;
723774;26.02.2021 07:16:37;8314;Return temp actual value;49.2;&deg;C
723965;26.02.2021 07:16:37;710;Room temperature Comfort setpoint;22.0;&deg;C
724182;26.02.2021 07:16:37;8310;Boiler temp actual value;58.4;&deg;C
724398;26.02.2021 07:16:38;900
Duration: 82
RGT1->HEIZ INF      3D2D0215 05 98 00 
DC 86 00 0E 02 3D 2D 02 15 05 98 00 83 5D 
```

proasnet, 2021-03-05:
And then I use /D0 and /D, serial:
```
READY
Using settings from config file
Reading done.
EEPROM schema v.0 Program schema v.5
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
Address EEPROM option 50: 1420
Address EEPROM option 51: 1440
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Size of cmdtbl1: 29568
Size of cmdtbl2: 38616
Size of cmdtbl3: 26496
free RAM: 85595
Init One Wire bus...
numSensors: 0
PPS settings:
Starting SD..ok
192.168.0.150
255.255.255.0
192.168.0.254
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...1907 MB free
Start network services
Device family: 123
Device variant: 231

Setup complete
Connect to MQTT broker, updating will topic
Will topic: BSB-LAN/status
HEIZ->EM1 INF      0504024E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
DC 80 03 1B 02 05 04 02 4E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 1B 
Connect to MQTT broker, updating will topic
Will topic: BSB-LAN/status
RGT1->HEIZ INF      3D2D0215 05 89 00 
DC 86 00 0E 02 3D 2D 02 15 05 89 00 B3 1F 
GET /D HTTP/1.1

Chars read: 52
logbuflen: 1024
bigBuff: Milliseconds;Date;Parameter;Description;Value;Unit
tf-8
Cache-Control: max-age=60, public

Duration: 2
HEIZ->ALL  INF      05000219 00 4A 00 0A 5A 7E FF FF FF FF 00 00 
DC 80 7F 17 02 05 00 02 19 00 4A 00 0A 5A 7E FF FF FF FF 00 00 30 D5 
HEIZ->EM1 INF      05040227 00 00 00 00 
DC 80 03 0F 02 05 04 02 27 00 00 00 00 D8 E6 
```

fredlcore, 2021-03-05:
Ok, this looks good: As you can see, bigBuff has the contents of your datalog.txt. Can you have a look at the source of the HTML page when you open /D? Is it empty or what does it look like? If there is any content, please post it here.
@dukess: What is strange to me is that the initial `dataFile.read(bigBuff, logbuflen)` returns `-1`, but bigBuff still contains the beginning of the logfile. If chars_read is less than 1, then transmitFile won't output the contents of the file to the browser. So that would explain the problem. But after resetting the file with /D0, chars_read is 52, so it should send bigBuff to the browser. Any ideas?

proasnet, 2021-03-05:
If I use IP/D, i see an empty page now.

proasnet, 2021-03-05:
I have a teamviewer if will be easy. Connected BSB. But login I can send via email. May be?

dukess, 2021-03-05:
Chars read: -1 - file not exist?

fredlcore, 2021-03-05:
That's what I thought as well, but if you look at the serial output, "bigBuff" clearly contains the beginning of the datalog.txt file...

fredlcore, 2021-03-05:
@proasnet: Could you open a port for both the webinterface and the telnet client and select telnet for debug output? That wold probably be easier.

proasnet, 2021-03-06:
If I change in WebUI config from EEPROM, save
Then debug from serial to telnet, save
a webUI dont work and telnet dont work...
Then I must reset it by button on shield and then WebUI working again.
Few days ago, I am tried telnet and worked it. I dont know why dont work today. From yesterday is BSB reflashed.
So I am sending you adresses, I am still here.

dukess, 2021-03-06:
You need to enable EEPROM using in WebUI too (option on top of list).

dukess, 2021-03-06:
@fredlcore 
```
bigBuff: Milliseconds;Date;Parameter;Description;Value;Unit
tf-8
Cache-Control: max-age=60, public
```
looks like string was copied without zero.

fredlcore, 2021-03-06:
Yes, now that you mention it - interesting! Why would this line come without a \0? `chars_read` seems to include it, however, with 52 counts.

proasnet, 2021-03-06:
> 
> 
> You need to enable EEPROM using in WebUI too (option on top of list).

I wrote upward, that I have selected in config.h for use config from file, but in the webUI I tried select config from EEPROM, this is working. But, than I select debug from serial to telnet, both crash and only one way is reset by button.
After reset by button, option read config from EEPROM is selected back to file and debug to serial. BSB not member settings from eeprom.

fredlcore, 2021-03-06:
Ok, that's strange, I just got onto your webinterface and made the changes (before I read about the crash problem), and indeed, now I can't access the webinterface anymore. We need to investigate this crash separately and for that we would need the serial logs right before and after you make these changes that lead to the crash.
Can you reset the board remotely or do you have to do it when you are back home?

proasnet, 2021-03-06:
I am at home. I can reset now, with serial protocol, wait a moment please

proasnet, 2021-03-06:
Reseted with this debug:
```
READY
Using settings from config file
Reading done.
EEPROM schema v.0 Program schema v.5
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
Address EEPROM option 50: 1420
Address EEPROM option 51: 1440
Address EEPROM option 52: 1460
Address EEPROM option 53: 1461
Address EEPROM option 54: 1462
Address EEPROM option 55: 1463
Size of cmdtbl1: 29568
Size of cmdtbl2: 38616
Size of cmdtbl3: 26496
free RAM: 85595
Init One Wire bus...
numSensors: 0
PPS settings:
Starting SD..ok
192.168.0.150
255.255.255.0
192.168.0.254
Waiting 3 seconds to give Ethernet shield time to get ready...
Calculating free space on SD...1907 MB free
Start network services
Device family: 123
Device variant: 231

Setup complete
Connect to MQTT broker, updating will topic
Will topic: BSB-LAN/status
HEIZ->EM1 INF      0504024E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
DC 80 03 1B 02 05 04 02 4E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 1B 
POST /JQ?Parameter=700,710,711,712,714,730,900,8000,8740,8749 HTTP/1.1

LAN->HEIZ QUR  700 Heating circuit 1 - Operating mode: 
DC C2 00 0B 06 3D 2D 05 74 9C 4B 
HEIZ->LAN ANS  700 Heating circuit 1 - Operating mode: 1 - Automatik
DC 80 42 0D 07 2D 3D 05 74 00 01 A7 52 
#700: 1 - Automatik
LAN->HEIZ QUR  710 Heating circuit 1 - Room temperature Comfort setpoint: 
DC C2 00 0B 06 3D 2D 05 8E D2 1E 
HEIZ->LAN ANS  710 Heating circuit 1 - Room temperature Comfort setpoint: 22.0 &deg;C
DC 80 42 0E 07 2D 3D 05 8E 00 05 80 10 E3 
#710: 22.0 &deg;C
LAN->HEIZ QUR  711 Heating circuit 1 - Komfortsollwert Maximum: 
DC C2 00 0B 06 3D 2D 05 A5 47 17 
HEIZ->LAN ANS  711 Heating circuit 1 - Komfortsollwert Maximum: 24.0 &deg;C
DC 80 42 0E 07 2D 3D 05 A5 00 06 00 FD 69 
#711: 24.0 &deg;C
LAN->HEIZ QUR  712 Heating circuit 1 - Room temp reduced setpoint: 
DC C2 00 0B 06 3D 2D 05 90 21 E1 
HEIZ->LAN ANS  712 Heating circuit 1 - Room temp reduced setpoint: 18.0 &deg;C
DC 80 42 0E 07 2D 3D 05 90 00 04 80 9A 2F 
#712: 18.0 &deg;C
LAN->HEIZ QUR  714 Heating circuit 1 - Room temp frost protection setpoint: 
DC C2 00 0B 06 3D 2D 05 92 01 A3 
HEIZ->LAN ANS  714 Heating circuit 1 - Room temp frost protection setpoint: 6.0 &deg;C
DC 80 42 0E 07 2D 3D 05 92 00 01 80 88 B2 
#714: 6.0 &deg;C
LAN->HEIZ QUR  730 Heating circuit 1 - Summer/winter changeover temp: 
DC C2 00 0B 06 3D 2D 05 FD 9C EA 
HEIZ->LAN ANS  730 Heating circuit 1 - Summer/winter changeover temp: 18.0 &deg;C
DC 80 42 0E 07 2D 3D 05 FD 00 04 80 FA 7B 
#730: 18.0 &deg;C
LAN->HEIZ QUR  900 Heating circuit 1 - Operating mode: 
DC C2 00 0B 06 3D 05 07 BE AD 48 
HEIZ->LAN ANS  900 Heating circuit 1 - Operating mode: 4 - Automatik
DC 80 42 0D 07 05 3D 07 BE 00 04 C3 E3 
#900: 4 - Automatik
LAN->HEIZ QUR 8000 Status - Status heating circuit 1: 
DC C2 00 0B 06 3D 05 07 A3 6E D4 
HEIZ->LAN ANS 8000 Status - Status heating circuit 1: 114 - Heizbetrieb Komfort
DC 80 42 0D 07 05 3D 07 A3 00 72 DC 80 
#8000: 114 - Heizbetrieb Komfort
LAN->HEIZ QUR 8740 Diagnosis consumer - Room temp 1 actual value: 
DC C2 00 0B 06 3D 2D 05 1E 51 A7 
HEIZ->LAN ANS 8740 Diagnosis consumer - Room temp 1 actual value: 22.1 &deg;C
DC 80 42 0E 07 2D 3D 05 1E 00 05 87 A6 9B 
#8740: 22.1 &deg;C
LAN->HEIZ QUR 8749 Diagnosis consumer - Raumthermostat 1: 
DC C2 00 0B 06 3D 05 0C 7D 98 9D 
HEIZ->LAN ANS 8749 Diagnosis consumer - Raumthermostat 1: 0 - Kein Bedarf
DC 80 42 0D 07 05 3D 0C 7D 00 00 E2 DF 
#8749: 0 - Kein Bedarf
```

proasnet, 2021-03-06:
I can set reading from eeprom in config.h and reflash if help this.

fredlcore, 2021-03-06:
Ok, we've just had a TeamViewer session and I could see that everything works fine until the log file is 1024 bytes long. I then increased the size of bigBuff to 4096 to see if the limit to 1024 bytes was the problem, but it was not. 
I then realized that dataFile.read() returns -1 only after the file is larger than 1024 bytes. So I assume that the problem is with the file system. @proasnet said that the card has been formatted with FAT32 and that it was working with version 1.0. I'm not sure, @dukess, whether we changed the SD card library sometime in between and that this may be the reason? But it would be strange if a newer version would no longer support FAT32 if it was working before. Since he tried two cards, I also don't think that it's a mechanical error.

He will check and see if the card is still working with version 1.0 and afterwards reformat with FAT16 to see if that makes a change.

proasnet, 2021-03-06:
I reflash to saved older version without any changes, v0.44 and logging working. You can see here
[image]

proasnet, 2021-03-06:
Than I return to v2, updated with you and logging not working, screenshot after one minute from start with FAT16 card formated. As you see here
[image]

fredlcore, 2021-03-06:
Well, it does display the log file header - that's more than before. Have you set logging parameters to 5 seconds / parameter 8700 to get the log file filled quickly? Also use CTRL+R to reload the datalog page.
The problem only occurred before if the log file was larger than 1024 bytes, so please wait until it's filled with a few more entries.

fredlcore, 2021-03-06:
One more thing you can try please:
Remove the folder /src/SdFat from the BSB-LAN v2.0 directory and replace it with the src/SdFat folder from 0.44 version. We changed the SD library at some point, so maybe the new one has some issues.

fredlcore, 2021-03-06:
Oh, and one more thing first before changing the library:
Please open /src/SdFat/FatLib/FatVolume.h and search for line 
#define DEBUG_MODE 0
and change it from 0 to 1 and flash again and have serial monitor running.
Then wait until the log file gets large enough so that it no longer shows in the browser. Then there should be an error message in the serial monitor.

dukess, 2021-03-06:
> Yes, now that you mention it - interesting! Why would this line come without a \0? `chars_read` seems to include it, however, with 52 counts.

It can be "bug" or we can ignore it:
In transmitFile() we read and write to client defined number of bytes, so here no problem. But when Serial.println(bigBuff); called it print string until first zero.

fredlcore, 2021-03-06:
Thanks, yes, I figured out that this is no problem because of the defined number of bytes. The problem lies elsewhere...

dukess, 2021-03-06:
>whether we changed the SD card library sometime in between and that this may be the reason? But it would be strange if a newer version would no longer support FAT32 if it was working before. 

I have no problem with FAT32 (8 Gb on Mega, 16 on Due).

dukess, 2021-03-06:
Idea:
// change SPI_DIV3_SPEED to SPI_HALF_SPEED if you are still having problems getting your SD card detected

proasnet, 2021-03-06:
Now after refresh site with /D is page empty again.
I am doing what you wrote.

proasnet, 2021-03-06:
With replace a library logging working. Now I am waiting until the log file gets large enough

proasnet, 2021-03-06:
Now, page /D is empty.
Here is serial log:
```
11: 495
Address EEPROM option 12: 499
Address Temp\arduino_build_61045\sketch\src\SdFat\FatLib\FatFileLFN.cpp401
Device family: 123
Device variant: 231

Setup complete
\Temp\arduino_build_61045\sketch\src\SdFat\FatLib\FatFileLFN.cpp401
Device family: 123
Device variant: 231

Setup complete
READY
EEPROM not ready
Using settings from config file
Reading done.
EEPROM schema v.0 Program schema v.5
Address EEPROM option 0: 0
Address EEPROM option 1: 1
Address EEPROM option 2: 2
Address EEPROM option 3: 6
Address EEPROM option 4: 226
Address EEPROM option 5: 306
Address EEPROM option 6: 490
Address EEPROM option 7: 491
Address EEPROM option 8: 492
Address EEPROM option 9: 493
Address EEPROM option 10: 494
Address EEPROM option 11: 495
Address EEPROM option 12: 499
Address EEPROM option 13: 500
Address EEPROM option 14: 660
Address EEPROM option 15: 661
Address EEPROM option 16: 665
Address EEPROM option 17: 825
Address EEPROM option 18: 831
Address EEPROM option 19: 832
Address EEPROM option 20: 836
Address EEPROM option 21: 840
Address EEPROM option 22: 844
Address EEPROM option 23: 848
Address EEPROM option 24: 850
Address EEPROM option 25: 854
Address EEPROM option 26: 858
Address EEPROM option 27: 922
Address EEPROM option 28: 986
Address EEPROM option 29: 987
Address EEPROM option 30: 988
Address EEPROM option 31: 998
Address EEPROM option 32: 999
Address EEPROM option 33: 1159
Address EEPROM option 34: 1160
Address EEPROM option 35: 1164
Address EEPROM option 36: 1165
Address EEPROM option 37: 1166
Address EEPROM option 38: 1167
Address EEPROM option 39: 1171
Address EEPROM option 40: 1203
Address EEPROM option 41: 1235
Address EEPROM option 42: 1267
Address EEPROM option 43: 1299
Address EEPROM option 44: 1301
Address EEPROM option 45: 1302
Address EEPROM option 46: 1303
Address EEPROM option 47: 1304
Address EEPROM option 48: 1336
Address EEPROM option 49: 1400
[... 1695 lines cut ...]
DC C2 00 0B 06 3D 2D 05 FD 9C EA 
HEIZ->LAN ANS  730 Heating circuit 1 - Summer/winter changeover temp: 18.0 &deg;C
DC 80 42 0E 07 2D 3D 05 FD 00 04 80 FA 7B 
#730: 18.0 &deg;C
LAN->HEIZ QUR  900 Heating circuit 1 - Operating mode: 
DC C2 00 0B 06 3D 05 07 BE AD 48 
HEIZ->LAN ANS  900 Heating circuit 1 - Operating mode: 4 - Automatik
DC 80 42 0D 07 05 3D 07 BE 00 04 C3 E3 
#900: 4 - Automatik
LAN->HEIZ QUR 8000 Status - Status heating circuit 1: 
DC C2 00 0B 06 3D 05 07 A3 6E D4 
HEIZ->LAN ANS 8000 Status - Status heating circuit 1: 122 - Raumtemperaturbegrenzung
DC 80 42 0D 07 05 3D 07 A3 00 7A 5D 88 
#8000: 122 - Raumtemperaturbegrenzung
LAN->HEIZ QUR 8740 Diagnosis consumer - Room temp 1 actual value: 
DC C2 00 0B 06 3D 2D 05 1E 51 A7 
HEIZ->LAN ANS 8740 Diagnosis consumer - Room temp 1 actual value: 22.2 &deg;C
DC 80 42 0E 07 2D 3D 05 1E 00 05 8D 07 D1 
#8740: 22.2 &deg;C
LAN->HEIZ QUR 8749 Diagnosis consumer - Raumthermostat 1: 
DC C2 00 0B 06 3D 05 0C 7D 98 9D 
HEIZ->LAN ANS 8749 Diagnosis consumer - Raumthermostat 1: 0 - Kein Bedarf
DC 80 42 0D 07 05 3D 0C 7D 00 00 E2 DF 
#8749: 0 - Kein Bedarf
```

proasnet, 2021-03-06:
About 10 record was displayed to moment, when /D page was empty. Then after /D0 logging again

fredlcore, 2021-03-06:
Ok, thanks - unfortunately the line numbers here don't match; can you please post three lines before and three lines after line number 786 in file \src\SdFat\FatLib\FatFile.cpp?
Also, do you have a larger SD card available? Maybe the new library is optimized for SDHC cards (larger than 4GB)? That would help narrow down the error.

@dukess: It looks like it's a FAT cluster issue. Line 775 (at least here) says
`fg = m_vol->fatGet(m_curCluster, &m_curCluster);`
and it seems that fatGet returns zero which is why 784 throws an error. The question is what changed between the old SdFat library version (which apparently does not have this problem) and the one we're using now...

fredlcore, 2021-03-06:
@proasnet: There is a newer version of SdFat available on the authors website. Could you please try the following:
- Download Version 2 from SdFat from here: https://github.com/greiman/SdFat/archive/master.zip
- Remove the _contents_ from /src/SdFat folder in BSB-LAN (i.e. leave the folder name "SdFat" in place)
- Move the _contents_ from the new library's folder /src into the /src/SdFat folder in BSB-LAN
Then compile again and see if the problem goes away.

proasnet, 2021-03-06:
I must now go out with family. I am back after about 3 hours.

dukess, 2021-03-06:
V2 of SdFat can't be compiled in without efforts: need to rwrite some code. So i can advise upgrade to 1.1.4 at this moment.

fredlcore, 2021-03-06:
True. Was there a specific reason why you upgraded SdFat to 1.1.4 back then? I remember we talked about this, but I forgot why (other than having an updated version is always good :) ). If @proasnet says that the older version works, it may be best to revert back to that one for now...

[26 more replies]

closed 2021-03-06 by fredlcore