fredlcore/BSB-LAN #337: /JQ crashes on ESP32

fredlcore, 2021-02-16:
When querying parameters via /JQ=x,x,x the ESP32 hangs indefinitively after having received all parameters from the heating system. This is an example output which looks completely normal, except that afterwards there is no reacation whatsoever from the ESP32:

```
GET /4444/JQ=8000,700,8740,8741,710 HTTP/1.1                                    
LAN->HEIZ QUR 8000 Status - Status Heizkreis 1:                                 
DC C2 00 0B 06 3D 05 07 A3 6E D4                                                
HEIZ->LAN ANS 8000 Status - Status Heizkreis 1: 114 - Heizbetrieb Komfort       
DC 80 42 0D 07 05 3D 07 A3 00 72 DC 80                                          
#8000: 114 - Heizbetrieb Komfort                                                
LAN->HEIZ QUR  700 Heizkreis 1 - Betriebsart:                                   
DC C2 00 0B 06 3D 2D 05 74 9C 4B                                                
HEIZ->LAN ANS  700 Heizkreis 1 - Betriebsart: 1 - Automatik                     
DC 80 42 0D 07 2D 3D 05 74 00 01 A7 52                                          
#700: 1 - Automatik                                                             
LAN->HEIZ QUR 8740 Diagnose Verbraucher - Raumtemperatur 1:                     
DC C2 00 0B 06 3D 2D 05 1E 51 A7                                                
HEIZ->LAN ANS 8740 Diagnose Verbraucher - Raumtemperatur 1: --- &deg;C          
DC 80 42 0E 07 2D 3D 05 1E 01 0C 80 5B D4                                       
#8740: --- &deg;C                                                               
LAN->HEIZ QUR 8741 Diagnose Verbraucher - Raumsollwert 1:                       
DC C2 00 0B 06 3D 2D 05 93 11 82                                                
HEIZ->LAN ANS 8741 Diagnose Verbraucher - Raumsollwert 1: 21.0 &deg;C           
DC 80 42 0E 07 2D 3D 05 93 00 05 40 EB 8E                                       
#8741: 21.0 &deg;C                                                              
LAN->HEIZ QUR  710 Heizkreis 1 - Komfortsollwert:                               
DC C2 00 0B 06 3D 2D 05 8E D2 1E                                                
HEIZ->LAN ANS  710 Heizkreis 1 - Komfortsollwert: 21.0 &deg;C                   
DC 80 42 0E 07 2D 3D 05 8E 00 05 40 C9 AF                                       
#710: 21.0 &deg;C             
```

fredlcore, 2021-02-16:
Some more findings: It's not limited to /JQ but it almost certainly happens after a /JQ. Note the `G` in the last line, this is the beginning of the debug output of the next `GET` request. The code is executed until this block:
```
        if ((c!='\n') && (c!='\r') && (bPlaceInBuffer<MaxArrayElement)) {
          cLineBuffer[bPlaceInBuffer++]=c;
          continue;
        }
```
Debug messages right before `continued` are still printed, but then the code gets stuck. I have added an ESP32 watchdog and it fires after the defined amount of time.
It's not an index-out-of-bound kind of error as bPlaceInBuffer is zero at the time of the first character coming in.

fredlcore, 2021-02-16:
It seems to be a problem when multiple http requests reach the ESP32 at the same time. I tested this on the Olimex EVB as well as the NodeMCU ESP32, with the former both LAN and WiFi.
I have several instances in FHEM that poll/send data from/to BSB-LAN, in addition to a homebridge plugin that does the same. There is no problem when I open two browser windows and call a long /JQ URL, they seem to be processed sequentially somehow, at least most of the time. But at least now that I turned off all requests from Homebridge and FHEM except two that run in a safe distance to each other, the watchdog hasn't kicked in for more than 20 minutes now (before it hardly went longer than a minute or two).

Since there is data coming in (as can be seen in the `G`and sometimes `GE` from the GET-Request as the last line before the system freezes), I would assume that either the test for `client.connected()` or for `client.available()` is stuck in some kind of endless loop. However, when I look at these functions, they are pretty short with no loops whatsoever. So if anybody has an idea how to approach this, I'd be more than glad to hear it!

fredlcore, 2021-02-16:
Ok, turning on and off Homebridge which sends at least two asynchronous queries to BSB-LAN every 30 seconds or so seems to be the culprit. I remember that this also lead to a problem with the Due, but it was Homebridge which was getting timeouts then. The Arduino in its basic single-tasking approach probably couldn't be bothered. Now the question is whether this matter can be dealt with somehow, because even though BSB-LAN is not the target of any heavy load in a home automation system, it still could happen once in a while that concurrent requests lock the ESP32 out. A watchdog is just a dirty hack to prevent further damage. Any ideas, @dukess, @do13 or anybody else?

dukess, 2021-02-17:
I think we should to add new flag (or use exist variable for this purpose) which will be set after client = server->available(); and cleared after client.stop(); for prevention of execution client = server->available(); if flag is set.
Ex:
```
if(client == NULL){
client = server->available();
}
....
client.stop();
client = NULL;
```

fredlcore, 2021-02-17:
Nice idea! Unfortunateley, it doesn't work here somehow, although it really makes sense. 
When using WiFi, I often get this error right error before it hangs:
`[E][WiFiClient.cpp:392] write(): fail on fd 60, errno: 104, "Connection reset by peer"`
Not always, but if this error comes, it will hang afterwards. So maybe the problem is one level higher/lower?

dukess, 2021-02-17:
Try to insert delay(1000); after client.stop();
https://github.com/gilmaimon/ArduinoWebsockets/issues/46

fredlcore, 2021-02-17:
Your first idea worked in the end, I just had to put the condition in a different place, thanks a lot!
Now I also figured out what the problem is: If there are two concurrent requests with a rather short timeout and the remote host closes the connection, this problem occurs. Because if I make three parallel requests in Firefox with rather long /JQ requests, all three requests are served without a problem one after the other.

I haven't figured out where exactly the problem is on the ESP32's side, but this workaround works and I hope it also does not interfere with the Due.

Thanks again!

closed 2021-02-17 by fredlcore