fredlcore/BSB-LAN #339: Multiple temperature units on the BSB/LPB bus?

sebi5361, 2021-02-19:
Dear Fred,
Thanks for this great project, it is exactly what I was looking for!

2 questions:
- Is it possible to connect several (two) temperature units (Atlantic T55) on that BSB/LPB bus?
- Do you still have a PCB for your project I could buy?

Sébastien

1coderookie, 2021-02-19:
> * Is it possible to connect several (two) temperature units (Atlantic T55) on that BSB/LPB bus?

Do you mean
1. connecting the BSB-LAN adapter AND the T55 (=QAA55) room unit to the BSB (room units are not being connected to the LPB!) at the same time or
2. connecting two T55 (=QAA55) room units? 

If you mean no. 1, then the answer is yes. 
Check out the manual: https://1coderookie.github.io/BSB-LPB-LAN_EN/chap02.html#23-connecting-the-adapter-to-the-controller
```
In cases where only one BSB port is available at the controller (e.g. RVS21 controller within heat pumps) you can connect the adapter parallel to an already installed room unit.
```
Which heating system do you have?

sebi5361, 2021-02-19:
I have an atlantic alféa extensa duo.
I don't know exactly what I want to do with it yet.
I would like to keep the existing installation as it is + add the possibility to control it remotely (homeassistant, google home...).
- One possibility is to get nest thermostat and connect it to the heating system via a modified t55.
(There is a trick I found on the internet to replace the thermistance by a 15k resistor, a 10k resistor respectively, to let the t55 believe it is 16.3°C, 24.6°C respectively.)
But I would like to make sure the two t55 (the original + the modified) do no interfere.
I don't have room units.
- I am also thinking of building your device as it will be definitely cheaper and will give me more control.
- Another possibility is to build an equivalent of this modified t55 based on your project, to send on/off commands to the heating system based on nest on/off inputs.

fredlcore, 2021-02-19:
Hi Sébastien,

yes, I still have PCBs available, just send me an e-mail to bsb (ät) code-it.de

As for your other question, two T55 don't make sense because both will send their local room temperature to the heater, so the heater will get confused as it can only process one room temperature. 
What you could do is to remove the T55 and use BSB-LAN to tell the heater the temperature. This is done via parameter 10000 and is explained in the manual. You could use a home automation system to collect the temperature from various rooms and then send a weighted average temperature to the heater for example. In that case, I don't think you would need a Nest thermostat. I, for example, use MAX! thermostats which send the local room temperature and collect these from several rooms and send them to the heater. You could use Shellys or Homematic thermostats if you have a home auotmation system that can poll their temperature. For the Shellys, you could even do this with a little Linux shell script running on your PC or NAS if you don't want to install a full-fledged home automation system for that single purpose.

1coderookie, 2021-02-19:
Ah ok, things are getting more clear now ;)
Short version: get yourself the adapter and have fun with all the possibilities you'll have! :D
Depending on how your home looks like, you could for example install the adapter setup next to your existing room unit by using the existing wire from the room unit to the heater (just connect the adapter to the connectors of the room unit) and then make an installation with up to 5 sensors (= 5 rooms) [using the adapter itself for emulating a room unit](https://1coderookie.github.io/BSB-LPB-LAN_EN/chap08.html#8213-room-unit-emulation).  
Or build one of the 'room unit' solutions I've mentioned in the manual (chapters 12.6.1 to 12.6.3) or invent a new solution (in that case please report, so that I can mention that in the manual also ;) ). 
Or (if you are using any kind of home automation software) install temp sensors in your rooms and send the values from the home automation software to BSB-LAN/the heater. And so on.
I recommend that you read the manual, (almost) everything is mentioned there: https://1coderookie.github.io/BSB-LPB-LAN_EN/index.html

sebi5361, 2021-02-19:
I will write you an email.
Indeed I was afraid of those two T55 sending temperatures on the bus. But maybe I can turn off one of them. It will only show me the local temperature, and I can let it in place for the day I want to remove the other one?
I have homeassistant running and some temperature sensors already.  I like Shelly as a brand. I will have a look at what they propose to measure temperature.
I will definitely take the time to read your manual as it looks really didactic. Good to know for the 10000 parameter. But if I understood well, the Duo can do all the heavy lifting and I can run the control loop on it  directly.

1coderookie, 2021-02-19:
Ok, @fredlcore was faster ;)

> As for your other question, two T55 don't make sense because both will send their local room temperature to the heater, so the heater will get confused as it can only process one room temperature.

Connecting two T55/QAA55 isn't even possible if you only have one heating circuit, because the QAA55 is pretty 'dumb' and you can only set it up for usage with hc1/2/3. So if you would have two QAA55 configured as room unit 1 for your heating circuit 1, then you would get a BSB address conflict. If you'd set up a second QAA55 as ru2, then that value wouldn't affect the hc1 anyway. 
So that just for clarification - as you only have one room unit anyway, it doesn't matter - just for your information..

1coderookie, 2021-02-19:
>   But I would like to make sure the two t55 (the original + the modified) do no interfere.
>   I don't have room units.

So save your money on that and don't buy them - as I said, they are pretty dumb and imho not really worth buying (in your case)..

fredlcore, 2021-02-19:
If you have homeassistant running and some temperature sensors installed, then you don't need anything else. Let homeassistant collect all the temperatures, generate an average temperature (weighted, if you want) and send it to parameter 10000 on BSB-LAN. That should be sufficient for what you want to do. The T55 has to be removed in any case because while it won't harm anything physically, the T55 will still send its own measured temperature, so the heater will always readjust based on the T55's and BSB-LAN's temperature.

closed 2021-02-19 by fredlcore

sebi5361, 2021-02-19:
Thanks guys!

sebi5361, 2021-02-19:
How easy is it to modify the code to get two digital inputs on the BSB-ESP32 board to trigger on/off heating system commands (parameter 10000)?

fredlcore, 2021-02-19:
Have a look at BSB_LAN_custom*.h

sebi5361, 2021-02-19:
I will. Thks