fredlcore/BSB-LAN #340: Reboot loop with "Stack smashing protect failure!" on ESP32

zambrovski, 2021-02-22:
Hi folks,

I got the adapter configured with ESP32 from a friend since he was upgrading to newer version and set it up for my environment. After some time playing around I was able to read BPB and LBP parameters and could see them in the web interface. Then, I did a change in the web interface and got stuck, so I had to checkout the code and flash it again.

I activated Wifi as said in the manual and could successfully flash the ESP32 with adapter on it. Aparently, the system is rebooting with an error "Stack smashing protect failure!".

I'm using the ESP32-Node-MCU clone. The interesting thing is that in my friend's setup (and obviously an older version of the library or ESP libraries) the adapter and the software was working without problems for pretty long time.

I was googling for the error message and received several hits regarding MQTT modules on ESP32... Just an idea..

Do you have an advice how to proceed with this issue further... 

Here is the entire output of the serial port monitor retrieved via Arduino IDE:

```
00:19:45.250 -> ets Jun  8 2016 00:22:57
00:19:45.250 -> 
00:19:45.250 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:45.250 -> configsip: 0, SPIWP:0xee
00:19:45.250 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:45.250 -> mode:DIO, clock div:1
00:19:45.250 -> load:0x3fff0018,len:4
00:19:45.250 -> load:0x3fff001c,len:1216
00:19:45.250 -> ho 0 tail 12 room 4
00:19:45.283 -> load:0x40078000,len:9720
00:19:45.283 -> ho 0 tail 12 room 4
00:19:45.283 -> load:0x40080400,len:6352
00:19:45.283 -> entry 0x400806b8
00:19:45.615 -> READY
00:19:45.648 -> Reading EEPROM
00:19:45.648 -> Reading done.
00:19:45.648 -> EEPROM schema v.5 Program schema v.5
00:19:45.648 -> Address EEPROM option 0: 0
00:19:45.648 -> Address EEPROM option 1: 1
00:19:45.648 -> Address EEPROM option 2: 2
00:19:45.648 -> Address EEPROM option 3: 6
00:19:45.648 -> Address EEPROM option 4: 226
00:19:45.648 -> Address EEPROM option 5: 306
00:19:45.648 -> Address EEPROM option 6: 490
00:19:45.648 -> Address EEPROM option 7: 491
00:19:45.648 -> Address EEPROM option 8: 492
00:19:45.648 -> Address EEPROM option 9: 493
00:19:45.681 -> Address EEPROM option 10: 494
00:19:45.681 -> Address EEPROM option 11: 495
00:19:45.681 -> Address EEPROM option 12: 499
00:19:45.681 -> Address EEPROM option 13: 500
00:19:45.681 -> Address EEPROM option 14: 660
00:19:45.681 -> Address EEPROM option 15: 661
00:19:45.681 -> Address EEPROM option 16: 665
00:19:45.681 -> Address EEPROM option 17: 825
00:19:45.681 -> Address EEPROM option 18: 831
00:19:45.681 -> Address EEPROM option 19: 832
00:19:45.681 -> Address EEPROM option 20: 836
00:19:45.681 -> Address EEPROM option 21: 840
00:19:45.714 -> Address EEPROM option 22: 844
00:19:45.714 -> Address EEPROM option 23: 848
00:19:45.714 -> Address EEPROM option 24: 850
00:19:45.714 -> Address EEPROM option 25: 854
00:19:45.714 -> Address EEPROM option 26: 858
00:19:45.714 -> Address EEPROM option 27: 922
00:19:45.714 -> Address EEPROM option 28: 986
00:19:45.714 -> Address EEPROM option 29: 987
00:19:45.714 -> Address EEPROM option 30: 988
00:19:45.714 -> Address EEPROM option 31: 998
00:19:45.714 -> Address EEPROM option 32: 999
00:19:45.714 -> Address EEPROM option 33: 1159
00:19:45.714 -> Address EEPROM option 34: 1160
00:19:45.747 -> Address EEPROM option 35: 1164
00:19:45.747 -> Address EEPROM option 36: 1165
00:19:45.747 -> Address EEPROM option 37: 1166
00:19:45.747 -> Address EEPROM option 38: 1167
00:19:45.747 -> Address EEPROM option 39: 1171
00:19:45.747 -> Address EEPROM option 40: 1203
00:19:45.747 -> Address EEPROM option 41: 1235
00:19:45.747 -> Address EEPROM option 42: 1267
00:19:45.747 -> Address EEPROM option 43: 1299
00:19:45.747 -> Address EEPROM option 44: 1301
00:19:45.747 -> Address EEPROM option 45: 1302
00:19:45.780 -> Address EEPROM option 46: 1303
00:19:45.780 -> Address EEPROM option 47: 1304
00:19:45.780 -> Address EEPROM option 48: 1336
00:19:45.780 -> Address EEPROM option 49: 1400
00:19:45.780 -> Address EEPROM option 50: 1420
00:19:45.780 -> Address EEPROM option 51: 1440
00:19:45.780 -> Address EEPROM option 52: 1460
00:19:45.780 -> Address EEPROM option 53: 1461
00:19:45.780 -> Address EEPROM option 54: 1462
00:19:45.780 -> Address EEPROM option 55: 1463
00:19:45.780 -> Size of cmdtbl1: 29664
00:19:45.780 -> Size of cmdtbl2: 39792
00:19:45.780 -> Size of cmdtbl3: 26640
00:19:45.813 -> free RAM: 306612
00:19:45.813 -> Init One Wire bus...
00:19:45.813 -> numSensors: 0
00:19:45.813 -> PPS settings:
00:19:45.813 -> Starting SD..Setting up WiFi interface.........
00:19:46.908 -> Attempting to connect to WPA SSID: <name-of-my-network-here>..
00:19:48.898 -> You're connected to the network:
00:19:48.898 -> SSID: <name-of-my-network-here>
00:19:48.898 -> IP Address: <valid-ip-address-assigned-by-dhcp-server-in-my-network>
00:19:48.898 -> Signal strength (RSSI): -59Bm
00:19:48.898 -> Waiting 3 seconds to give Ethernet shield time to get ready...
00:19:48.898 -> Calculating free space on SD...1326535 Bytes free
00:19:52.113 -> Start network services
00:19:52.510 -> LAN->HEIZ QUR 6225 Konfiguration - Gerätefamilie: 
00:19:52.510 -> DC C2 00 0B 06 3D 05 00 02 52 88 
00:19:52.510 -> HEIZ->LAN ANS 6225 Konfiguration - Gerätefamilie: 107
00:19:52.510 -> DC 80 42 0E 07 05 3D 00 02 00 00 6B EF 55 
00:19:52.510 -> #6225: 107
00:19:52.676 -> LAN->HEIZ QUR 6226 Konfiguration - Gerätevariante: 
00:19:52.676 -> DC C2 00 0B 06 3D 05 00 03 42 A9 
00:19:52.709 -> HEIZ->LAN ANS 6226 Konfiguration - Gerätevariante: 100
00:19:52.709 -> DC 80 42 0E 07 05 3D 00 03 00 00 64 68 0E 
00:19:52.709 -> #6226: 100
00:19:52.709 -> Device family: 107
00:19:52.709 -> Device variant: 100
00:19:52.709 -> 
00:19:52.709 -> Setup complete
00:19:52.709 -> 
00:19:52.709 -> Stack smashing protect failure!
00:19:52.709 -> 
00:19:52.709 -> abort() was called at PC 0x400e6904 on core 1
00:19:52.709 -> 
00:19:52.709 -> Backtrace: 0x4008c434:0x3ffb1e90 0x4008c665:0x3ffb1eb0 0x400e6904:0x3ffb1ed0 0x400d7f87:0x3ffb1ef0 0x400e23ff:0x3ffb1fb0 0x40088b7d:0x3ffb1fd0
00:19:52.742 -> 
00:19:52.742 -> Rebooting...

```

fredlcore, 2021-02-23:
Thanks for the bug report - we have the interesting situation that the ESP32 is more sensitive to buffer overflows than the Arduino. That means that coding accidents such as buffer overflows are somehow not noticeable on the Arduino while they throw such an error on the ESP32. Since I do not use MQTT and most parts of the code are not from me, there's nothing much I can do at this stage.
What you could do is to disable MQTT in the first place and see if that stops the reboot loop. If that is the case and you know a little bit about programming, you can add the line
`Serial.println("Checkpoint 1...");`
at several point in the loop() function (and increase the number each time) to see where the error hits. If disabling MQTT works, you'll have to look out for `#ifdef MQTT` sections as this is then most likely the part where the error occurs.

1coderookie, 2021-02-23:
If you figure it out that it's really the MQTT function causing that, then please report it, cuz maybe then also @dukess, @torbennehmer and @Luposoft63 could probably have a look at it, as they developed the latest MQTT functions.

fredlcore, 2021-02-23:
I'm not sure anyone of them is running it on the ESP32 (yet)...

zambrovski, 2021-02-25:
Ok. I had an idea and now it is confirmed. Since the software was running and stopped running, I assumed something went wrong with the ESP32. So I ordered a new one and flashed it today again. Everything is running as expected. 

So sorry for fearing you, I'll close the bug.

closed 2021-02-25 by zambrovski