fredlcore/BSB-LAN #341: Problems with DHT22

iTobs, 2021-02-27:
BSB-LAN sends the values via MQTT every minute.
Only the values from DHT22 bring only "---" in over 50%.
All other values including DS18B20 are always transmitted correctly.
At Frederik's suggestion, I have the script from https://www.bastelgarage.ch/index.php?route=extension/d_blog_module/post&post_id=1
let it run longer - there was never a bug.
I process the values in OpenHAB.
What could be the cause of this problem?
Regards
Tobias

1coderookie, 2021-02-27:
Are you using the current version of BSB-LAN?
We had an issue with the DHT22 about a month ago: #318, but @dukess fixed it (#319) - after that, DHT22 worked fine when I tested it. 

> Only the values from DHT22 bring only "---" in over 50%.

Does this affect all of the three readings/values of the DHT22 at the same time, so that no value at all is transmitted, or is one of them transmitted in any case?

iTobs, 2021-02-28:
It always affects all three values:

[image]

Version 2.0.83-20210206130531
Should I update? Current is probably 2.0.88? 
Is there a change log for these small version steps? 

Thank you.

1coderookie, 2021-02-28:
No, a change log for these small steps doesn't exit. Judging by your version number I'd say that you already should have it included, but anyway - yes, please update and check again. 

How do you supply your setup with power? 

> At Frederik's suggestion, I have the script from https://www.bastelgarage.ch/index.php?route=extension/d_blog_module/post&post_id=1
> let it run longer - there was never a bug.

Did you test that with the same setup (Arduino Due&LAN-Shield) plus the same power supply?

dukess, 2021-02-28:
I can advise to enable telnet/serial console and look for messages when 
dht22 will be queried. Just for understanding what error is appearing.

fredlcore, 2021-02-28:
The previous DHT22-related error was fixed a month ago, Tobias' firmware is from 6 February, so the fix is included. But as dukess said, having a serial log would help then to see if any kind of errors are reported there when the values are queried.

iTobs, 2021-03-03:
Because the adapter is installed on the heater, it is a bit complicated with the serial protocol.
Here is the excerpt from the Telnet protocol: 

> One Wire, DHT & MAX! Sensors
> Virtual parameter 20101 queried. Table line 3919
> DHT22 sensor: 39
> Time out error,	One Wire, DHT & MAX! Sensors
> Virtual parameter 20102 queried. Table line 3920
> DHT22 sensor: 39
> Time out error,	One Wire, DHT & MAX! Sensors

correctly:
> One Wire, DHT & MAX! Sensors
> Virtual parameter 20101 queried. Table line 3919
> DHT22 sensor: 39
> OK,	#dht_temp[0]: 18.20, hum[0]:  75.40
> One Wire, DHT & MAX! Sensors
> Virtual parameter 20102 queried. Table line 3920
> DHT22 sensor: 39
> OK,	#dht_temp[0]: 18.20, hum[0]:  75.40
> One Wire, DHT & MAX! Sensors

Can I log more details?

During the test, I had the Due powered by USB, normally i use a power supply.

fredlcore, 2021-03-03:
Hm, that's strange because the it is the library (not BSB-LAN which returns the result of the DHT-read function. In that case, it states a timeout which would not even break the output if a valid (i.e. between 1 and 99) value would be returned.

There are two things I can think of:
1. I'm not sure how pin 39 is used on the Due, maybe it has other functions that might interfere?
2. Could you check if the plain example you tried before and which was working fine was actually using the same library as we ship with BSB-LAN? It could be that in the example code, you were using a more recent version of the library and therefore it is more stable? For convenience reasons, we ship some libraries with BSB-LAN so users don't have to install them manually, but after that it's mostly "never change a running system", so ours may be already outdated.

fredlcore, 2021-03-03:
Actually, I figured that our library is rather old and that the author has come up with a new one (DHTnew) that also supports ESP32. Unfortunately, it's not directly a drop-in replacement, so BSB-LAN's code would have to be adjusted. It shouldn't be much work, but I don't have a DHT22 installed anymore, so I wouldn't want to implement it without being able to test it. If you are slightly knowledgeable in C, you could probably easily do it yourself and then submit the update via a Pull Request so we can merge it into the main code.

fredlcore, 2021-03-03:
I think I found one more thing: Could it be that you only have one DHT22 sensor active? If yes, could you just add another random pin? That one will certainly throw an error or invalid results, but it could be that the correct one is then working fine. We have some kind of mandatory pause between reading the same pin again which may be the issue here. Let me know if that changes anything.

dukess, 2021-03-03:
2 seconds between queries

1coderookie, 2021-03-03:
When I did the tests with the DHT22, I also only had one sensor connected, no problem occured at my system in that case. 
But I had a problem once with not getting values (after the fix) due to a bad power supply, where the sensor just didn't get enough power/voltage from the setup to work fine. That's why I asked previously about the power supply, cuz I could be that the final setup with the additional DS18B20 sensors draws too much power and therefore it doesn't really work correct with the DHT22. When @iTobs tested with the mentioned script from bastelgarage and DHT22 only (without querying the DS18B20) it could be that the power was just enough. Just an idea..

iTobs, 2021-03-03:
I think nothing else is running on pin 39.
At the beginning I activated a second PIN, then deleted it in the settings, and now added another one. It hasn't changed anything.
The power supply was a 9V 1A power supply.
Now I use a micro PC to experiment and the Due is attached directly to it.
It does not work stably again. 
I will report.

fredlcore, 2021-03-03:
Ok, I have now added the DHTNew library, still untested. Could you please download the latest master repository and first of all see if it gets you any results and secondly, hopefully stable ones?

dukess, 2021-03-03:
New library more powerful. It allow to save state between queries and can 
be used as "one object per sensor" (but need more ram). I think we can 
rewrite dht subsystem after discontinuing mega support.

fredlcore, 2021-03-03:
Oh, does that mean that the current version would break it for Mega users because of RAM usage?

dukess, 2021-03-04:
It is _possible_, but not a fact that will _happen_.
But i found problem: 
DHTNEW dht_sensor(DHT_Pins[log_sensor]); will creating local object every time when DHT queried.
First value will return successful but second query can return error or random/zero value i think. So we need to cache values from DHT.read22.

fredlcore, 2021-03-04:
Ah, ok, I had hoped that since it is always the same variable, it would not instantiate a new object, just as if we would recreate a variable. Is that not the case?

And yes, I will leave the old library until the release, but wanted to test first if that solves @iTobs problem. Hopefully we will have feedback about this soon, then I can revert back.

iTobs, 2021-03-04:
I will install and test the current version This evening.
I just downloaded it.

iTobs, 2021-03-04:
The current version 2.0.89-20210303003632 hangs when the DHT22 is queried: 

[image]

the log file hangs :

> One Wire, DHT & MAX! Sensors
> Virtual parameter 20051 queried. Table line 3083
> One Wire, DHT & MAX! Sensors
> Virtual parameter 20100 queried. Table line 3979
> One Wire, DHT & MAX! Sensors
> Virtual parameter 20101 queried. Table line 3980

if I do nothing, the log file hangs on:

> #8830: 44.2 &deg;C
> One Wire, DHT & MAX! Sensors
> Virtual parameter 20101 queried. Table line 3980

probably with the mqtt transfer.

fredlcore, 2021-03-04:
Ok, that sucks. Because I can't test it without an DHT22, and if the current code does not work for you, we don't really know if it's the code or some other issue...
Anyone else any ideas?

dukess, 2021-03-04:
Dumb question but: all needed pull-up resistors are solded?

iTobs, 2021-03-04:
Does anyone have any idea why my due simply "freezes" with the dhtnew? Do you want to change the code back or should I check the problem on my due?

I had installed a 10 kOhm pull-up resistor.
Then I noticed that a 4.7kOhm resistor is integrated in the DHT22. 

I'll probably get a new DHT22.

fredlcore, 2021-03-04:
Hm, that's strange because it does not crash here. I don't have a sensor attached, but if the crash was code-related, I would think that it would crash here, too.  Can you try and check the result if you unplug the DHT22?

iTobs, 2021-03-04:
Currently the older code version only gives me '---'  - strange!
I have applied the new code.
If I set a PIN to which no sensor is connected, it also crashes as soon as I query the sensors: 
[image]

fredlcore, 2021-03-04:
Ok, thanks - I'm currently testing on the ESP32, not on the Due. Will check on the Due. If I have these crashes as well, I will try to fix it  or revert to the old code.

fredlcore, 2021-03-04:
In the meantime, could you please check which DHT library and version your system is using when you used the example above that worked fine with your sensor? I'll then try and add this one to the code.

What wonders me in DHTNew is this section in DHTNEW::read():
```
  if (_type != 0)
  {
    while (millis() - _lastRead < _readDelay)
    {
      if (!_waitForRead) return DHTLIB_WAITING_FOR_READ;
      yield();
    }
    return _read();
  }
```
`_lastRead` is only set in the `_read()` function. So when read() is called, _lastRead is still 0, so millis() - _lastRead will always be larger than _readDelay which is 2000. I only wonder why the code does not crash on the ESP32, maybe because of the yield()? Strange...

iTobs, 2021-03-04:
The test used: https://github.com/adafruit/DHT-sensor-library
but at the moment I get an error with this script too.
Sorry, but i have to get another DHT22 first... 
Probably my DHT22 runs after power off better for a while, but on bsb-lan the sporatic error occurred immediately, not on the test-script.

fredlcore, 2021-03-04:
Ok, that's a different library alltogether, two actually. I've reverted back to the old(er) DHT22 library and will wait for your tests once you have a new sensor. But as @1coderookie said, it could simply be a power issue, so for testing, please remove all other sensors.

fredlcore, 2021-03-05:
Ok, now I wired up a DHT22 to the Arduino myself and it's working instantly with the old library. So since @1coderookie just made tests a couple of weeks ago on his Arduino, I assume that there is either a faulty sensor or some kind of power issue. Therefore, I'll close this issue for now.

fredlcore, 2021-03-05:
I checked again a few times and once in a while, I also get a timeout error. I figured that this could be due to the different processor architecture. The timeout is only 100 microseconds, so they are using an empty loop for x iterations. x is based on the processor frequency and stored in DHTLIB_TIMEOUT.
What you could try is to change this line in src/DHT/dht.cpp
`const int DHTLIB_TIMEOUT = (F_CPU/40000);`
to
`const int DHTLIB_TIMEOUT = (F_CPU/10000);`
and see if the errors are getting less...

fredlcore, 2021-03-05:
I tried it myself as well - with the extended timeout, I now get a checksum error once in a while. So it seems that this is really a timeout - and by extending the timeout period, I still don't get better data, it's just that the data is not complete and therefore a checksum error occurs. The only thing I can think of now is that this is a library issue (and there is a bug in addressing/querying the sensor) or the library is not suitable for the Due...

fredlcore, 2021-03-05:
Also no change when updating to the current version of DHTstable. Maybe it's a hardware matter after all because do get the timeouts, they are maybe 1 out of 5 or 1 out of 10, but not every other request...

iTobs, 2021-03-05:
I have the line changed in:
const int DHTLIB_TIMEOUT = (F_CPU/10000);

The BSB-LAN sends the values to MQTT every minute.
I currently have 40 values, 25 of which are '---'.

Strangely, I haven't received any values from DHT22 since yesterday evening, only when I disconnected the Due from power today did it work again. A reset was not enough.

I found a similar problem, it was related to interupts:
https://forum.arduino.cc/index.php?topic=470599.0

fredlcore, 2021-03-05:
Please be more specific: What 40 values? DHT22? 25 "---" values from DHT22?
To my understanding, the linked thread says that turning off the interrupts hasn't changed the result reliably, but that using a different library has. I will try the one linked there.

iTobs, 2021-03-05:
Sorry, i have some problems with english. 
Meanwhile, a total of 70 temperature values from DHT22, 46 of which are '---' in the mqtt server.

But please don't make any unnecessary effort just because of me ...

fredlcore, 2021-03-05:
No, I'm having the same issues, and of course it would be good to know the root cause. I'll get back when I have updated the library.

fredlcore, 2021-03-05:
The linked library looks good here, please try the latest version from master repository.

iTobs, 2021-03-05:
Is the new version online? The last modification is 8 hours old.

fredlcore, 2021-03-05:
Sorry, forgot to push. Now updated.

iTobs, 2021-03-05:
I now have the new version running for 160 minutes. I have 160 temperature-values and no errors. 
Thank you!

fredlcore, 2021-03-05:
Great :) - no idea what the problem could have been because both libraries are rather minimal, but in the end the result is what counts ;)...

fredlcore, 2021-03-05:
BTW: I have for a long time now exactly the same temperature value (15.0 degrees) - do you have variations in temperature and humidity?

closed 2021-03-05 by fredlcore

iTobs, 2021-03-05:
The values have changed continuously :
[image]

Can I ask another question here? 
The following query appears very often: 
> DISP->HEIZ QUR 8310 Diagnose Erzeuger - Kesseltemperatur: 
> DC 8A 00 0B 06 3D 0D 05 19 4F 8C 
> HEIZ->DISP ANS 8310 Diagnose Erzeuger - Kesseltemperatur: 28.7 &deg;C
> DC 80 0A 0E 07 0D 3D 05 19 00 07 2B B3 15 

Can I easy find out why?

fredlcore, 2021-03-05:
Because that's what is shown in your boiler display :). 
And thanks for the values, maybe my basement is really stable in terms of temperature ;)...

iTobs, 2021-03-05:
So everything is stable now.
Thanks again.

dukess, 2021-03-06:
My comment: main difference between old and current libraries is in initialization delay before reading (1 millisecond vs 800 microseconds). So these microseconds can be very significat.

Note: in current version dht11 sensors will always return TIMEOUT (but i'm not sure that BSB-LAN declared dht11 support).
I think i rewrite DHT subsystem little bit later with moving DHT.setup() to setup() and using array of objects. Current algorithm i will leave with #define

fredlcore, 2021-03-06:
Yes, that's correct, but we always mention DHT22 because DHT11 is quite a bit more inaccurate (even DHT22 is less accurate than BME280). So just because of DHT11, I wouldn't make any changes.
What current algorithm do you want to keep?

iTobs, 2021-03-06:
My dht22 was stable until 2:50 this morning. Then he failed completely. Resetting the Due does not help.
I had to power off the sensor. Now it works again.
The sensor is probably defective also.

1coderookie, 2021-03-06:
Just to have it clear for myself: You are now only using the built-in 4.7k pullup, right? And connected VCC of the DHT to the _3.3V_-pin of the Due (and NOT to the 5V pin)?

dukess, 2021-03-06:
@fredlcore You can look at this
https://github.com/dukess/bsb_lan/commit/784bef22c4023e5a7e374238634f87a485f7fd9f
Pro: we do not need external timer (DHT_Timer) and variables for saving state.
Contra: memory consumption.

iTobs, 2021-03-06:
Yes, that's right. I have not additional Resistor installed. The DHT is connected to the 3,3 V Pin. 

I've already thought about the power connect to a output pin. Then i can the DHT remote switch on and off.

1coderookie, 2021-03-06:
Do you have and know how to use a multimeter for measuring the voltage? If so, you could measure the voltage of the 3,3V pin (3,3V and GND), just to make sure, that there really are 3,3V and not less..

fredlcore, 2021-03-06:
@dukess: Looks good! But I would not leave the old DHT code then, the code is already getting quite messy ;). If this is really a problem for Mega users, they can either disable DHT22 code or implement it differently in _custom.h code. At some point we have to be able to make changes without carrying on old code.
I plan to release v2.0 very soon since we now have most substantial changes running for quite some time, and afterwards I would like to no longer worry about the RAM and flash constraints of Mega users (who still can use the _defs.h updates which is the core functionality).
@iTobs and @1coderookie: I have always used my DHT being powered from a GPIO which gives me exactly the option to turn it on and off. This has worked on the Mega (5V) and now also on the Due (3.3V).

1coderookie, 2021-03-06:
> @iTobs and @1coderookie: I have always used my DHT being powered from a GPIO which gives me exactly the option to turn it on and off. This has worked on the Mega (5V) and now also on the Due (3.3V).

Pretty cool - never thought about that ;)

dukess, 2021-03-06:
@fredlcore i see two ways: we can leave current code without changes until release (and after release i will do PR with new clean code) or i can make PR now (while we have testers for DHT :) ) and clean it immediately after release.

fredlcore, 2021-03-06:
Ok, good idea! Then me and @iTobs can test the new version as well.

iTobs, 2021-03-06:
I like to take part.
My 3,3V pin has 3,32V. 
I have connected the power of the DHT22 to pin 33. 
Of course I have to turn this on with /G33=1.
Can I implement this funktion in the BSB_LAN_custom_setup.h pretty easy?

fredlcore, 2021-03-06:
@iTobs: New version is on GitHub, can you test if DHT22 works as well as before?
And yes, you can implement it in _custom_setup.h easily:
```
pinMode(33, OUTPUT);
digitalWrite(33, 1); // turn DHT22 on
digitalWrite(33, 0); // turn DHT22 off
```

1coderookie, 2021-03-06:
> My 3,3V pin has 3,32V.

Perfect :)

iTobs, 2021-03-06:
Also normal arduino code, no special function. 
to include BSB_LAN_custom_setup.h i must acrivate #define CUSTOM_COMMANDS ?
Then I get the error : BSB_LAN_custom.h: No such file or directory
So I created an empty file BSB_LAN_custom.h
That's OK, but i really only need BSB_LAN_custom_setup.h.

I have uploaded on my due the current version. 
I don't get any values from the DHT22:
```
Virtual parameter 20101 queried. Table line 3980
DHT22 sensor: 39 - Time out error

```

[17 more replies]