fredlcore/BSB-LAN #344: Some ideas in Logger graph

proasnet, 2021-03-07:
Hi, as I wrote yesterday, my ideas is some minor adjustments, as is in my picture.
[image]

fredlcore, 2021-03-07:
Negative scale has been fixed, not sure about the possibility of the D3 framework to provide a right scale. As for parameter numbers, I can have a look...

dukess, 2021-03-15:
Instead right scaling i can propose horizontal lines:
[image]

fredlcore, 2021-03-15:
Horizontal lines would be nice!

dukess, 2021-03-15:
https://github.com/fredlcore/BSB-LAN/pull/349

fredlcore, 2021-03-15:
Thanks! Will close this now as I think this is as far as we can get. If there's anything else or new, feel free to re-open!

fredlcore, 2021-03-15:
Now that I have a closer look at your image, I can see that you have an option to select the date, that is pretty cool!
Could we do something like displaying just the values of the last week by default? Because now, the data is always loaded completely into the browser (which can take quite a long time with longer logs). 
I guess we would have to add an option to the 'D' parameter for that. But on the other hand, we don't always have the (correct) date in the logfile (PPS only has day of week, so we have to use the year 2018).
Or maybe rather than taking a date, we could add a parameter to only send the last X lines of log file. But then we don't beforehand, how many lines there are in a log file. Hm... Maybe too complicated ;)...

dukess, 2021-03-15:
>Could we do something like displaying just the values of the last week by default? Because now, the data is always loaded completely into the browser (which can take quite a long time with longer logs).

i just added 
```
        data = data.filter(function(d) {
          return (d.Date >= cutoffDateS && d.Date <= cutoffDateE);
        });

```
after 
```
data.forEach(function(d) {
          d.Date = parseDate(d.Date);
          d.Value = +d.Value;
        });
```
block.
Date values will be got from date picker
      var cutoffDateS = new Date(gemById('startd3date').value + "T00:00:00");
      var cutoffDateE = new Date(gemById('endd3date').value + "T23:59:59");

where gemById() is alias for getElementById() but we can get current date and calculate week ago.

Unfortunately we must read whole file in any case. But browser can cache datalog if we reading it through external web server.

proasnet, 2021-03-15:
Yes. This is nice.
I found the next logging problem.
I have a pen with Relative output, thus is power of burner in %, now I dont
know a number, I can write later. In night is this parameter iddle and have
not zero value, but --. Graphic loger can not refresh in morning drawing
and not continue.

dukess, 2021-03-15:
Can you look in browser console? It is possible than d3 crashed when it received "---" instead numerical value.

fredlcore, 2021-03-30:
Can we close this?

proasnet, 2021-03-31:
Yes of course. Thank you very much for your work. With regards.

closed 2021-10-31 by fredlcore