fredlcore/BSB-LAN #350: http server hangs?

iTobs, 2021-03-15:
The BSB-LAN adapter no longer reacts to requests on port 80. Neither from the browser or via telnet. 
It sends values to mqtt. A ping also answers.
In the seriel monitor the arduino shows that it is working.
But i cannot access the website.
In the log i don't can see any abnormalities. 
The last working http-question was OK:
GET /K49 HTTP/1.1
Now i don't can see any http questions in the log. 

Restarting the serial monitor restarts the due. 
Now everything works fine.

Does anyone have any ideas about the cause?

fredlcore, 2021-03-15:
Is it reproducible or just a one-time glitch?

iTobs, 2021-03-15:
Some time ago, when the adapter was already hanging, I hadn't opened a serial monitor...
I will watch it.

fredlcore, 2021-03-15:
Ok, closing for now, feel free to re-open when the problem persists...

closed 2021-03-15 by fredlcore

iTobs, 2021-03-20:
The error occurs because I cannot query the log via telnet and access bsb-lan via http at the same time.

fredlcore, 2021-03-20:
You can change the debug output to serial and run the terminal either from the Arduino IDE or any other terminal program.