fredlcore/BSB-LAN #352: Backup and restore editable parameters from heating system

proasnet, 2021-03-19:
Hi, your work is perfectly. I have idea, if is possible add to your BSB LAN system Backup and Restore all editable parameters to SD card as a file. A lot of heating systems with Siemens units has not this function, that will be good. Your system can read and write all parameters, and I thing that is possible.
Thank you.

fredlcore, 2021-03-19:
@1coderookie, what do you think of that?

1coderookie, 2021-03-20:
Hmm, not sure about the (regular) use-case of this, because usually the settings won't be lost (if one doesn't initiate a complete factory reset). 
It could be a useful feature though if one starts optimizing his unit and therefore maybe wants to make sure that the settings before the changes occur can be restored with one click (even though that for this one could also just make a full query and save the output as a txt-file to have it on hand if needed) for example. 
I'm still thinking about 
a) what could be the worst case if somehow an error occurs during the restore though and 
b) if it somehow could be a problem to the controller if so many settings would occur within a short time (but for this I'm not sure how you could/would realize this restore function @fredlcore).. 
Why do you think you'd need this funtion @proasnet?

fredlcore, 2021-03-20:
I think the amount of settings should not be a problem because we will only write the next one once we have received an "ACK" telegram from the heater. But I was also thinking about potential errors. The telegrams in itself are safe because they have a checksum, but if data is read incorrectly from SD card, a wrong value might be written. This could also be prevented with a checksum on the SD card, however. But I don't know if there are values that depend on another so that you have to change let's say two settings right after each other or so?
I wonder whether this should be included into BSB-LAN as an extra function because you can already download all parameter settings via JSON or plain text via /0-9999. If you really have to reset the system, you can download them again and run a diff and see which ones (if any) have changed.

1coderookie, 2021-03-20:
> But I don't know if there are values that depend on another so that you have to change let's say two settings right after each other or so?

Not right after each other in a manner of 'critical' timing, but at least there are certain parameters that only _can_ be set if other parameters are set/activated before, yes. So that's a good point, didn't think of that before - this could be a problem.

> I wonder whether this should be included into BSB-LAN as an extra function because you can already download all parameter settings via JSON or plain text via /0-9999. 

I was looking for that JSON function (do you mean /JL?), but I (based on the description) thought that only affects the settings of BSB-LAN..? So if that also works on the whole parameters of the controller, then I don't see any need to implement this function for the SD storage. I share the concerns about a probably faulty value due to card problems and therefore maybe some 'serious' issues for the configuration of the heater.

> If you really have to reset the system, you can download them again and run a diff and see which ones (if any) have changed.

Oh, a diff is also a good idea! See, I'm more oldschool in this and like to print out that query for setting it manually at the heater itself, but running a diff and using BSB-LAN for setting is the way to do it in the new century I guess ;-)

fredlcore, 2021-03-20:
No, not /JL, that's for the configuration, I thought of /JQ=0-6, but that does not work. @dukess, would that be (easily) possible to implement? Or does it not make sense in the way that JSON is usually parsed?

As for the faulty values written/read from SD card, that won't be an issue if we use checksums on the values written to the SD card, I just mentioned this as a "note-to-self" that this would need to be implemented in order to prevent that problem. But it's relatively easy as we already have checksum functions as part of BSB-LAN.

proasnet, 2021-03-20:
There can be about two reasons. Occasionally, the main board in the heating system will fail and a reload is required after replacing the main board. This is less common. You have described the second reason well, and that is my case. I'm looking for the optimal setting of a new heating system and restore points would be a useful feature. It might be enough to save the bulk query to a txt file. But, printing to papers after changing any more parameters? I dont know. It's a working compromise, but we have 2021 and you're a great programmers. It's just an idea that might help more people than just me.

dukess, 2021-03-20:
> No, not /JL, that's for the configuration, I thought of /JQ=0-6, but that does not work. @dukess, would that be (easily) possible to implement? Or does it not make sense in the way that JSON is usually parsed?

I did save/load program values to/from .json file on JavaScript. Certainly it possible to compare two files. The only difficulty is that processing a request to save parameters takes more than 10 minutes.

fredlcore, 2021-03-20:
Could you explain more or give an example how you queried more than one parameter via JS? Thanks!

dukess, 2021-03-22:
Saving config.
1. Get list of categories and cache it. (/JK)
2. Get list of programs for every category and cache they. (/JK=1-49)
3. Query all writable programs through AJAX and cache returned values. (/JQ=program,program,program)
4. Build JSON like this:  {"programId": {"dataType": dataType, "value": "value"}, ....}

This algorithm can be implemented inside Arduino. For config restoration we can use /JS command (now it allowed multiple SET in one query but need different JSON structure as you know).

fredlcore, 2021-03-27:
@proasnet: No need for writing down anything on paper. Just save the output of /0-9999 and use this as a basis to run `diff` or a program such as BeyondCompare and you will immediately see which values have changed since compared to the previous version.
Thus I would have normally voted against additional code, but since @dukess has already done the work for you, I'll include it once it's tested.

closed 2021-03-27 by fredlcore

proasnet, 2021-03-27:
The nice work. Thank you very much.