fredlcore/BSB-LAN #356: Upgraded to DUE, now LED goes out and Heating says "no connection"

hacki11, 2021-03-21:
Hi,

i replaced the MEGA today with a DUE with software from master.
The BSB PCB is still the same.

When i plug in the Heating, the red LED is on.
As soon as i connect the power to the due, the led goes off and a few seconds later my heating says "Keine Verbindung".

I switched back in the old MEGA and the LED keeps on when on power.

Is the BSB PCB not compatible with the due? Do i have to chnage something here?
I connected it onto the same ports as the Mega.

1coderookie, 2021-03-21:
You are using the PCB v2, right? Nope, not compatible, see the note at the top of the toc and follow the link to https://1coderookie.github.io/BSB-LPB-LAN/anhang_d.html
You could try to use hw-serial of the pcb v2 and use an ESP32 instead of the Due: https://1coderookie.github.io/BSB-LPB-LAN/kap01.html#133-esp32-mit-due-kompatiblem-bsb-lan-adapter-v2

hacki11, 2021-03-21:
Oh dear, i was not aware of that.
Is it possible i broke someting in my heater? 
(hope not, if it is only connected with a opto?)

1coderookie, 2021-03-21:
No, I don't think so. As you said, the controller itself is protected due to the optos and even shorting CL+&CL- didn't harm our controllers when we messed something up during tests (iirc it only said 'Kurzschluss')..  
Try to use the mega setup again and see if it works ;)

hacki11, 2021-03-21:
Ok i will undo my change.

Where do i get that v4 adapter? Can you send me one?

fredlcore, 2021-03-21:
Send me an e-mail to bsb (ät) code-it.de

closed 2021-03-21 by fredlcore