fredlcore/BSB-LAN #362: Summer- and Wintertimestart will not take set value

hacki11, 2021-03-27:
Set is successful (INF):
```
curl -X POST "http://bsb-lan/JS" -d "{\"Parameter\":\"5\",\"Value\":\"26.03\",\"Type\":\"0\"}"
{
  "5": {
    "status": 1
  }
}
``` 

Reading the value still shows the old (25.03. instead of 26.03.):
```
 curl http://bsb/JQ=5
{
  "5": {
    "name": "Sommerzeitbeginn Tag/Monat",
    "dataType_name": "SUMMERPERIOD",
    "dataType_family": "DDMM",
    "error": 0,
    "value": "25.03.",
    "desc": "",
    "dataType": 6,
    "readonly": 0,
    "unit": ""
  }
}
```

fredlcore, 2021-03-27:
These are SET not INF parameters.

hacki11, 2021-03-27:
I used INF because i get status error otherwise:
```
curl -X POST "http://bsb-lan/JS" -d "{\"Parameter\":\"5\",\"Value\":\"26.03\",\"Type\":\"1\"}"
{
  "5": {
    "status": 0
  }
}
```

fredlcore, 2021-03-27:
Yes, it's a bug still not fixed. If you send "26.3" instead of "26.3.", it'll work.

fredlcore, 2021-03-27:
Well, /S parameter also accepts "26.3.", so I guess it's just the /JS implementation.

hacki11, 2021-03-27:
Hm, not working for me :(
```
curl -X POST "http://bsb-lan/JS" -d "{\"Parameter\":\"5\",\"Value\":\"26.3\",\"Type\":\"1\"}"
{
  "5": {
    "status": 0
  }
}
```

fredlcore, 2021-03-27:
Hm, strange, I'll have a look at it if time permits...

fredlcore, 2021-12-25:
Sorry for the long wait, but I just tested it and it works:
```
curl -X POST "http://192.168.1.50/4444/JS" -d "{\"Parameter\":\"5\",\"Value\":\"27.3.\",\"Type\":\"1\"}"
{
  "5": {
    "status": 1
  }
}
```
I read back parameter 5 from the web interface and it reflected the changes. I close this issue now, but feel free to re-open it if the matter persists.

closed 2021-12-25 by fredlcore