fredlcore/BSB-LAN #363: /Y no longer shows parameter number, category and parameter name

fredlcore, 2021-03-30:
@dukess: I only realized now that by using decoded_telegram, it seems that the output of /Y is affected in such a way that it only displays the extracted value, no longer the parameter number etc.
While this is not a (widely) publiy used function and it's also not really necessary, I wanted to ask if it is possible to bring back at least the matching parameter number, because that helped to find out whether the command ID was already known in the _defs.h or not. If it is a bigger change, then please forget about it.

dukess, 2021-03-31:
Here is "typo" bug I think. Try to replace 0 to 1 in build_pvalstr() (line 7761):
```
#endif
  // TODO: replace pvalstr with data from decodedTelegram structure
            build_pvalstr(0);
            if (outBuf[0]>0) {
              printToWebClient(outBuf);
              printToWebClient(PSTR("<br>"));
            }
```

fredlcore, 2021-03-31:
Awesome, thanks!

fredlcore, 2021-03-31:
Sorry, the problem is still not solved because now an unknown command ID makes the ESP32 crash with a "Prohibited load" error. It seems that because decodedTelegram.catdescaddr is empty (and possibly in conjunction with decodedTelegram.prognr being "-1" in these cases) is makeing some kind of moves that result in the crash.
Can we certainly know that .prognr is -1 for an unknown command ID? If yes, then we could probably just change the 
`if (extended) {`to `if (extended and decodedTelegram.prognr > -1) {`?

fredlcore, 2021-03-31:
Ok, found the solution myself already via decodedTelegram.error :)...

fredlcore, 2021-04-05:
Hi @dukess, not sure if this is related, but now(?) I no longer get the parameter numbers, names and category for unsupported parameters, for example see here:
[image]
Do you have any idea what needs to be fixed?

dukess, 2021-04-06:
Looks like build_pvalstr(0) instead build_pvalstr(1) was called. I need to 
read code but i have no time right now.

fredlcore, 2021-04-06:
Thanks for the hint, build_pvalstr(1) was correct, but inside the build_pvalstr() function there was this condition:
`if (extended && !decodedTelegram.error) {`
Once I removed `&& !decodedTelegram.error`, things were back to normal. 
We already have a check for `decodedTelegram.error == 7` in the output function, so I guess this additional check is unnecessary, at least for the webinterface. Or did you have anything else in mind with this?

dukess, 2021-04-06:
AFAIK you add "&& !decodedTelegram.error" because /Y not shown info for unknown telegrams. If /Y work fine now then we don't need change something else.

fredlcore, 2021-04-06:
Thanks! You know the code better than I do :)...

closed 2021-04-06 by fredlcore