fredlcore/BSB-LAN #364: IP adress not bound after lease time

proasnet, 2021-04-01:
Hi guys. Today I found in by Mikrotik router (RouterOS), that BSBLan with configured as DHCP client, and I have make as static, is not bounded after lease time. Only BSBLan in my network, other devices are bounded periodicaly. I can connect to WebUI of BSBLan, working good, but I dont know, that is or not problem. After BSBLan restart, is bounded, and after lease time is still in "waiting" - for bound. Any idea?
Thank you.

dukess, 2021-04-01:
I'm use Mikrotik too but haven't problem. So i have "standart" idea: try to update packages and firmware.

fredlcore, 2021-04-01:
Ok, then I'll close this for now, feel free to reopen if the problem with BSB-LAN persists.

closed 2021-04-01 by fredlcore