fredlcore/BSB-LAN #368: One wire sensor not working

Charodej, 2021-04-19:
Hello,
Inn version 20210121102308 one wire sensor works correctly DS18B20 but in newer versions with the same HW and no change in SF, only update -> one wire sensor does not work. 
Only DHT22 sensors works in both versions - new and old.

fredlcore, 2021-04-19:
Please post a log of what's going on when you query the sensor(s). @1coderookie, do you have DS18B20s running in your setup at the moment and a recent version of BSB-LAN? Do you encounter the same problem?

1coderookie, 2021-04-19:
@Charodej : Just for my information & my testing scenario: do you have more than one DS18B20 sensor or only one? Which system, Due or ESP32? Which pin(s) do you use for the OneWire(s)?
@fredlcore :  I'll try with a test-setup and the current version asap..

fredlcore, 2021-04-19:
@1coderookie: Thanks!

Charodej, 2021-04-19:
Using more DS18B20 on pin 7 with DS18B20 adapter on Arduino Due.

1coderookie, 2021-04-19:
Sorry, but I don't really understand what you mean by that.. :(
How many sensors and which kind of "DS18B20 adapter"? The one I show in the manual at the end of this chapter https://1coderookie.github.io/BSB-LPB-LAN_EN/chap07.html#712-notes-on-ds18b20-temperature-sensors or something similar? Or any kind of a busmaster chip..? Or..?

Charodej, 2021-04-19:
5 sensors ds18b20 (same as in pdf manual) and adapter looks like that
[image]

Than it goes to data pin 7.

1coderookie, 2021-04-19:
Ok, thanks. So if it worked before the update with that setup w/o any problems, it shouldn't be a HW-problem - but just for you to keep in mind: There _could_ occur problems due to the value of the resistor: 4k7 is good for one sensor, but the more sensors you add (and the longer the wires become), the smaller the resistor might has to become..
And now you don't get any value at all?
As fredlcore said, please post a log of the query.
I'll test asap..

Charodej, 2021-04-19:
There is no value given by the sensors in the parametres at all - through webinterface.
I tried already 2 new versions after the working version as stated above. Last working version is the one I am actually using.

1coderookie, 2021-04-19:
Tested with 5 sensors, works as expected:
[image]

Check the startup in the serial monitor (https://1coderookie.github.io/BSB-LPB-LAN_EN/chap12.html#122-serial-monitor), there should appear this (with a different timestamp though ;) ):
```
13:58:58.400 -> Init One Wire bus...
13:58:58.632 -> numSensors: 5
```
If it doesn't appear: 
Could it be that you didn't activate OneWire in the config.h? Check that setting:

// Compile module with one wire bus support.
#define ONE_WIRE_BUS
bool enableOneWireBus = **_false_**;  //true - enable oneWire bus, false - disable
byte One_Wire_Pin = 7;         // Define the pin for one wire temperature sensors

-> if "bool enableOneWireBus" is set to "false", enter "true" instead, flash again, observe the startup using serial monitor and do another test please by clicking on the "Sensors" button at the top of the BSB-LAN-page.

Charodej, 2021-04-19:
OK right now I have sensors reading but DHT22 stop to read 
[image]

1coderookie, 2021-04-19:
It would be nice if you would share with us what exactly you did and why suddenly the OneWire works..

1coderookie, 2021-04-19:
And _again_: the full logs from the serial monitor would be helpful!

Charodej, 2021-04-19:
I went back to the Arduino program a set 
bool enableOneWireBus = true;  //true - enable oneWire bus, false - disable
byte One_Wire_Pin = 7;    

[log.txt](https://github.com/fredlcore/BSB-LAN/files/6336182/log.txt)

1coderookie, 2021-04-19:
Ok, so your log shows no DHT22 at bootup (which should be the case iirc) and it says there is a timeout error when querying: 
```
Virtual parameter 20101 queried. Table line 4096
DHT22 sensor: 5 - Time out error
One Wire, DHT & MAX! Sensors
Virtual parameter 20102 queried. Table line 4097
DHT22 sensor: 5 - Time out error
One Wire, DHT & MAX! Sensors
Virtual parameter 20103 queried. Table line 4098
DHT22 sensor: 5 - Time out error
```
But I'm not sure why, it seems to me that no sensor is connected though.. 
Is the HW-setup still ok, or maybe a wire is loose or so? 
Did you change anything on the DHT22-config in the config.h-file? And/or did you check, that the settings for DHT22 (regarding the activation and the used pin(s)) are still ok?
It should still look like this (if you are using pin 5 for the DHT22; if you are using more than 1 sensor, then of course each belonging pin has to be assigned):
```
// Compile module with DHT temperature/humidity sensors support
#define DHT_BUS
// Define the pins for DHT temperature/humidity sensors (Up to 10)
byte DHT_Pins[10] = {5};
```

--

Another point: 
Are you using MAX!-components? Your log:
```
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Adding known Max ID to list: 00000000
Connection to max_cul: failed
```
If you don't have MAX!, then you can/should deactivate it.

dukess, 2021-04-20:
@Charodej try to use version before 03.03.2021. 03-05.03 new DHT library was added.

1coderookie, 2021-04-20:
But we tested it with the new lib and it worked, right? So I'd guess the problem is probably somewhere in the config.h or in the hw-setup as described above..

dukess, 2021-04-20:
Yes, Ulf, you right, it worked. But i take consideration small probability for program failure :)

1coderookie, 2021-04-26:
@Charodej : Any news here..?

Charodej, 2021-04-26:
excuse me not resposding - some other priorites right now. 
Some strange things are happening in my system - both types of sensors started to work, but then as right now it is not running again. Same situation as before - so I ordered some new HW to see what is the issue...
Give me some time and come back if fixed or what was cause..

1coderookie, 2021-04-27:
No problem.
Make sure that  
a) you connect the sensors to 3,3V at the Due (so that the GPIOs only get 3,3V signal) and
b) you're using a sufficient power supply. In case you are using USB as power-supply, you can measure the voltage at the 3,3V and 5V pins at the DUE for example, just to check if at least the voltage is fine there.  
Good luck!

1coderookie, 2021-09-15:
@Charodej  : Did you get it working - can we close this issue?

Charodej, 2021-09-19:
Hello, thank you so much for the information. At the end it was power supply issue and short circuit in one of the sensors as well. It is right now fix and everything running. Thank you for your work!

closed 2021-09-19 by Charodej