fredlcore/BSB-LAN #370: Need to rewrite "programIsreadOnly"?

fredlcore, 2021-05-03:
@dukess: I have now found out a way to get the information from the heater whether a parameter is read-only or not as well as whether a parameter is OEM or not.
Still, it would take too long to query this each time before trying to write a parameter, so we will still store this information in BSB-LAN_defs.h.

The problem is that currently, we make all OEM parameters writeable, even if they are read-only. So what I'm thinking of is this:
- Each OEM parameter that is read-only will get the flags `FL_OEM+FL_RONLY`. 
- Each OEM parameter that is writeable will get the flag `FL_OEM`.

The logic for programIsreadyOnly should now be like this:
0 - everything is read-only
1 - All FL_OEM and all FL_RONLY parameters are read-only, the rest is writeable
2 - All FL_RONLY parameters are read-only, the rest (including FL_OEM that do not have FL_RONLY) are writeable (so FL_OEM that have FL_RONLY are read-only).

I'm not sure if I 100% understand the different logics in programIsreadOnly, so if you could have a look at it that would be great!

dukess, 2021-05-04:
In this case we should change 
`      case 2: if ((param_len & FL_RONLY) == FL_RONLY && (param_len & FL_OEM) != FL_OEM) return true; else return false; //All writable except read-only`
to 
`      case 2: if ((param_len & FL_RONLY) == FL_RONLY) return true; else return false; //All writable except read-only
`

fredlcore, 2021-05-06:
Great, thanks!

closed 2021-05-06 by fredlcore