fredlcore/BSB-LAN #373: Problems with ClientID "BSB-LAN" at the mosquitto mqtt broker, when using two or more bsb-lan Adapters in the network

eikejenn, 2021-05-26:
Version: Master branch.

Hi guys, 
when i am using an additional bsb-lan adapter for test szenarios in my network, i can see problems in the mosquitto logfile.
The bsb-lan controller uses a hard coded ClientID for the PubSub Connection. My mosquitto closes the old connection, because
the clientID has to be uniq.

May 25 22:53:37 raspi20 mosquitto[1970]: 1621976017: Client BSB-LAN already connected, closing old connection.
May 25 22:53:37 raspi20 mosquitto[1970]: 1621976017: Socket error on client BSB-LAN, disconnecting.
May 25 22:53:37 raspi20 mosquitto[1970]: 1621976017: New client connected from 192.168.10.224 as BSB-LAN (c1, k120, u'heizung').
May 25 22:54:28 raspi20 mosquitto[1970]: 1621976068: Client BSB-LAN already connected, closing old connection.
May 25 22:54:28 raspi20 mosquitto[1970]: 1621976068: Socket error on client BSB-LAN, disconnecting.
May 25 22:54:28 raspi20 mosquitto[1970]: 1621976068: New client connected from 192.168.10.98 as BSB-LAN (c1, k15, u'heizung').
May 25 22:54:28 raspi20 mosquitto[1970]: 1621976068: Client BSB-LAN already connected, closing old connection.
May 25 22:54:28 raspi20 mosquitto[1970]: 1621976068: Socket error on client BSB-LAN, disconnecting.
May 25 22:54:28 raspi20 mosquitto[1970]: 1621976068: New client connected from 192.168.10.224 as BSB-LAN (c1, k120, u'heizung').
May 25 22:55:18 raspi20 mosquitto[1970]: 1621976118: Client BSB-LAN already connected, closing old connection.
May 25 22:55:18 raspi20 mosquitto[1970]: 1621976118: Socket error on client BSB-LAN, disconnecting.
May 25 22:55:18 raspi20 mosquitto[1970]: 1621976118: New client connected from 192.168.10.98 as BSB-LAN (c1, k15, u'heizung').
May 25 22:55:18 raspi20 mosquitto[1970]: 1621976118: Client BSB-LAN already connected, closing old connection.
May 25 22:55:18 raspi20 mosquitto[1970]: 1621976118: Socket error on client BSB-LAN, disconnecting.
May 25 22:55:18 raspi20 mosquitto[1970]: 1621976118: New client connected from 192.168.10.224 as BSB-LAN (c1, k120, u'heizung').

a solution example:

use a #define in the config.h to set the clientID for the mqtt connection.
Additionally, a note about this problem could help you understand what this parameter does.

You will find this in BSB_LAN.ino at line 9370:
>    String MQTTWillTopic = mqtt_get_will_topic();
>     int retries = 0;
>     while (!MQTTPubSubClient->connected() && retries < 3) {
>       MQTTPubSubClient->connect(PSTR("BSB-LAN"), MQTTUser, MQTTPass, MQTTWillTopic.c_str(), 0, true, PSTR("offline"));
>       retries++;
>       if (!MQTTPubSubClient->connected()) {

Regards

Eike

1coderookie, 2021-05-27:
Thanks for the feedback.
@torbennehmer and @Luposoft63  what do you guys think about it?

fredlcore, 2021-05-27:
We have MDNS_HOSTNAME which we could (re)use, but this may not be set in case someone disables MDNS.

torbennehmer, 2021-05-28:
@1coderookie @fredlcore @Luposoft63 I'll look into this. Could be start of next week until I get around though.
@eikejenn How urgent is this for you?

eikejenn, 2021-05-29:
That's okay with me. My workaround is to change the value in the BSB_LAN.ino directly to another value

torbennehmer, 2021-06-03:
@fredlcore I've a fix for this prepared, but I'm having trouble to get the EEPROM Config management working, appearantly I'm missing something trivial, I'd guess. How can we handle that, should I just prepare a PR or do you want to take a peek directly at this branch:

https://github.com/torbennehmer/bsb_lan/tree/issue-373-mqqt-clientid

Specifically:
https://github.com/torbennehmer/bsb_lan/commit/94b37ba07ec21457f3af08ca5f5806c084979aec

I can see the option in the config screen, but as soon as I activate EEPROM driven configuration, I'm always ending up with an empty MQTTClientID. 

Is there any howto on handling the EEPROM Config?

@eikejenn If you can live without configurability, you can give that branch a test-run, it looks like this here:
[image]

Best regards,
Torben

fredlcore, 2021-06-03:
As for the EEPROM storage, I think it's best if @dukess has a look at it first.
Also, once we're at it, I would like to be able to define one single hostname that is used for all network related features, so besides MQTT also MDNS and whereever else it might be useful/necessary. The webinterface would get really cluttered if we have different settings for different hostname functions.

torbennehmer, 2021-06-03:
@fredlcore Sounds good to me. do you want this in a single fix? Then I'd drop the current MQTT specific implementation and replace it with something like what you suggest.

In respect to MQTT: I'd suggest to leave the Client ID Option at least in the Header File but not in the UI, for those edge cases, where you have exotic MQTT configurations at hand. The default hostname as client ID will suffice in 99% of the cases. What do you think?

dukess, 2021-06-03:
I wrote some comments at commit.

fredlcore, 2021-06-04:
@dukess: Thanks!
@torbennehmer: Since I'm currently in the process of adding hundreds of new parameters and this issue is non-critical, I would prefer to include a single fix once it is done, including the proper adding in the UI. I would still prefer to have one hostname in the UI as well as in the header file for simplicity and clarity. Those 1% of the users would still have the option to change it in the source code directly, just as @eikejenn did. The code already has quite a few "special interest" configuration options that I would rather reduce, not increase ;)...

torbennehmer, 2021-06-05:
@fredlcore Fair enough. I've been thinking about it once more: For our home-based usage scenario a quasi-random client id based on the mac address should be reasonable transparent. Most MQTT Installations will handle their access permissions via the user/password combo. There are edge cases, where the client ID is actually relevant, but since it is stable per BSB-LAN, even this should be ok.
So I see two options:
a) stick with the quasi random ID I've implemented, this might mean for some users to adapt their MQTT Broker settings accordingly. But that should be a one-time change, even though it's a breaking one.
b) Modify the code to use the existing MQTTDeviceID option, used in the JSON Payload. I haven't thought about that one for whatever reasons, but it actually is pretty close to what we need here. If we have two adapters in the same network, you'd use two separate Device IDs in JSON context as well as on the protocol level. The option defaults to BSB-LAN, so it'll be backwards compatible for those, who didn't change that option.

I'd actually choose b) as best option, so I'll update the branch in question accordingly (push to my repo will follow shortly).

Let me know if I should rebase to master and send in a PR.

fredlcore, 2021-06-05:
I agree that option b) would be better, not only because it prevents a breaking change. If there is a need for a differing hostname (e.g. due to two adapters in the same network), there is likely also a need for a different MQTT client id, so in terms of configuration, it should all be based on the same name. 

Once it's tested and ready, please send me a PR. Thanks for all your work!

torbennehmer, 2021-06-05:
@fredlcore PR #376 is in, that should do it I guess.

PR #376 (2021-08-28): Make MQTT client id configurable
Fix #373 for two adapters conflicting on the broker with the same client id. Use the MQTTDeviceID instead of a hard coded default. If the option is unset, use "BSB-LAN" for backwards compatibility (in most cases anyway).

Attention: Breaks compatibility if and only if you already have a custom MQTTDeviceID *and* your broker relies on the client ID in any way for authorization purposes. In almost all other cases, the change will not be noticeable.

Removed a duplicate entry in the .gitignore while at it.
commit dc360729 (2021-08-28): Make MQTT client id configurable

Fix #373 for two adapters conflicting on the broker with the same client
id. Use the MQTTDeviceID instead of a hard coded default. If the option
is unset, use "BSB-LAN" for backwards compatibility (in most cases
anyway).

Attention: Breaks compatibility if and only if you already have a custom
MQTTDeviceID *and* your broker relies on the client ID in any way for
authorization purposes. In almost all other cases, the change will not
be noticeable.
closed 2021-08-28 by fredlcore