fredlcore/BSB-LAN #374: setting time range (in my case TWW) doesn't work

eikejenn, 2021-05-26:
Hi guys,

when i open the URL "http://heizung/K6", the actual time range will read and shown correctly.
The time range settings are filled in the textbox in this format: "1. 05:30 - 07:00 2. 11:30 - 12:00 3. 16:30 - 18:00"
When i change the time range and keep the format, the controller creates this set-URL:
"http://heizung/S560=1._05:30_-_07:00_2._11:30_-_12:00_3._16:30_-_18:00"

But the expected format is "http://heizung/S560=05:30-07:00_11:30-12:00_16:30-18:00"

When i correct the format in the textbox to this one "05:30-07:00 11:30-12:30 16:30-18:00" and click on set,
the controller will set the time range fine.

It would be better to fill the text box directly with the correct format.
Regards 

Eike

eikejenn, 2021-05-26:
When only one time range is defined, this format is expected: "06:00-22:00 xx:xx-xx:xx xx:xx-xx:xx"
The textbox output from the bsb-lan controller ist actual: "1. 06:00 - 22:00 2. --:-- - --:-- 3. --:-- - --:--"
the minus characters "-" should be converted to "x"

dukess, 2021-06-18:
Main problem is backward compatibility. Same format is used in JSON and MQTT. If @fredlcore decides to drop compatibility (while preparing for the 2.0 release), then yes, this issue can be resolved quickly.

fredlcore, 2021-06-18:
I wouldn't mind breaking compatibility, but maybe that's not necessary. The way it's displayed in the web-interface can be changed without breaking compatibility. We only need to stick with the URL format, i.e. /S560=05:30-07:00_11:30-12:00_16:30-18:00

I wonder if the "-" would need to be converted to "x" because all the code does to parse the string is this:
`ret=sscanf(val,sscanf_buf,&h1s,&m1s,&h1e,&m1e,&h2s,&m2s,&h2e,&m2e,&h3s,&m3s,&h3e,&m3e);`
The h.. and m.. variables are set before and apparently if the value cannot be parsed (i.e. with because of an 'x'), it sticks with the set value. The question is whether this also works when using a '-' instead of an 'x'. If yes, then this can stay as it is and I just need to adjust the way the string is displayed in the webinterface. @dukess, do you know more about the behaviour of sscanf in that regard?

dukess, 2021-06-18:
> The way it's displayed in the web-interface can be changed without breaking compatibility. 

We will have to rewrite code, because value is directly taken from printTimeProg function via variable decodedTelegram.value

>The question is whether this also works when using a '-' instead of an 'x'. If yes, then this can stay as it is and I just need to adjust the way the string is displayed in the webinterface.

Behavior will not changing i think

fredlcore, 2021-06-18:
Yes, but inside printTimeProg we just need to change the sprintf command? This will not break JSON/MQTT compatibility, or does it?

fredlcore, 2021-06-18:
Ok, setting the program with a string in the web interface like this works:
`06:30-07:00 16:30-18:00 --:-----:--`
It looks a bit strange because the '-' in between the range is not easily visible, but it works. So it's more a question of taste which one we shall use.
Would it break compatibility if we change this line:
`len+=sprintf_P(decodedTelegram.value+len,PSTR("%02d:%02d - %02d:%02d"),msg[k],msg[k + 1],msg[k + 2],msg[k + 3]);`
to this one?
`len+=sprintf_P(decodedTelegram.value+len,PSTR("%02d:%02d-%02d:%02d"),msg[k],msg[k + 1],msg[k + 2],msg[k + 3]);`

dukess, 2021-06-19:
> It looks a bit strange because the '-' in between the range is not easily visible, but it works. So it's more a question of taste which one we shall use.

sscanf just can't recognize values and then skip it.

> len+=sprintf_P(decodedTelegram.value+len,PSTR("%02d:%02d-%02d:%02d"),msg[k],msg[k + 1],msg[k + 2],msg[k + 3]);

Here should be not a problem.

> len+=sprintf_P(decodedTelegram.value+len,PSTR("%d. "), i + 1);

There are no problems here for me but I do not know how others do parsing the parameters: they are can consider numbering order.

fredlcore, 2021-06-19:
Ok, I see. I'm not sure if any external module has actually done the work of parsing it, but that should not be an issue at this stage. @eikejenn: Can you please confirm that minusses also work, and that converting these to 'x' is not necessary? The only necessary change is to omit the whitespace before/after the minus that connects each of the two times, I guess...

fredlcore, 2021-06-19:
i.e.: 
`http://heizung/S560=05:30-07:00_11:30-12:00_--:-----:--`
should work the same as 
`http://heizung/S560=05:30-07:00_11:30-12:00_xx:xx-xx:xx`
or even
`http://heizung/S560=05:30-07:00_11:30-12:00_##:##-##:##`

fredlcore, 2021-12-25:
Since there has not been an update here for a while, I assume it's working as it is and therefore close this issue. Feel free to re-open if any issues in that matter come up again.

closed 2021-12-25 by fredlcore