fredlcore/BSB-LAN #375: [FEATURE REQUEST] Powering the microcontroller through the BSB.

sebi5361, 2021-05-27:
Can my ESP32 be powered by the Boiler System Bus, the same way an Atlantic T55 thermostat is?
A capacity should do the job to overcome temporary BSB communication protocol short-circuits, and a 12V to 3.3V buck converter should provide the required voltage. Did someone experiment such a circuit?

1coderookie, 2021-05-27:
We had that discussion a (long) while ago in the FHEM-Forum, but we came to the conclusion that we don't want to try anything like that. The point is, that we don't really know how much current the controller could offer. 
If you want to try that out (probably using G+ as it supports the 12V constantly for the backlight of the room units) you are of course free to do so on your own risk, but I would highly recommend to NOT do so, because you could damage the controller of your heating system. That couldn't only be dangerous, it would get really expensive if you would have to get a new controller installed. So just stick with the little extra power supply for the ESP and everything is fine.. ;)

1coderookie, 2021-05-27:
..in case you speak German, the discussion started in december 2018 and was a few 'pages' long: https://forum.fhem.de/index.php/topic,29762.msg870270.html#msg870270

sebi5361, 2021-05-27:
Thank you for your answer.

closed 2021-05-27 by sebi5361