fredlcore/BSB-LAN #377: [BUG]

proasnet, 2021-06-11:
**BSB-LAN Version**
2.096

**Architecture**
Arduino due

**Bus system**
BSB

**Describe the bug**
HA Integration not working good

Hi,
from version 2.092 not working fine a integration in Home Assistant. I know, that problem may be in the integration, but I write about in here too.
In dashboard I have a integration with actual room temperature, mode and setpoint. In version 2.088 it working fine, but from 2.092 and up, I have room teperature under zero, for example now -4,3, but real value is 26
[image]
Any idea where is a problem?

fredlcore, 2021-06-12:
As stated here: https://www.home-assistant.io/integrations/bsblan/ the integration currently only works with version BSB-LAN version 1.0
Please get in touch with the author, @liudger to solve this matter. If there is a need for change on the BSB-LAN side, please feel free to open a new issue with a concrete requirement what needs to be done (or fails) on BSB-LAN side.

closed 2021-06-12 by fredlcore

proasnet, 2021-06-17:
Hmm, it's interesting. As I said at the beginning, everything worked in version 2.0.88 which is already version 2 and not 1 and after the update to 2.0.92 and above it did not work. There has been a change that is causing the fault. Dashboard support for HA seems slow and unresponsive to me. Today I will try to downgrade to 2.0.88

proasnet, 2021-06-18:
I uploaded version 2.0.88 and I confirm that the integration started working properly in HA again.

fredlcore, 2021-06-18:
Can you run a `diff` between the two versions? If not, then send me the one that works and the one that doesn't (BSB_LAN.ino only).

proasnet, 2021-06-21:
Okay. If you have not your own versions saved, I send you today.
API working fine with version 2.0.88 and lower...
Not working from 2.0.92 and higher.
Versions between 2.0.88 and 2.0.92 I am not have and I dont know about working API.

fredlcore, 2021-06-22:
So far, there are hundreds, if not thousands different commits (i.e. "versions"), and no, I don't keep a copy of each of them.