fredlcore/BSB-LAN #379: Remove "protected pins"

fredlcore, 2021-06-30:
As I keep getting questions regarding which pins could be used for external sensors, especially on the various ESP32 boards, I'm thinking that it probably would be best to remove the "protection" of "protected pins".

This feature was meant to prevent interference between an incorrectly set up sensor and BSB-LAN, but in practice, there is hardly a (sane) situation where this could lead to a significant problem. The main "harm" that could be done is that you pick a pin for your sensor that is already in use with a different device/service on the microcontroller. Examples would be the I2C, SPI or serial port pins. That would mean that either this device would not function properly (i.e. you could not read/write to the SD card) or it may have no effect at all because such inconsistent data would simply be ignored by the other device.

With the growing number of boards that this project supports and the stark variations even with a board family, I don't think it is still feasible to "protect" the user from one's own sloppiness/lazyness. Instead, if the user wants to use an external sensor, they would have to take a look at the datasheet of the board they are using and identify freely available pins and then use these without us having to constantly check the various board specifications (with the ESP32, we would already need three different "protected pins" sets). The way it is now is only more confusing for the user because we're currently using "overly protective" settings just to cover any possible configuration.

Thoughts?

dukess, 2021-07-12:
We can make it more intelligent: when changing the parameters, we can check whether the same contact values are used for different types of sensors and display a warning. The list of protected contacts should be created dynamically, using the known values of used contacts from variables.
Variables for the list: Serial pins for adapter, SD-card pin, Ethernet pin, one wire pin, DHT22 pin(s), BME280 pin, RGT buttons pins. I don't think I've forgotten anything.

Proof-of-concept on the knee:
```
//These constants should be set for every architecture/board
static const sd_pin = 4;
static const eth_pin = 10;
static const bme280_pin = 66; //just for example. i not remember pin for i2c

bool pinIsUsed(byte pin){
  if (bitRead(protected_GPIO[pin / 8], pin  % 8)){
    return true;
  }
  bitWrite(protected_GPIO[pin / 8], pin  % 8, 1);
  return false;
}

//Should clear protected_GPIO array before first pinIsUsed() call
void clearPinList(){
  for (i=0; i<anz_ex_gpio; i++) {
    protected_GPIO[i] = 0;
  }
}

//Return true if any intersection found
bool pins_intersection_check(){
  clearPinList();
  if(pinIsUsed(sd_pin)) return true;
  if(pinIsUsed(eth_pin)) return true;
  if(pinIsUsed(bus_pins[0])) return true;
  if(pinIsUsed(bus_pins[1])) return true;
  if(One_Wire_Pin && pinIsUsed(One_Wire_Pin)) return true;
  for (i=0; i<sizeof(DHT_Pins)/sizeof(DHT_Pins[0]); i++) {
    if(DHT_Pins[i] && pinIsUsed(DHT_Pins[i])) return true;
  }
  if(pinIsUsed(bme280_pin)) return true;
//RGT buttons
  for (i=0; i<sizeof(button_on_pin)/sizeof(button_on_pin[0]); i++) {
    if(button_on_pin[i] && pinIsUsed(button_on_pin[i])) return true;
  }
  return false;
}
```

protected_GPIO can be removed from config file in this case.

fredlcore, 2021-07-19:
This sounds good, but the problem is that on the Olimex, for example, we have a number of pins that are occupied with other functions already, even if BSB-LAN is not using them (CAN bus, i2C etc.). It may be good to have a check if the pins used within BSB-LAN are not used twice (e.g. a OneWire pin isn't also used for DHT22). But what we cannot do (at least since we have various ESP32 boards in use) is to have a set of protected pins for each board type.

Then again, if we do this kind of sanity check on a page where people can see directly what they enter for OneWIre, DHT22 etc. next to each orther, and they don't realize if they enter the same value twice, we would have to include similar sanity checks at other, more important parts of the program. 

If we limit the checks to prevent a double entry of the same pin number for the different sensors and RGT, we could also do this via JavaScript inside the browser?

dukess, 2021-07-19:
You right :)

>If we limit the checks to prevent a double entry of the same pin number for the different sensors and RGT, we could also do this via JavaScript inside the browser?

It is not easy now because web interface have no info about variables types and their relations. 
The main problem: to implement this function, we will have to add another field to the parameter table and write a fairly large piece of JS code.

fredlcore, 2021-07-19:
I think if we give each field an ID, we could do something like this (untested):

```
var fields = ['field1', 'field2', 'field3'];
for (let x = 0; x < fields.length; x++) {
  for (let y = 0; y < fields.length; y++) {
    if (x != y && document.getElementById(fields[x]).value == document.getElementById(fields[y]).value) {
      alert('Double use of pin ' + document.getElementById(fields[x]).value);
    }
  }
}
```
For fields where several pins can be entered, we would have to break the value up into an array again and add an additional loop, but that should'nt be more than a few more lines. Or am I missing something?

dukess, 2021-07-19:
The solution is quite working, the main obstacle is that there is no guarantee that the parameter numbers will not change. That's why I suggested adding another field in configuration_struct with a numeric identifier. By fact, it means the group number. For example: 0 - do not check the parameters, 1...254 - compare parameters with the same identifier.

By the way we can call checking routine by onChange event.

fredlcore, 2021-07-19:
Hm, what do you mean with "no guarantee that the parameter numbers will not change"? If I'm not mistaken, the sorting of the cf_params ENUM does never change, new parameters are always added at the end, correct? So could we use the ENUM position of each entry and add it as the ID of the <input> tag? Then the JavaScript could look at specific IDs and perform sanity checks?

dukess, 2021-07-20:
Yes, you are right, but I do not exclude the possibility of human error or the removal of an unnecessary parameter(s) in the future. We can protect ourselves from such errors if we add warnings to the source code, but it will be better if the id mapping will occurs automatically.

fredlcore, 2021-07-20:
That's correct, but the more I think about it, the more I think it's better to have a proper "warning" in the manual about double-checking the pins you want to use and check with the datasheet of the microcontroller/dev-board instead of having the software do it for just a few configuration options that isn't even doing a complete check.
If we remove this kind of protection, people will know that they have to do some checking before using the pins for sensors. If we (just) check for doublettes, we'll still get people having problems because they think that it should work because BSB-LAN is not complaining, even if the problem is related to some other, microcontroller-related use of that pin.

Thanks anyway for brainstorming on this matter!

closed 2021-07-20 by fredlcore