fredlcore/BSB-LAN #380: [BUG]

DaSigg, 2021-07-17:
**BSB-LAN Version**
2.0.99 20210624231250

**Architecture**
Arduino Due

**Bus system**
BSB

**Describe the bug**
To delete the vacation parameter 632 on my Elco Thision S13 Plus the day and month must be set to a meaningful value, e.g. 1.1. 
All other parameters (Year, Hour, Minute, Sec and also DayOfWeek) must be set to 0.
BSP-LAN set the year to 0x93 (the last byte from 0 - 1900), all others to 0xFF.

fredlcore, 2021-12-25:
Sorry for the long delay, I've changed the code now so that it sends all zeros on the unused fields. Could you run a test on your site and please log the telegrams in that process? The strange thing is that my Thision S17.1 returns 0xFF for all unused fields, not zeros. At least it accepts a zero in the unused fields, so it shouldn't clash with the requirements of your heater, but I'd be interested to see whether the return telegram on your Thision is actually different.
Thanks.

DaSigg, 2022-01-01:
Thank you.
Tested with vaction parameter 632. It looks OK. 
Log:

GET /S632=02.02 HTTP/1.1
set ProgNr 632 = 02.02
day/month: 2.2.
setting line: 632 val: 06 00 02 02 00 00 00 00 17
LAN->HEIZ SET  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat: ---
DC C2 00 14 03 3D 05 09 C4 06 00 02 02 00 00 00 00 17 F2 2C
HEIZ->LAN ACK  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat:
DC 80 42 0B 04 05 3D 09 C4 14 4E
LAN->HEIZ QUR  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat:
DC C2 00 0B 06 3D 05 09 C4 51 9A
HEIZ->LAN ANS  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat: 02.02.
DC 80 42 14 07 05 3D 09 C4 00 00 02 02 00 00 00 00 17 B2 5E
#632: 02.02.

GET /S632= HTTP/1.1
set ProgNr 632 =
setting line: 632 val: 05 00 01 01 00 00 00 00 17
LAN->HEIZ SET  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat: ---
DC C2 00 14 03 3D 05 09 C4 05 00 01 01 00 00 00 00 17 C9 0A
HEIZ->LAN ACK  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat:
DC 80 42 0B 04 05 3D 09 C4 14 4E
LAN->HEIZ QUR  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat:
DC C2 00 0B 06 3D 05 09 C4 51 9A
HEIZ->LAN ANS  632 Ferien Heizkreis 1 - Periode 1 Beginn Tag/Monat: ---
DC 80 42 14 07 05 3D 09 C4 01 00 02 02 00 00 00 00 17 59 7D
#632: ---

fredlcore, 2022-01-01:
Great, will close this issue now.

closed 2022-01-01 by fredlcore