fredlcore/BSB-LAN #381: [BUG] parsing of webif config creates wrong character

SebastianJuch, 2021-07-19:
**BSB-LAN Version**
#define MAJOR "2"
#define MINOR "0"
#define PATCH "99"
#define COMPILETIME "20210624231250"

**Architecture**
ESP32 NodeMCU

**Bus system**
BSB

**Describe the bug**
After saving config settings over the webif, the SSID (stored in BSB_LAN_config.h) get's manipulated if it has spaces. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set the SSID in BSB_LAN_config.h to a name with spaces "The Nexus"
2. compile and flash
3. Open WebIf
4. SSID will shown correctly (see before.png)
5. Change one Setting (as example enable mqqt service) and press save
6. SSID is schowing now "The+Nexus"
7. 
**Logs**
/

**Expected behavior**
SSID is not changing

**Screenshots**
before.png
[image]
after.png
[image]

**Desktop (if applicable, please complete the following information):**
 - OS: WIN10
 - Browser Chrome up to date
 - Version last

**Smartphone (if applicable, please complete the following information):**
/

**Additional context**
/

dukess, 2021-07-19:
Thanks for report!
Here is no strings conversion to URL format and vice versa so we got this unsatisfactory behavior.

fredlcore, 2021-07-19:
Thanks, @dukess!

closed 2021-07-19 by fredlcore

SebastianJuch, 2021-07-19:
Thanks for fixing!