fredlcore/BSB-LAN #386: [FEATURE REQUEST] HTTPS connection and secure MQTT for BSB-LAN

GSzabados, 2021-08-16:
BSB-LAN provides an amazing variety of features for gas heaters, but I think these features should be accessed through a secure connection considering current security  trends and zero-trust on any device on your local network. (I am not comfortable with the fact that other devices on the same network can see without any encryption the traffic to/from BSB-LAN.)

The feature request would be to add encryption to both HTTPS access and secure MQTT. I am fine with self-signed certificates.

Partial alternative is the limited IP access, and it is a good approach, but the communication is still unencrypted.

To support the development of the feature request, my contribution would be testing the new option as I have limited programming knowledge.

---- 

This feature request has been made according our discussion (https://github.com/fredlcore/BSB-LAN/discussions/385) and @torbennehmer's request to open it separately. 

I really appreciate all of your efforts and contribution in the development of BSB-LAN. It is an amazing piece of hardware and software.

fredlcore, 2021-08-16:
To my knowledge, the hardware is too limited (at least on the Arduino, but I think RAM-wise, also on the ESP32) to implement even just self-signed SSL encryption. It would mean a lot of work for again just limited additional security. Someone who is able to attack the white-listed IP address will also likely be able to perform a man-in-the-middle attack with a forged self-signed SSL certificate.
The actual data transmitted to/from BSB-LAN is actually not sensible information (unless you open BSB-LAN to the internet): Someone who is inside your WLAN is inside your house and could just go to the heater and change the settings at the room controller or at the heater itself. If you trust people enough to give them access to your local network, but don't trust them enough regarding a potential attack on BSB-LAN, you could put BSB-LAN into a VLAN separated from the "public" WLAN. It could then be accessible only via your home automation software which most likely runs on more powerful hardware and thus provides a much more sensible layer of (access) protection.

Somehow, I cannot add @torbennehmer as an assignee here, but if you have a different idea/approach, I'm happy to listen. However, since space, both in terms of RAM and Flash, is always an issue, it would have to run on a rather tight footprint because I don't want to force users on a different platform just because of that.

GSzabados, 2021-08-16:
> Someone who is inside your WLAN is inside your house and could just go to the heater and change the settings at the room controller or at the heater itself.

It is more about other devices on the network, like a hacked NAS. Or dodgy IoT device...
My real concern if the harmful actor changes some settings on the heater (ie. disable freeze protection, or overheats the system... I am not sure what physical measures are built to the heater itself outside of the LMS what would prevent those cases.)

> However, since space, both in terms of RAM and Flash, is always an issue, it would have to run on a rather tight footprint because I don't want to force users on a different platform just because of that.

I truly understand this issue. I will look into this as well, what solutions are available currently for ESP devices. Maybe we can find something what fits the bill.

fredlcore, 2021-08-16:
I understand your concern, and even if the possibility of a hacked NAS that targets a niche solution such as BSB-LAN is rather unlikely, it's still a valid point. But compared to other harmful actions that a hacked NAS could do, the chances of really damaging your heating system are rather minimal. I'm sure that @1coderookie will disagree to a certain degree, but so far all the heating systems that I have come across have built-in minimum and maximum values. These are hard-coded into the devices (as I found out during the disassembly of the complete dump), so it's not possible to set the freeze protection to less than 4 degress Celsius (IIRC). Sure, you can set certain values in a way that might put a strain on the valves etc., but that would be long term damage rather easily noticeable due to performance drops and no hacker would have in interest to make himself "heard" by making a valve or relay turn on and off all the time, making suspicious noises ;). 

But again, it's a valid point and if the cost-benefit ratio is favorable, I'm more than happy to add any code that you guys come up with.

1coderookie, 2021-08-16:
> My real concern if the harmful actor changes some settings on the heater (ie. disable freeze protection, or overheats the system... I am not sure what physical measures are built to the heater itself outside of the LMS what would prevent those cases.)

Overheating:
Afaik (and what the LMS manuals say) the max boiler temp (2212 - setpoint max) is usually pre-set to 80°C; the setpoint max OEM (2213) is set to 85°C. One can't set 2212 higher than 2213. However, manuals say that the setpoint max OEM (2213) _could_ be set to 120°C - but for that one has to change the OEM setting. To do that, an 'intruder' a) has to find that setting to be able to change OEM settings and b) has to know about this OEM temp limitation already. So that guy has to have knowledge about this heater or heating systems in general already - and if that really would be the case, then he probably would change other settings just to drive you crazy instead of just setting the max possible boiler temp trying and hoping to damage your system by doing so (at least I would do it in this way if I would be the nasty guy in this scenario.. ;) ). 

Frost protection:
In general, the boiler setpoint min (2210) is set to 40° as default, same with 2211 (setpoint min OEM). 2211 could be set to minimum = 8°C though. These settings determine when the heating system should start the burning process for heating up the boiler in general. 
Setpoint _frost protection_ (2217) is 8°C per default with a possible setting range from -20°C to 20°C and determines the absoulte minimum boiler temp which is 'allowed' if the system isn't in general heating mode. So if one really changes the value to -20°C or turns it off completely - yes, then you probably could get some problems if the temperatures (within your house) drop that low.. But in my opinion you'll most likely already have other problems if temperatures are as low as -20°C (like frozen pipes or so)..

> I am not sure what physical measures are built to the heater itself outside of the LMS what would prevent those cases.)

Afaik: none. The LMS ist the BMU (boiler management unit) and therefore the system that is in charge. Any temp sensor is connected to it, afaik there is no 'temperature sensitive hardware relay' or something like that that kicks in if the LMS is somehow faulty - there is only the 2213 OEM which limits the maximum boiler temp at last instance. 
In the other way, if the LMS or a component is faulty and therefore problems occur, usually an error occurs somewhere within the system which most likely will generate a decent error message within the LMS and prevent the system to work anyway (depending on the kind of problem of course). 

I don't want to say that there is no risk at all (also while using the different security options of BSB-LAN and a decent VPN etc), but I think (or hope ;) ) that the risk of really harming your heating system (in a way that something really 'breaks') is little compared to just misconfiguring your settings as much as possible for driving you crazy if an intruder really gets that far and is interested in heating systems somehow..

GSzabados, 2021-08-17:
> I don't want to say that there is no risk at all (also while using the different security options of BSB-LAN and a decent VPN etc), but I think (or hope ;) ) that the risk of really harming your [...]

Honestly, I do not want to over exaggerate things, but a niche segment was the target of Stuxnet as well. ;)

Anyhow, I've just had a quick look on what HTTPS webserver libraries are available, and came across the following one(s), unfortunately this is for ESP32 only:

https://github.com/fhessel/esp32_https_server_compat - which uses the following one

https://github.com/fhessel/esp32_https_server

With the compat it should be working with the same codebase as the built-in WebServer.h

fredlcore, 2021-10-31:
I was only able to have a look at this now, sorry for the delay, but I don't think that this is a viable solution. It would require a separate openssl installation or other means to generate a self-signed certificate. Most modern browsers will also display a security warning about self signed certificates which might raise further questions by novice users.
Without providing a certificate dummy, compilation will fail. But if we provide a certificate dummy, most people will just be using that which would basically mean zero protection - on the contrary, people might feel secure because they have an https URL.

Yes, we could only activate it via a confiuration option, but still people just activate it because they want the security and then run into the same problems. Furthermore, once we start providing pre-compiled binaries, we would have to include a generic certificate and therefore run into the same problems as above.

Unless I'm overlooking something here, I don't think that an SSL server would be feasible.

fredlcore, 2021-12-25:
Since there hasn't been any further discussion for a while, I'm closing this for now. Feel free to discuss further, but for the moment, it'll be off of my todo list...

closed 2021-12-25 by fredlcore

GSzabados, 2021-12-25:
@fredlcore, thanks! I agree with what you have wrote before regarding the dummy certificate and indeed it is making things far more complicated...