fredlcore/BSB-LAN #388: Changing the bustype via webconfig vs. /Px,y,z

1coderookie, 2021-09-04:
I recently discovered that changing the busytpe via webconfig doesn't seem to initiate a new query of the parameters 6225 and 6226 as it does when using the /Px,y,z command. 

Here is an example when changing the bustype from LPB to BSB using /P0,66,0. After setting it, 6225&6226 are queried for the autodetection of the connected controller:
```
11:05:50.777 -> GET / HTTP/1.1
11:05:51.009 -> GET /P0,66,0 HTTP/1.1
11:05:51.805 -> LAN->HEIZ QUR 6225 Konfiguration - Gerätefamilie: 
11:05:51.805 -> DC C2 00 0B 06 3D 05 00 02 52 88
11:05:51.805 -> HEIZ->LAN ANS 6225 Konfiguration - Gerätefamilie: 90
11:05:51.805 -> DC 80 42 0E 07 05 3D 00 02 00 00 5A C9 27
11:05:51.838 -> #6225: 90
11:05:52.004 -> LAN->HEIZ QUR 6226 Konfiguration - Gerätevariante: 
11:05:52.004 -> DC C2 00 0B 06 3D 05 00 03 42 A9
11:05:52.004 -> HEIZ->LAN ANS 6226 Konfiguration - Gerätevariante: 109
11:05:52.004 -> DC 80 42 0E 07 05 3D 00 03 00 00 6D F9 27
11:05:52.037 -> #6226: 109
11:05:52.037 -> Device family: 90
11:05:52.037 -> Device variant: 109
```
If you do the same by using the webconfig interface (change the type and click 'save settings'), this doesn't occur.

dukess, 2021-09-07:
If device was automatically restarted after bus changing via webconfig then it should be ok (device will reconfigure at startup), in other case it is a problem.

1coderookie, 2021-09-07:
After changing bustype via webconfig,  no reboot happens  - just the EEPROM dump appeares in the serial monitor. 
When changing via /P command, also no reboot takes place, but 6225&6226 will be queried right afterwards.

dukess, 2021-09-07:
clear. i will investigate it.

1coderookie, 2021-09-07:
Great, thanks! But maybe check again at your system before you start, just to make sure that it's not a local problem somehow.. ;)

dukess, 2021-09-07:
It is global problem. Long time ago device was being restarted after bus changing. But after code rewriting i forgot to add bus reinitialization after config changing. :-(

1coderookie, 2021-09-07:
Wow, that was quick! :) 
Thanks!

closed 2021-09-07 by 1coderookie

dukess, 2021-09-07:
You thanks me too early, first check the fix! ;-)